pbatard/rufus #1763: Error: No default or UI configuration directive found - when trying boot android pendrive on Windows XP

meucat, 2021-07-23:
﻿

Checklist
---------
- [ ] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [ ] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [ ] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [ ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
First I installed Rufus 3.14 on Windows 10 and formatted 32 GB pendrive with android-x86-9.0-r2.iso image , then I put pendrive in Windows XP notebook (32 bits) and error came "No default or UI config...".
Then I downloaded Rufus 2.18 on Windows XP notebook and ran Rufus direct on that machine, and again the same error.
Next I learnt that this error could be because files and directories should be renamed from "isolinux" to "syslinux" and same error came again
Next I leartn that this error could be because fast format so uncheck fast format and error came again

Thanks for any solution

Miguel

Log
---
```
<FULL LOG>
```

pbatard, 2021-07-23:
No log and did not follow the check list. Sorry, but the check list exists for a reason, and, as mentioned when you created the issue, if you __choose__ not not follow the check list as requested, and especially if you choose not to provide the requested log, then you agree that I may close the issue without further notice.

closed 2021-07-23 by pbatard

meucat, 2021-07-23:
Ok sorry I will run rufus again with log and check all entries to resubmit this post

pbatard, 2021-07-23:
Okay, if you edit the post and attach a log, I will reopen it.
Please be mindful however that, as indicated in the FAQ, I no longer support anything that has to do with Windows XP. So make sure you create the drive using Windows 10.