pbatard/rufus #1764: [Error : Drive has a unknown master boot record]

AiryPP, 2021-07-31:
﻿

Checklist
---------
- [x ] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [x ] I also tried one or more of the following:
  - [x] Using a different USB drive.
  - [x] Plugging the USB into a different port.
  - [x] Running Rufus on a different computer.
- [x ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
<Please describe your issue here>

Log
---
```
<FULL LOG>
```
Rufus x86 v3.14.1788
Windows version: Windows 10 64-bit (Build 19041.746)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.04
System locale ID: 0x0409 (en-US)
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Notice: The ISO download feature has been deactivated because 'Check for updates' is disabled in your settings.
Found USB 2.0 device 'SanDisk Cruzer Blade USB Device' (0781:5567)
1 device found
Disk type: Removable, Disk size: 16 GB, Sector size: 512 bytes
Cylinders: 1869, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0xAAB49E03
Drive has a Windows 7 Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 14.3 GB (15375269888 bytes)
  Start Sector: 2048, Boot: Yes
Scanning image...
ISO analysis:
  Image is an ISO9660 image
  Detected Grub version: 2.06
Disk image analysis:
  Image has an unknown Master Boot Record
  Image is a bootable disk image
ISO label: 'GARUDA_DR460NIZED_SOARING_'
  Size: 2.3 GB (Projected)
  Uses: GRUB2
  Uses: EFI
Note: DD image mode enforced since this ISOHybrid is not ISO mode compatible.
Using image: garuda-dr460nized-linux-zen-210729.iso (2.3 GB)

Format operation started
Requesting disk access...
Will use 'E:' as volume mountpoint
Deleting partitions:
● Partition 1 (offset: 1048576, size: 14.3 GB)
Opened \\.\PhysicalDrive1 for exclusive write access
Requesting logical volume handle...
+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++Remounted \\?\Volume{ffb13b3c-f13e-11eb-8acc-5454e598ba10}\ as E:

Found USB 2.0 device 'SanDisk Cruzer Blade USB Device' (0781:5567)
1 device found
Disk type: Removable, Disk size: 16 GB, Sector size: 512 bytes
Cylinders: 1869, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x00000000
Drive has an unknown Master Boot Record
Partition 2:
  Type: EFI FAT (0xef)
  Size: 4 MB (4194304 bytes)
  Start Sector: 4824948, Boot: No
```

pbatard, 2021-07-31:
I'm not seeing any error here (`Drive has an unknown Master Boot Record` is __NOT__ an error message), and you wrote Garuda in DD mode. This means that, whatever issue you observe has nothing to do with Rufus (you will get the exact same problem if you use Etcher or win32diskimager) and you need to take it with Garuda support.

&rarr; Closing.

closed 2021-07-31 by pbatard

AiryPP, 2021-07-31:
This also happens with other arch distro's or any other linux distro's like Manjaro

pbatard, 2021-07-31:
Again, `Drive has an unknown Master Boot Record` is not an error. It's normal behaviour.

I'm going to assume that you are simply being confused about your drive content somehow "disappearing" from Windows after you write it in DD mode, which is the only modes Manjaro and Garuda allow, and are under the false impression that, because you can't see data in Windows (or you see the drive being strongly reduced in size, because only the EFI System Partition is visible), you think that the drive has a problem.

This is a __wrong assumption__.

Your drive was created properly and it's not because you don't seem to see its data in Windows that it will not work.

__Please try to boot it__

Unfortunately, this issue is the result of Linux distro maintainers like Garuda, Red Hat/CentOS, Manjaro and others, thinking that only DD mode should ever be used to create bootable media, whereas it is actually __SUPER CONFUSING FOR WINDOWS USERS__ (and the reason why [Rufus tries to promote ISO mode over DD mode](https://github.com/pbatard/rufus/wiki/FAQ#Why_doesnt_Rufus_recommend_DD_mode_over_ISO_mode_for_ISOHybrid_images_Surely_DD_is_better), __where possible__, because in ISO mode, you can always see the media content in Windows).

So please try to boot your media.

And please please please, report your confusion, and why you thought that your drive had an issue, to the Garuda and Manjaro maintainers, so that they stop ignoring a real issue that is actually preventing people from installing their distro.

There are plenty of other distributions (Debian, Arch, Mint, openSUSE, Ubuntu, etc) that make sure that they are compatible with both ISO mode and DD mode, even when using an ISOHybrid image, so it makes no sense for Garuda and Manjaro to penalize their users by not doing the same. However, they will only acknowledge that this is a problem if users like you report that it effectively prevented them to try to boot the media, because you thought that it wasn't created properly.

AiryPP, 2021-07-31:
Sorry I read it wrongly. I'll try and boot the USB and try and update you. Thank you for the support!