pbatard/rufus #1766: Operation system nor found

Florian-Mahlberg, 2021-08-09:
﻿

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [ ] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [ ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [x] I also tried one or more of the following:
  - [x] Using a different USB drive.
  - [x] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
Hi, I am using rufus a long time and on many PCs. But this time, when I boot any Image/File that is bootable (works fine in VM or was created by myself) I always get an error on my Laptop, where I am trying to boot from the USB. This Error I had since Rufus version 3.14.* or lower. After I boot from the USB (legacy without secure boot) I just get the message "operation system not found". I am not completely sure if this message is hard coded in the bios, but I am using a Dell Laptop. The last time I installed Windows on it, I used the official Microsoft tool to install Windows, but now I can't do that with some other OS. 

Sidenote:
I tried booting from Etcher too, and got another error. It was written in pixels (hex) and seemed like a permission's problem.

Log
---
```
Rufus x86 v3.15.1812
Windows version: Windows 10 64-bit (Build 19042.1110)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.06
System locale ID: 0x0409 (en-US)
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found USB 2.0 device 'Intenso Rainbow Line USB Device' (058F:6387)
1 device found
No volume information for drive 0x81
Disk type: Removable, Disk size: 64 GB, Sector size: 512 bytes
Cylinders: 7648, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 0
Disk ID: 0x00000000
Drive has an unknown Master Boot Record
Scanning image...
ISO analysis:
  Image is an ISO9660 image
Disk image analysis:
  Image does not have a Boot Marker
ISO label: 'CDROM'
  Size: 4 KB (Projected)
Scanning image...
ISO analysis:
  'C:\Users\flo\Desktop\OWNOS\bootHelloWorld\boot.img' doesn't look like an ISO image
Disk image analysis:
  Image has an unknown Master Boot Record
  Image is a bootable disk image
Using image: boot.img (512 bytes)
Scanning image...
ISO analysis:
  Image is an UDF image
Disk image analysis:
  Image does not have a Boot Marker
ISO label: 'ESD-ISO'
  Size: 4.6 GB (Projected)
  Note: File on disk is larger than reported ISO size by 570 KB...
  Uses: EFI
  Uses: Bootmgr (BIOS and UEFI)
  Uses: Install.esd (version 0.14.0+)
Using image: Windows.iso (4.6 GB)

[... 933 lines cut ...]
Extracting: E:\support\logging\setupugcetw.dll (13.4 KB)
Extracting: E:\support\logging\sysprepetw.dll (17.9 KB)
Extracting: E:\support\logging\windeployetw.dll (13.9 KB)
Extracting: E:\support\logging\winsetupetw.dll (64.9 KB)
Extracting: E:\autorun.inf (128 bytes)
Extracting: E:\bootmgr (404 KB)
Extracting: E:\bootmgr.efi (1.5 MB)
Extracting: E:\setup.exe (72.4 KB)
Finalizing, please wait...

Found USB 2.0 device 'Intenso Rainbow Line USB Device' (058F:6387)
1 device found
Disk type: Removable, Disk size: 64 GB, Sector size: 512 bytes
Cylinders: 7648, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 1
Disk GUID: {3BD56A9E-66E9-4290-A92C-D2C95353C7B3}
Max parts: 128, Start Offset: 17408, Usable = 62914525696 bytes
Partition 1:
  Type: Microsoft Basic Data Partition
  Name: 'Main Data Partition'
  ID: {7CC3D773-3C0E-44E3-9749-8A4F7086E35C}
  Size: 58.6 GB (62913494528 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000

```

pbatard, 2021-08-09:
>  I always get an error on my Laptop, where I am trying to boot from the USB.

So if I understand correctly, you only get the issue with a specific laptop. All the other computers you tested with the same Rufus created drive seem fine, and it's just that laptop that has an issue, right?

Also, if you get an "error" then I'm going to assume that there is an error message. If so what is it?
If not, you may have to replace "error" with something like "freeze" or "black screen" or something else, as "error" is generally reserved to indicate an operation that produces an error message onscreen.

Or is your error this _"I just get wierd Pixels"_ you describe?

Now, you are also saying: _"After I boot from the USB (legacy without secure boot)"_, which, considering the output of your log, makes me pretty confident that your issue is that you are trying to boot a **UEFI-only** media in BIOS/Legacy mode, which would normally produce the following screen:
[image]

So, my guess is that your _"weird pixel"_ screen is just a garbled version of this message, which has everything to do with [this relevant FAQ entry](https://github.com/pbatard/rufus/wiki/FAQ#Why_doesnt_Rufus_create_a_Windows_installation_USB_that_can_be_booted_in_dual_BIOSUEFI_mode).

In short, to be able to boot the Windows installation media on your laptop, you __MUST__ create it with a _Target system_ of `BIOS (or UEFI-CSM)` and not `UEFI (non CSM)` __as you are currently doing__.

So please change _Partition scheme_ from `GPT` to `MBR` in Rufus and make sure _Target system_ says `UEFI (non CSM)` and, if you recreate your Windows USB, you should find that it works much better...

Florian-Mahlberg, 2021-08-09:
Sorry for opening this "Bug". You were right:

> So, my guess is that your "weird pixel" screen is just a garbled version of this message

closed 2021-08-09 by Florian-Mahlberg