pbatard/rufus #1768: Unusable flash drive after a failed Rufus install

kiobu, 2021-08-15:
﻿

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [x] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [x] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [x] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
I originally attempted to flash Debian 11 Bullseye onto my flash drive, which had been working flawlessly prior to this. About 90% of the way through copying the files, I received the "Flash drive is in use by another process (explorer.exe)" error. My flash drive was then put in an unusable state, I am unsure if it gave it write protection or what, but now Rufus cannot clear any partitions left over on the flash drive from the Debian install, and it seems as if the flash drive disconnects or Windows loses a handle on it immediately when attempting to write to it. I have tried to use `diskpart` to clean the drive, remove readonly flags, etc., but to no avail. Just like Rufus, it seems to disconnect the drive immediately when anything attempts to write to the drive. Re-plugging the flash drive in though, Windows detects it fine; however, the drive letter is removed each time.

I'm not convinced the flash drive has any bad blocks or is dead for any reason, as I had just used it previously to install Solus without any issues, and I had been removing and adding partitions to it readily only minutes before formatting it with Rufus and installing Debian. It seems to have been caused by the interruption in copying Debian to the flash drive. I know Rufus does not brick or corrupt hardware, but I am unsure exactly what happened.

I do not seem to be using any Rufus-incompatible software.

[image]

Here is the log where I attempt to flash a non-bootable image with the 'Check device for bad blocks' selected, and a drive letter already assigned.

Log
---
```
Rufus x86 v3.15.1812
Windows version: Windows 10 64-bit (Build 17134.1304)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.06
System locale ID: 0x0409 (en-US)
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found USB 3.0 device 'Kingston DataTraveler 3.0 USB Device' (0951:1666)
Using autorun.inf label for drive R: 'Install Debian GNU/Linux'
Found USB 3.0 device 'TOSHIBA External USB 3.0 USB Device' (0480:0900)
Device eliminated because it was detected as a Hard Drive (score 23 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives under 'advanced drive properties'
1 device found
Disk type: Removable, Disk size: 16 GB, Sector size: 512 bytes
Cylinders: 1881, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x033D81A3
Drive has a Syslinux Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 14.4 GB (15470998528 bytes)
  Start Sector: 2048, Boot: Yes

Format operation started
Requesting disk access...
Will use 'R:' as volume mountpoint
Deleting partitions:
● Partition 1 (offset: 1048576, size: 14.4 GB)
Could not delete partition: [0x80070002] The system cannot find the file specified.
Could not open GUID volume '\\?\Volume{9f84dc0a-fddf-11eb-93df-2cf05d8aa5ff}': [0x000003EE] The volume for a file has been externally altered so that the opened file is no longer valid.
Could not open \\.\PhysicalDrive6: [0x00000002] The system cannot find the file specified.
Could not open \\.\PhysicalDrive6: [0x00000002] The system cannot find the file specified.
```

For posterity, here is a log where I attempt to flash a non-bootable image with the 'Check device for bad blocks' selected, but a drive letter is assigned by Rufus (which results in the absence of the 0x000003EE error).

```
Rufus x86 v3.15.1812
Windows version: Windows 10 64-bit (Build 17134.1304)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.06
System locale ID: 0x0409 (en-US)
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found USB 3.0 device 'Kingston DataTraveler 3.0 USB Device' (0951:1666)
Found USB 3.0 device 'TOSHIBA External USB 3.0 USB Device' (0480:0900)
Device eliminated because it was detected as a Hard Drive (score 23 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives under 'advanced drive properties'
1 device found
Disk type: Removable, Disk size: 16 GB, Sector size: 512 bytes
Cylinders: 1881, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x033D81A3
Drive has a Syslinux Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 14.4 GB (15470998528 bytes)
  Start Sector: 2048, Boot: Yes
Found USB 3.0 device 'Kingston DataTraveler 3.0 USB Device' (0951:1666)
Found USB 3.0 device 'TOSHIBA External USB 3.0 USB Device' (0480:0900)
2 devices found
Disk type: Removable, Disk size: 16 GB, Sector size: 512 bytes
Cylinders: 1881, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x033D81A3
Drive has a Syslinux Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 14.4 GB (15470998528 bytes)
  Start Sector: 2048, Boot: Yes
Found USB 3.0 device 'Kingston DataTraveler 3.0 USB Device' (0951:1666)
Found USB 3.0 device 'TOSHIBA External USB 3.0 USB Device' (0480:0900)
Device eliminated because it was detected as a Hard Drive (score 23 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives under 'advanced drive properties'
1 device found
Disk type: Removable, Disk size: 16 GB, Sector size: 512 bytes
Cylinders: 1881, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x033D81A3
Drive has a Syslinux Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 14.4 GB (15470998528 bytes)
  Start Sector: 2048, Boot: Yes

Format operation started
Requesting disk access...
No drive letter was assigned...
Will use 'D:' as volume mountpoint
Deleting partitions:
● Partition 1 (offset: 1048576, size: 14.4 GB)
Could not delete partition: [0x80070002] The system cannot find the file specified.
Could not open \\.\PhysicalDrive6: [0x00000002] The system cannot find the file specified.
Could not open \\.\PhysicalDrive6: [0x00000002] The system cannot find the file specified.
```

kiobu, 2021-08-15:
Also, in the event that this flash drive is unrecoverable, I have a few spare USB drives. Are there steps I can take to ensure that no program will interfere with Rufus when attempting to write to the flash drive? I am aware you cannot keep a write lock in the flash drive forever since Windows requires it to be unmounted eventually, but I would like to minimize the chance if possible.

kiobu, 2021-08-15:
New flash drive worked fine, so I guess maybe the old one had some hardware issue.

pbatard, 2021-08-15:
Well, the problem is that, as far as USB flash drives are concerned, things tend to work fine... until they fail completely,

A lot of people seem to expect that they'll see a gradual failure, and therefore come here with the same message that their drive can't have failed on its own, and that there must either be a cause for the failure, or a way to recover it.

My message here is always the same, short of a couple very specific Windows bugs (that wouldn't manifest themselves with a Debian 11 ISO), if you see a failure from every possible avenue in Windows, it means your drive is most likely day, and that trying to recover it is a waste of time. At this stage, you do want to confirm that you also can't recover your drive from a different machine, preferably running Linux, and, if that is the case, just stop wasting your time trying to:

1. Figure out what might have happened that could have caused the failure (No, it wasn't software related. The Windows Hardware Abstraction layer, that an application like Rufus uses to access the drive prevents it).
2. Recover the drive by trying to reflash its internal firmware.

Instead, you should RMA your drive, if it's still under warranty or, if it's no longer under warranty, understand that if flash drive warranties aren't (usually) that long, it's probably because the manufacturers themselves don't expect their hardware to last...

kiobu, 2021-08-15:
I didn't expect a gradual failure, but I also didn't expect it to fail so soon after working so well. Really I just wanted a possible explanation as to why it failed during an install through Rufus, or maybe if you've seen the same issue before (drive connecting and disconnecting when written to.)

Not an issue anyways, I've used my other flash drive successfully. Thanks for the response.

closed 2021-08-15 by kiobu