pbatard/rufus #1769: W10 1507 ISO download failed

djsomi, 2021-08-16:
﻿

Checklist
---------
- [X] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [X] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [X] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [ X The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
Cant download old version of W10 1507 (anything above is OK), getting the following error in the log:

Log
---
```
Rufus x86 v3.15.1812
Windows version: Windows 10 64-bit (Build 19043.1165)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.06
System locale ID: 0x0409 (en-US)
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found USB 2.0 device 'Kingston DataTraveler 3.0 USB Device' (0951:1666)
NOTE: This device is a USB 3.0 device operating at lower speed...
Found USB 2.0 device 'APPLE SD Card Reader USB Device' (05AC:8403)
Device eliminated because it appears to contain no media
1 device found
Disk type: Removable, Disk size: 16 GB, Sector size: 512 bytes
Cylinders: 1911, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x011C312C
Drive has a Rufus Master Boot Record
Partition 1:
  Type: NTFS (0x07)
  Size: 14.6 GB (15723659264 bytes)
  Start Sector: 2048, Boot: Yes
Downloading https://github.com/pbatard/Fido/releases/download/v1.21/Fido.ps1.lzma
File length: 15.3 KB
Successfully downloaded 'Fido.ps1.lzma'
Download signature is valid ✓
Script signature is valid ✓
Please Wait...
Querying https://www.microsoft.com/en-US/software-download/
Querying https://www.microsoft.com/en-US/api/controls/contentinclude/html?pageId=a8f8f489-4c7f-463a-9ca6-5cff94d8d041&host=www.microsoft.com&segments=software-download,Windows10ISO&query=&action=getskuinformationbyproductedition&sessionId=e3913272-501c-4040-8075-97573623bb06&productEditionId=79&sdVersion=2
Exception calling "ShowDialog" with "0" argument(s): "Cannot process argument transformation on parameter 'SkuId'. Cann
ot convert value "FQC-08929" to type "System.Int32". Error: "Input string was not in a correct format.""
At C:\Users\xxx\AppData\Local\Temp\{ACF0B8A3-1FAE-44D4-8199-412D74B5E2D8}.ps1:1061 char:1
+ $XMLForm.ShowDialog() | Out-Null
+ ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
    + CategoryInfo          : NotSpecified: (:) [], ParentContainsErrorRecordException
    + FullyQualifiedErrorId : ParameterBindingArgumentTransformationException
Exited download script with code: 1

```

pbatard, 2021-08-19:
Thanks for the report. I can replicate the issue and I should have a fix for it soon.

closed 2021-08-19 by pbatard

djsomi, 2021-08-22:
Thanks, working fine now!