pbatard/rufus #1770: EFI (only bios) unknown filesystem

lorenzomarialolli, 2021-08-18:
﻿

Checklist
---------
- [X] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [X] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [X] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [X] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [X] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [X] I also tried one or more of the following:
  - [X] Using a different USB drive.
  - [X] Plugging the USB into a different port.
  - [X] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
Greetings,
I've found this problem after many hours of testing and googling. Basically I've created a modified version of ubuntu live iso, with distroshare imanager, and then transfered to usb key via rufus. It doesnt' work, on EFI ONLY bios (tested on 2 different devices, HP Probook): it returns the grub error "unknown filesystem". I started my search thinking of a problem related to my own linux version, and after a lot of tests, I've tryed to replicate the problem using the command (from linux):

dd if=live-cd.iso of=/dev/sdb status=progress bs=8192

and it worked!
I've tested many different way with rufus to create the usb key, but none of them worked on EFI BIOS (but it worked on old/CSM bios), but none worked for me. Can you please help me out?

Any help would be appreciated

Log
---
```
<FULL LOG>
Rufus x86 v3.15.1812 (Portable)
Windows version: Windows 10 64-bit (Build 19042.1165)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.06
System locale ID: 0x0410 (it-IT)
Will use default UI locale 0x0410
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'it-IT'
Notice: The ISO download feature has been deactivated because 'Check for updates' is disabled in your settings.
Found USB 2.0 device 'S31B0912 USB DISK USB Device' (090C:1000)
1 device found
No volume information for drive 0x81
Disk type: Removable, Disk size: 4GB, Sector size: 512 bytes
Cylinders: 487, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 4
Disk GUID: {E37A8CAC-AB4C-4CD3-AABB-702263A332E7}
Max parts: 176, Start Offset: 32768, Usable = 4009698816 bytes
Partition 1:
  Type: Microsoft Basic Data Partition
  Name: 'Gap0'
  ID: {E37A8CAC-AB4C-4CD3-AABA-702263A332E7}
  Size: 104 KB (106496 bytes)
  Start Sector: 64, Attributes: 0x1000000000000001
Partition 2:
  Type: EFI System Partition
  Name: 'EFI boot partition'
  ID: {E37A8CAC-AB4C-4CD3-AAB9-702263A332E7}
  Size: 2.8 MB (2949120 bytes)
  Start Sector: 272, Attributes: 0x1000000000000001
Partition 3:
  Type: Apple HFS+ Partition
  Name: 'HFSPLUS'
  ID: {E37A8CAC-AB4C-4CD3-AAB8-702263A332E7}
  Size: 3 GB (3242848256 bytes)
  Start Sector: 6032, Attributes: 0x1000000000000001
Partition 4:
  Type: Microsoft Basic Data Partition
[... 686 lines cut ...]
Extracting: D:\System\Library\CoreServices\.disk_label (3.1 KB)
Extracting: D:\System\Library\CoreServices\.disk_label.contentDetails (23 bytes)
Extracting: D:\.disk\release_notes_url (23 bytes)
Extracting: D:\EFI\BOOT\BOOTX64.EFI (from '/efi.img', 152 KB)
Finalizing, please wait...
Created: D:autorun.inf
Created: D:autorun.ico

Found USB 2.0 device 'S31B0912 USB DISK USB Device' (090C:1000)
Using autorun.inf label for drive D: 'ISOIMAGE'
1 device found
Disk type: Removable, Disk size: 4GB, Sector size: 512 bytes
Cylinders: 487, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 1
Disk GUID: {4B77FE82-EAA3-4D66-914B-650292F1A01B}
Max parts: 128, Start Offset: 17408, Usable = 4009720320 bytes
Partition 1:
  Type: Microsoft Basic Data Partition
  Name: 'Main Data Partition'
  ID: {B33BA6F4-0FA5-47AC-9CF6-5945C545194D}
  Size: 3.7 GB (4008689152 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
0 devices found

```

pbatard, 2021-08-19:
> I've created a modified version of ubuntu live iso

Unless you are very familiar with the latest ISOHybrid image generation from Ubuntu, I wouldn't try creating a modified Live ISO. You are asking for trouble.

> it returns the grub error "unknown filesystem"

Most likely because you rely on the infamous `efi.img` hack instead of mapping the EFI bootloaders from the ESP on top of the ISO, like the latest Ubuntu does. The 152 KB `BOOTX64.EFI` that's extracted from your `efi.img` looks way too small to be a proper GRUB EFI bootloader (for instance if you look at `ubuntu-21.04-desktop-amd64.iso`'s `/EFI/boot/bootx64.efi` you'll see that it's about 1.3 MB.

So I'm not surprised you are getting GRUB errors.

You should first follow what Ubuntu maintainers are doing exactly, and make sure that works. Then add your changes.

Remember that because ISOHybrid is essentially a hack, creating a proper ISOHybrid requires a lot of expertise, so if you try to take a shortcut, you're most likely going to end with exactly the problem you ran into above.

Please spend time replicating the current Ubuntu ISO generation and then take it from here.

closed 2021-08-19 by pbatard

lorenzomarialolli, 2021-08-20:
Hello,

thank you for your prompt answer.

> > I've created a modified version of ubuntu live iso
> 
> Unless you are very familiar with the latest ISOHybrid image generation from Ubuntu, I wouldn't try creating a modified Live ISO. You are asking for trouble.

Probably you are right, anyway I need it.
> Most likely because you rely on the infamous `efi.img` hack instead of mapping the EFI bootloaders from the ESP on top of the ISO, like the latest Ubuntu does. The 152 KB `BOOTX64.EFI` that's [...]

I don't think so. there are some reason for that:

- the ISO is working with dd and etcher
- the ISO is working in EFI with rufus using some command during grub-rescue stage (I can share them, but those are not important I think)

> 
> So I'm not surprised you are getting GRUB errors.
> 
> You should first follow what Ubuntu maintainers are doing exactly, and make sure that works. Then add your changes.

Agree, thats what I've done, and the ISO is working aside of rufus
> Remember that because ISOHybrid is essentially a hack, creating a proper ISOHybrid requires a lot of expertise, so if you try to take a shortcut, you're most likely going to end with exactly the [...]

I've used rufus for so much time and I appreciate your job, so please do not take my issue in the wrong way. I think rufus is a great piece of software and you are very knowledgable. I've appreciated so much information found in you FAQ section and the experience with the software itself. I've used it for year without any issue for old LEGACY bios or with CSM.

Regards,
LL

pbatard, 2021-08-20:
> the ISO is working with dd and etcher

Read what I said about the `/EFI/boot/bootx64.efi` from your `efi.img`.

If your argument is that dd and Etcher work but not Rufus in ISO mode, clearly you don't understand what ISOHybrid is about. A proper ISOHybrid implementation needs to work in 3 ways:
1. When burned to optical media and booted from CD/DVD
2. When written to a disk-based media (e.g. USB Flash Drive) as a flat image (i.e. DD mode), which is the __ONLY__ mode that Etcher and `dd` allow, and then booted as a BIOS or UEFI disk.
3. When the ISO content is extracted to a FAT32 partition and then booted as a UEFI disk (i.e. ISO mode)

Unlike official Ubuntu images, you haven't done 3. So it's not a Rufus issue. The issue is that you haven't done your homework and don't understand that ISOHybrid doesn't stop at DD mode. You should have UEFI bootloaders in your ISO content, which you don't have. Fix that.

> Agree, thats what I've done, and the ISO is working aside of rufus

If the ISO isn't working in Rufus, you __clearly__ haven't done that.

Trying to shift your problems onto Rufus will not accomplish anything. __YOU__ need to fix your image generation, instead of expecting me to magically alter Rufus to handle half-assed ISOHybrids.

lorenzomarialolli, 2021-08-20:
> Read what I said about the `/EFI/boot/bootx64.efi` from your `efi.img`.

Yes, I've read about that.
> If your argument is that dd and Etcher work but not Rufus in ISO mode, clearly you don't understand what ISOHybrid is about. A proper ISOHybrid implementation needs to work in 3 ways:
>
> 1. When [...]
I'll make more investigation on point 3 and on  152 KB BOOTX64.EFI

> Unlike official Ubuntu images, you haven't done 3. So it's not a Rufus issue. The issue is that you haven't done your homework and don't understand that ISOHybrid doesn't stop at DD mode. You should have proper UEFI bootloaders in your ISO content, which you don't have. Fix that.

I'll check it.
> 
> > Agree, thats what I've done, and the ISO is working aside of rufus
> 
> If the ISO isn't working in Rufus, you **clearly** haven't done that because all the official Ubuntu ISOs do.
> 
> Trying to shift your problems onto Rufus will not accomplish anything. **YOU** need to fix your image generation, instead of expecting me to magically alter Rufus to handle half-assed ISOHybrids.

There is no need to answer on defensive. Probably not the best manner, but I'll come back after more investigations.
Thank you for your kind support.