pbatard/rufus #1773: Error 0xC0030057 when creating persistent drive on Windows 10

Teramanbr, 2021-08-30:
Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [x] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [x] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [x] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
Was trying to install Ubuntu with persistent memory on my 64GB Pen Drive when i got this error at 53.1%.
Searched for it on rufus github, found one old post about Windows 7, but i am using Windows 10, 
Have tried installing it without persistence, works like a charm, but can't make a persistent Ubuntu.
I did NOT run it on compatibility mode.
Should i try doing it? What do i do?

Log
---
```
Rufus x86 v3.15.1812
Windows version: Windows 10 64-bit (Build 19043.1149)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.06
System locale ID: 0x0416 (pt-BR)
Will use default UI locale 0x0416
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'pt-BR'
NOTE: The Portuguese Brazilian (Português do Brasil) translation requires an update, but the current translator hasn't submitted one. Because of this, some messages will only be displayed in English.
If you think you can help update this translation, please e-mail the author of this application
Found USB 2.0 device 'USB 2.0 PenDrive USB Device' (0123:4567)
1 device found
Disk type: Removable, Disk size: 67 GB, Sector size: 512 bytes
Cylinders: 8158, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 2
Disk ID: 0x2CF4BA3A
Drive has an unknown Master Boot Record
Partition 1:
  Type: Extended LBA (0x0f)
  Size: 62.5 GB (67106799616 bytes)
  Start Sector: 2048, Boot: No
Partition 5:
  Type: NTFS (0x07)
  Size: 62.5 GB (67106766848 bytes)
  Start Sector: 2112, Boot: No
Scanning image...
ISO analysis:
  Image is an ISO9660 image
  Will use '/isolinux/isolinux.cfg' for Syslinux
  Detected Syslinux version: 6.04/20191223 (from '/isolinux/isolinux.bin')
Disk image analysis:
  Image has an unknown Master Boot Record
  Image is a bootable disk image
ISO label: 'Ubuntu 20.04.3 LTS amd64'
  Size: 2.9 GB (Projected)
  Has a >64 chars filename
  Uses: Syslinux/Isolinux v6.04
  Uses: EFI
  Note: This ISO uses symbolic links, which will not be replicated due to file system
  limitations. Because of this, some features from this image may not work...
Using image: ubuntu-20.04.3-desktop-amd64.iso (2.9 GB)
Will reuse 'ldlinux.sys' and 'ldlinux.bss' from './Rufus/Rufus/syslinux-6.04/20191223/' for Syslinux installation

Format operation started
Requesting disk access...
Will use 'H:' as volume mountpoint
Deleting partitions:
● Partition 0 (offset: 1048576, size: 62.5 GB)
Could not delete partition: [0x80042408] The extended partition is not empty.
● Partition 1 (offset: 1081344, size: 62.5 GB)
Opened \\.\PhysicalDrive2 for shared write access
Analyzing existing boot records...
Drive has an unknown Master Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 2048 sectors
Initializing disk...
Partitioning (MBR)...
● Creating Main Data Partition (offset: 1048576, size: 22.5 GB)
● Creating Linux Persistence Partition (offset: 24159179776, size: 40.0 GB)
Waiting for logical drive to reappear...
Using Ubuntu-like method to enable persistence
Notice: Using physical device to access partition data
Formatting (ext3)...
655359 possible inodes out of 10485759 blocks (block size = 4096)
524287 blocks (5.0%) reserved for the super user
Creating 320 inode sets: [1 marker = 4.0 set(s)]
++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
Creating 32768 journal blocks: [1 marker = 409.6 block(s)]
++++++
Could not create ext3 journal: Windows error code 0xC0030057 (FormatMessage error code 0x00003AFC)
Re-mounted volume as H: after error

Found USB 2.0 device 'USB 2.0 PenDrive USB Device' (0123:4567)
1 device found
No volume information for drive 0x82
Disk type: Removable, Disk size: 67 GB, Sector size: 512 bytes
Cylinders: 8158, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 2
Disk ID: 0x0005526D
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 22.5 GB (24158131200 bytes)
  Start Sector: 2048, Boot: Yes
Partition 2:
  Type: GNU/Linux (0x83)
  Size: 40.0 GB (42949670400 bytes)
  Start Sector: 47185898, Boot: No
Found USB 2.0 device 'USB 2.0 PenDrive USB Device' (0123:4567)
1 device found
No volume information for drive 0x82
Disk type: Removable, Disk size: 67 GB, Sector size: 512 bytes
Cylinders: 8158, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 2
Disk ID: 0x0005526D
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 22.5 GB (24158131200 bytes)
  Start Sector: 2048, Boot: Yes
Partition 2:
  Type: GNU/Linux (0x83)
  Size: 40.0 GB (42949670400 bytes)
  Start Sector: 47185898, Boot: No
Closed Windows format prompt
Will reuse 'ldlinux.sys' and 'ldlinux.bss' from './Rufus/Rufus/syslinux-6.04/20191223/' for Syslinux installation

Format operation started
Requesting disk access...
Will use 'H:' as volume mountpoint
Deleting partitions:
● Partition 1 (offset: 1048576, size: 22.5 GB)
● Partition 2 (offset: 24159179776, size: 40.0 GB)
Opened \\.\PhysicalDrive2 for shared write access
Analyzing existing boot records...
Drive has a Zeroed Master Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 2176 sectors
Initializing disk...
Partitioning (MBR)...
● Creating Main Data Partition (offset: 1048576, size: 22.5 GB)
● Creating Linux Persistence Partition (offset: 24159179776, size: 40.0 GB)
Waiting for logical drive to reappear...
Using Ubuntu-like method to enable persistence
Notice: Using physical device to access partition data
Formatting (ext3)...
655359 possible inodes out of 10485759 blocks (block size = 4096)
524287 blocks (5.0%) reserved for the super user
Creating 320 inode sets: [1 marker = 4.0 set(s)]
++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
Creating 32768 journal blocks: [1 marker = 409.6 block(s)]
++++++
Could not create ext3 journal: Windows error code 0xC0030057
Re-mounted volume as H: after error

Found USB 2.0 device 'USB 2.0 PenDrive USB Device' (0123:4567)
1 device found
No volume information for drive 0x82
Disk type: Removable, Disk size: 67 GB, Sector size: 512 bytes
Cylinders: 8158, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 2
Disk ID: 0x000600CD
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 22.5 GB (24158131200 bytes)
  Start Sector: 2048, Boot: Yes
Partition 2:
  Type: GNU/Linux (0x83)
  Size: 40.0 GB (42949670400 bytes)
  Start Sector: 47185898, Boot: No
Closed Windows format prompt
Found USB 2.0 device 'USB 2.0 PenDrive USB Device' (0123:4567)
1 device found
No volume information for drive 0x82
Disk type: Removable, Disk size: 67 GB, Sector size: 512 bytes
Cylinders: 8158, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 2
Disk ID: 0x000600CD
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 22.5 GB (24158131200 bytes)
  Start Sector: 2048, Boot: Yes
Partition 2:
  Type: GNU/Linux (0x83)
  Size: 40.0 GB (42949670400 bytes)
  Start Sector: 47185898, Boot: No
```

pbatard, 2021-09-12:
Sorry for the late reply.

The fact that this error occurs in the middle of creating the journal blocks could indicate an issue with your drive. Especially, the error code you get seems to come from a call to the basic `write()` within a loop, and there's very little reason to see that fail for one specific block of data when the others were okay (and when it certainly doesn't seem like we screwed up parameters such as trying to write a block that's outside our boundaries).

Do you mind posting a log containing the results of the bad blocks check?

Teramanbr, 2021-09-23:
Well, turns out its actually a problem on my pendrive apparently, here is the log
,,,
Rufus x86 v3.15.1812
Windows version: Windows 10 64-bit (Build 19043.1165)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.06
System locale ID: 0x0416 (pt-BR)
Will use default UI locale 0x0416
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'pt-BR'
Notice: The ISO download feature has been deactivated because 'Check for updates' is disabled in your settings.
NOTE: The Portuguese Brazilian (Português do Brasil) translation requires an update, but the current translator hasn't submitted one. Because of this, some messages will only be displayed in English.
If you think you can help update this translation, please e-mail the author of this application
Found USB 2.0 device 'USB 2.0 PenDrive USB Device' (0123:4567)
1 device found
Disk type: Removable, Disk size: 67 GB, Sector size: 512 bytes
Cylinders: 8158, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x55555555
Drive has an unknown Master Boot Record
Partition 1:
  Type: NTFS (0x07)
  Size: 62.5 GB (67106766848 bytes)
  Start Sector: 2048, Boot: No
Scanning image...
ISO analysis:
  Image is an ISO9660 image
  Will use '/isolinux/isolinux.cfg' for Syslinux
  Detected Syslinux version: 6.04/20191223 (from '/isolinux/isolinux.bin')
Disk image analysis:
  Image has an unknown Master Boot Record
  Image is a bootable disk image
ISO label: 'Linux Mint 20.2 Cinnamon 64-bit'
  Size: 2 GB (Projected)
  Uses: Syslinux/Isolinux v6.04
  Uses: EFI
Using image: linuxmint-20.2-cinnamon-64bit.iso (2 GB)
Will reuse 'ldlinux.sys' and 'ldlinux.bss' from './Rufus/Rufus/syslinux-6.04/20191223/' for Syslinux installation

Format operation started
Requesting disk access...
Will use 'H:' as volume mountpoint
Deleting partitions:
● Partition 1 (offset: 1048576, size: 62.5 GB)
Opened \\.\PhysicalDrive2 for shared write access
Analyzing existing boot records...
Drive has an unknown Master Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 2176 sectors
Initializing disk...
Bad Blocks: Checking from block 0 to 127999 (1 block = 512 KB)
Bad Blocks: Using offset 95837 for fake device check
Bad Blocks: Writing test pattern 0x55
Bad Blocks: Reading and comparing
Bad Blocks: 7552
Bad Blocks: 7553
Bad Blocks: 7554
Bad Blocks: 7555
Bad Blocks: 7556
Bad Blocks: 7557
Bad Blocks: 7558
Bad Blocks: 7559
Bad Blocks: 7560
Bad Blocks: 7561
Bad Blocks: 7562
Bad Blocks: 7563
Bad Blocks: 7564
Bad Blocks: 7565
Bad Blocks: 7566
Bad Blocks: 7567
[... 233 lines cut ...]
Bad Blocks: 7801
Bad Blocks: 7802
Bad Blocks: 7803
Bad Blocks: 7804
Bad Blocks: 7805
Bad Blocks: 7806
Bad Blocks: 7807
Too many bad blocks, aborting test
Bad Blocks: Check completed, 256 bad blocks found. (0/0/256 errors)
Re-mounted volume as H: after error

Found USB 2.0 device 'USB 2.0 PenDrive USB Device' (0123:4567)
1 device found
No volume information for drive 0x82
Disk type: Removable, Disk size: 67 GB, Sector size: 512 bytes
Cylinders: 8158, Tracks per cylinder: 255, Sectors per track: 63
Partition type: SFD (Super Floppy Disk) or unpartitioned
Found USB 2.0 device 'USB 2.0 PenDrive USB Device' (0123:4567)
1 device found
No volume information for drive 0x82
Disk type: Removable, Disk size: 67 GB, Sector size: 512 bytes
Cylinders: 8158, Tracks per cylinder: 255, Sectors per track: 63
Partition type: SFD (Super Floppy Disk) or unpartitioned
Closed Windows format prompt
,,,

pbatard, 2021-09-23:
There you go. There's a __very good reason__ the check list asks you to run a bad blocks check, and you also indicated that _"I ran a bad blocks check"_ when you filled it.

Please don't tick checkboxes just for the sake of it. The check list is not there just to look pretty, but it exists to help __YOU__ identify what your issue might be. Thank you.

closed 2021-09-23 by pbatard