pbatard/rufus #1775: Odd issue where SCCM image gets error 8007000B during apply Image when using new Rufus

mangomatt2, 2021-09-15:
﻿

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [ ] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [ ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [x] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [x] Using a different USB drive.
  - [x] Plugging the USB into a different port.
  - [x] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
Odd issue.   With a 64 GB Sandisk USB drive, we load our custom 37 GB sccm ISO image on it with Rufus.    We then use the USB drive to load Windows 10 by selecting a USB UEFI boot option on our Dell computer.  It kicks off our SCCM task sequence.  With Rufus 3.8.1.580, our custom Windows 10 image successfully installs.   However, when we use Rufus 3.15.1812, it always bombs out on the "Apply Operating System (wim)" task (SCCM task sequence) with the the error 8007000B.   The SCCM log says incorrect format.      I am not sure what has changed over time.

We use GPT, UEFI (non CSM), Large FAT32, 32 kilobytes Cluster size.

Sorry if this is the wrong place to submit this.   Feel free to delete if so.

Log
---
```
<FULL LOG>
```

pbatard, 2021-09-15:
Hmmm, as you can see [there have been a lot of changes between 3.8 and 3.15](https://github.com/pbatard/rufus/blob/master/ChangeLog.txt#L1-L83), so it may help if you could identify the one version where the issues started to happen.

On the top of my head, I genuinely have no idea what change may have caused what you report. Especially, considering that we're very careful to try not to introduce alterations from the content of the ISO, and UEFI boot is very straightforward (even with UEFI:NTFS, we're really booting the original Microsoft bootloader and letting is access the unmodified original content), I don't really see how something like what you describe could be caused by something that Rufus changed, unless Dell computers have some weird issues with partition alignment when applying WIM content, as there isn't much else that I see could have an impact.

I suppose you validated that the SHA from the WIM file that was extracted is the same as the one from the ISO, just to make confirm that this isn't due to data corruption. Outside of that, if you can pinpoint the version that appears to have introduce the behaviour, it would certainly help with trying to guess what's happening...

mangomatt2, 2021-09-16:
Hello Peter,

Thank you for the reply.   I will test newer versions asap and double check the SHA from the WIM file when it fails.

mangomatt2, 2021-09-16:
Hello Peter,

I started with these:

rufus 3.12.1710 - errors after format is done and it tries to apply the wim/swm files
rufus 3.11.1678 - errors after format is done and it tries to apply the wim/swm files
rufus 3.10.1647 - errors after format is done and it tries to apply the wim/swm files
rufus 3.9.1624 - no issue.    Works just like Rufus 3.8.1580 did.   So seems to be starting with 3.10.1647.

I tested the hashes.   For the ones that do not work, 1 of the 3 swm files (split wim file) had an incorrect hash.    I simply copied the swm file from the ISO to the USB to replace the bad file and tried again.  The image then applied.

My co-worker also did the same result with Rufus 3.15.1812(portable).   She got the error applying the OS.   We checked the hash and 1 of the 3 files had a incorrect ash.   We replaced the file as I did and it then worked.

pbatard, 2021-09-17:
> 1 of the 3 swm files (split wim file) had an incorrect hash

Was it always the same file? And can you check precisely where the first difference occurs in that file (by binary comparing the problematic file with the original one)?

I would also really like you to run a comprehensive bad blocks check on your USB drives, because there's genuinely no change that I can think of that was introduced in 3.10 that could even remotely have an impact on ISO file extraction.

We did not update the libcdio library (which is what we use to read the data) and we certainly did not change how we copy the data (which we do by simply invoking the standard `WriteFile()` API).

The only thing that changed in our ISO extraction code between 3.9 and 3.10, really, is that we added detection for an extra error code ([here](https://github.com/pbatard/rufus/commit/b19f47f9b81bdf7ed079538d103ddacb4742ce35)), which means that, unless you get an error in Rufus during ISO file extraction, which you did not, the code being used by Rufus 3.9 and the code being used by Rufus 3.10 to extract ISO content is exactly the same.

Which is why the only explanation I can have for your issue is either coincidental/environmental or, if not, a compiler issue (as it's always possible that a newer version of a compiler introduces changes in behaviour that impact data handling or that).

Now, one thing that I'd also like clarification about (since you did not provide a log, which is __ALWAYS__ problematic &mdash; please don't assume that the providing a log does not apply to you) is that I'm still not sure how your process works.

I'm going to assume, from the fact that you use large FAT32, that you have a < 4GB `.wim` and a set of <4 GB `.swm` file, that this is how your ISO is structured and that you are using the regular Windows installer from a `boot.wim` to install Windows.

However you're talking about a 37 GB ISO, but only 3 `.swm` files, which, since they all must be < 4 GB leaves us far behind 37 GB (1 x 4 GB max `.wim` + 3 x 4 GB max `.swm` = 16 GB max of data... where are the remaining 21 GB?).

I am also not familiar with SCCM, which I would expect, when used in a corporate environment, would rely on fetching the image data from the network rather than have to go through a lengthy process of large file extraction to slow and unreliable USB media, so your process still has a lot of question marks here.

Can you please at least provide a log (from Rufus 3.9 and 3.10, so that we can compare while we're at it) so that I get a better idea of what goes into your media exactly?

mangomatt2, 2021-09-17:
Hello Peter,

I'll work on getting the log for you.   

There was one occasion that the hash was wrong on the first swm file and fine on the next two.    My co-worker had a wrong hash on the 2nd swm file but not the first or third.   

The other 21 GB is other portions of the task sequence.  Most of it is drivers.   SCCM goes through a series of tasks.  Once the operating system is applied, we have a WMI query that will apply a drivers package if it matches the computer model.    This is a standard SCCM practice.

We use USB drives over network because we have over a hundred sites.   Not all of our sites have a distribution point or pxe server.   It would be much too slow to go over the WAN.   This isn't an uncommon practice for many corporate environments.

It is very possible it something coincidental/environmental.    It's just odd that the older version never has the issue (thus far from our testing).   

Thank you for your time on this.   Luckily we have an easy resolution by just simply manually copying the swm files over after Rufus makes the USB drive.

mangomatt2, 2021-09-17:
[rufus_09172021_0907ET.log](https://github.com/pbatard/rufus/files/7185862/rufus_09172021_0907ET.log)

Log attached.

Latest test with Rufus 3.15.1812 = first and third swm file hashes are wrong.

I'll check if any bad blocks with Rufus

I checked a couple other large files on the USB drive and those hashes are ok.

================ USB DRIVE ==========================
PS D:\SMS\PKG\CS100764> get-filehash B5B30520E8DA7CFCD080F9ABD6BF7CD97B33768E2952C4DB7CBAD725D8359282000.swm

Algorithm       Hash                                                                   Path
---------       ----                                                                   ----
SHA256          9F43F03693482D47E06BD2E315DDA42DD14B2D550D5C40B8635F2EB860905DDD       D:\SMS\PKG\CS100764\B5B30520E8D…

PS D:\SMS\PKG\CS100764> get-filehash B5B30520E8DA7CFCD080F9ABD6BF7CD97B33768E2952C4DB7CBAD725D8359282001.swm

Algorithm       Hash                                                                   Path
---------       ----                                                                   ----
SHA256          C426DD2F645616514B8A10EABB01BAC6477E3BED7606BC2B68ED339BA535DE84       D:\SMS\PKG\CS100764\B5B30520E8D…

PS D:\SMS\PKG\CS100764> get-filehash B5B30520E8DA7CFCD080F9ABD6BF7CD97B33768E2952C4DB7CBAD725D8359282002.swm

Algorithm       Hash                                                                   Path
---------       ----                                                                   ----
SHA256          DA0FC585366451D8AAEB4255373FA0616E157F31C126B6258D3FF7829CC3296A       D:\SMS\PKG\CS100764\B5B30520E8D…

================ ISO IMAGE ======================

PS I:\SMS\PKG\CS100764> get-filehash B5B30520E8DA7CFCD080F9ABD6BF7CD97B33768E2952C4DB7CBAD725D8359282000.swm

Algorithm       Hash                                                                   Path
---------       ----                                                                   ----
SHA256          8159B86B821CA43062C4D2B668D360E6EC9299F0FDF46FB5CF00ABA83C81C8F1       I:\SMS\PKG\CS100764\B5B30520E8D…

PS I:\SMS\PKG\CS100764> get-filehash B5B30520E8DA7CFCD080F9ABD6BF7CD97B33768E2952C4DB7CBAD725D8359282001.swm

Algorithm       Hash                                                                   Path
---------       ----                                                                   ----
SHA256          C426DD2F645616514B8A10EABB01BAC6477E3BED7606BC2B68ED339BA535DE84       I:\SMS\PKG\CS100764\B5B30520E8D…

PS I:\SMS\PKG\CS100764> get-filehash B5B30520E8DA7CFCD080F9ABD6BF7CD97B33768E2952C4DB7CBAD725D8359282002.swm

Algorithm       Hash                                                                   Path
---------       ----                                                                   ----
SHA256          46A56C5AF3E593802277296CCC3A3E5D1E2B6C3E07422D8A49D6CCB5BF139B2F       I:\SMS\PKG\CS100764\B5B30520E8D…

pbatard, 2021-09-17:
Thanks. Another thing that I'd really like to see would be repeated tests with Rufus 3.9, to make absolutely certain that there is a different of behaviour betwen 3.9 and 3.10. A couple of tests with 3.9 are not enough to conclusively rule out that this is an external issue that randomly affects all versions, and that you simply happen not to have picked up with 3.9 yet.

Obviously, if there is any chance that Rufus may introduce data corruption, I'd like to get to the end of it. But I also want to make sure that I'm not wasting my time with an _"Oh yeah, it was our environment randomly adding corruption errors, and we simply had not seen in with older versions of Rufus yet_"...

mangomatt2, 2021-09-17:
Hello Peter,

Will do.  I'll do repeated tests this weekend.  I'll use my personal laptop as well instead of my company issued laptop.

Ran the usb drive bad block test.    Here is that portion of the log.

No bad blocks found.
Format operation started
Requesting disk access...
Will use 'D:' as volume mountpoint
Deleting partitions:
● Partition 1 (offset: 1048576, size: 57.3 GB)
Opened \\.\PhysicalDrive1 for exclusive write access
Analyzing existing boot records...
Drive has a Zeroed Master Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 2176 sectors
Initializing disk...
Bad Blocks: Checking from block 0 to 117359 (1 block = 512 KB)
Bad Blocks: Using offset 109659 for fake device check
Bad Blocks: Writing test pattern 0x00
Bad Blocks: Reading and comparing
Bad Blocks: Writing test pattern 0xFF
Bad Blocks: Reading and comparing
Bad Blocks: Writing test pattern 0x55
Bad Blocks: Reading and comparing
Bad Blocks: Writing test pattern 0xAA
Bad Blocks: Reading and comparing
Bad Blocks: Check completed, 0 bad blocks found. (0/0/0 errors)

pbatard, 2021-09-17:
Thanks. The bad blocks check makes it pretty clear that the USB Flash drive is unlikely to be the cause of the issue here.

mangomatt2, 2021-09-17:
So I've created the USB drive twice now with 3.15.1812 (Portable) and all three hashes were good both times.    Shrug.   

That proves that 3.15.1812 is capable of making a good drive.

pbatard, 2021-09-17:
Hmmm, I almost wish that you were always getting a failure with 3.15, because it would of course make troubleshooting _slightly_ easier, on account of replicability.

When I have a chance (it'll probably be some time before I do that though) I'll pick one of the official Windows 10 ISOs that has an `install.wim` that's a shade under 4 GB, and master a new ISO with duplicated version of that file until I get a >36 GB image that I can use to test for corruption errors. Regardless of the outcome, it'll be a good validation of how Rufus behaves when a large amount of data being copied over, which can't hurt as an application test anyway...

mangomatt2, 2021-09-17:
Thank you, Peter.  I'll continue testing as well and report back if I have any new info.

It would be cool if there was a way for Rufus to check hashes on wim and swm files and recopy if they don't match.

pbatard, 2022-04-15:
I have just tested a custom 39GB Windows 10 image where I have 10 duplicates of `install.wim` and saw no issues with the file extraction (using the same Large FAT32, 32 KB cluster size, UEFI (non CSM) as you did):
```
/mnt/iso/sources> sha256sum install.*
58e13f63bf784ae63239bdf4df4c7153e100c6ab0591de634b018b3e2dd83ff7  install.wim
58e13f63bf784ae63239bdf4df4c7153e100c6ab0591de634b018b3e2dd83ff7  install.wm1
58e13f63bf784ae63239bdf4df4c7153e100c6ab0591de634b018b3e2dd83ff7  install.wm2
58e13f63bf784ae63239bdf4df4c7153e100c6ab0591de634b018b3e2dd83ff7  install.wm3
58e13f63bf784ae63239bdf4df4c7153e100c6ab0591de634b018b3e2dd83ff7  install.wm4
58e13f63bf784ae63239bdf4df4c7153e100c6ab0591de634b018b3e2dd83ff7  install.wm5
58e13f63bf784ae63239bdf4df4c7153e100c6ab0591de634b018b3e2dd83ff7  install.wm6
58e13f63bf784ae63239bdf4df4c7153e100c6ab0591de634b018b3e2dd83ff7  install.wm7
58e13f63bf784ae63239bdf4df4c7153e100c6ab0591de634b018b3e2dd83ff7  install.wm8
58e13f63bf784ae63239bdf4df4c7153e100c6ab0591de634b018b3e2dd83ff7  install.wm9
```
I've also gone through the commits that intervened between 3.9 and 3.10, and I really can't see any specific one that I can see affect the extraction of ISO data.

I therefore would think that there exists some kind of compounded issue linked to your environment that may be triggered by something external to the code, such as the manner in which the executable was produced by the compiler (because, we of course update the compiler used to produce Rufus along with the executable), or, since you are working in a corporate environment, a security solution that should only ever monitor data being transferred, but that ends up interfering with said data.

I will therefore close this issue and wait to see if anybody else, coming from a completely separate environment from yours, appears to report a similar problem, as there really isn't much I can do about this report (I may eventually add written data validation in Rufus, but it's not going to happen in any short or medium term).

One thing I would still suggest you try is see if [the Windows Store version of Rufus](https://www.microsoft.com/en-us/p/rufus/9pc3h3v7q9ch) also exhibits this issue, as you'll be running a 64-bit VS2022 compiled executable as opposed to the 32-bit MinGW compiled regular version, and, for a weird issue like this one, testing a different executable may actually yield some data that may help pinpoint the source of the issue.

For the time being however, I will close this issue (but you can still comment on it if you wish).

closed 2022-04-15 by pbatard