pbatard/rufus #1776: Rufus(GPT)

Emanuele-1998, 2021-09-17:
Why if create the usb in GPT, the process format is in MBR?

[image]

pbatard, 2021-09-17:
Did not follow the check list. I can't help you because I don't have enough data.

closed 2021-09-17 by pbatard

Emanuele-1998, 2021-09-17:
Sorry for mistake
[rufus 3.15.1812.log](https://github.com/pbatard/rufus/files/7186549/rufus.3.15.1812.log)

[image]

pbatard, 2021-09-17:
Here's where your confusion comes from.

As the log indicates, your partition scheme was properly created as GPT:
```
Partitioning (GPT)...
```
What appears to confuse you is that, as a partition scheme, GPT may also contain a **protective** Master Boot Record, that is used for systems that cannot understand GPT (e.g. old BIOS based computers) so that they don't flag the disk as unpartitioned or empty.

And one can also use the MBR to display a message, like Rufus does, to let users know that they booted a UEFI-only media in BIOS mode:
[image]

So Rufus does actually write an MBR when creating a 100% GPT compliant drive. And you will see a message indicating this. But that does not mean that your drive is partitioned as MBR (even though it technically is, because a GPT drive with a protective MBR is a dual MBR **and** GPT disk, but this is usually not explicitly stated, precisely so that users don't get confused).

In short, do not assume that `Writing MBR (Master Boot Record)` is the same as `Partitioning drive as MBR` or `Partitioning as something that isn't GPT`. Rufus is simply writing data into the MBR that also exists on a GPT drive, and, for the sake of transparency, telling you precisely what it is doing, even if, if you aren't aware about the fact that one can write an MBR onto a GPT drive, this can be confusing.

Hope that clarifies it.

Emanuele-1998, 2021-09-17:
> Here's where your confusion comes from.
> 
> As the log indicates, your partition scheme was properly created as GPT:
> 
> ```
> Partitioning (GPT)...
> ```
> 
> What appears to confuse you is that, as a partition scheme, GPT may also contain a **protective** Master Boot Record, that is used for systems that cannot understand GPT (e.g. old BIOS based computers) so that they don't flag the disk as unpartitioned or empty.
> 
> And one can also use the MBR to display a message, like Rufus does, to let users know that they booted a UEFI-only media in BIOS mode:
> [image]
> 
> So Rufus does actually write an MBR when creating a 100% GPT compliant drive. And you will see a message indicating this:
> 
> ```
> Writing Master Boot Record...
> Using Rufus protective MBR
> ```
> 
> But that does not mean that your drive is partitioned as MBR (even though it technically is, because a GPT drive with a protective MBR is a dual MBR **and** GPT disk, but this is usually not explicitly stated, precisely so that users don't get confused).
> 
> In short, do not assume that `Writing MBR (Master Boot Record)` is the same as `Partitioning drive as MBR` or `Partitioning as something that isn't GPT`. Rufus is simply writing data into the MBR that also exists on a GPT drive, and, for the sake of transparency, telling you precisely what it is doing, even if, if you aren't aware about the fact that one can write an MBR onto a GPT drive, this can be confusing.
> 
> Hope that clarifies it.

Thanks for explanation