pbatard/rufus #1777: Rocky Linux 8.4 does not work in ISO mode

marcosfrm, 2021-09-17:
[Rocky](https://rockylinux.org/) is a new RHEL 8 clone after the CentOS Stream drama (aka premature CentOS 8 EOL).

We are using it to replace CentOS 8 in our machines, but unfortunately Rocky also drinks the ISOHybrid kool-aid:

[image]

[image]

If I select "ISO file" it still says "Error setting up base repository".

It would be nice add it to the ISOHybrid kool-aid drinkers list (and probably [AlmaLinux](https://almalinux.org/) too). Anaconda should not be this dumb...

```
Rufus x86 v3.15.1812
Windows version: Windows 10 64-bit (Build 19043.1237)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.06
System locale ID: 0x0416 (pt-BR)
Will use default UI locale 0x0416
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'pt-BR'
NOTE: The Portuguese Brazilian (Português do Brasil) translation requires an update, but the current translator hasn't submitted one. Because of this, some messages will only be displayed in English.
If you think you can help update this translation, please e-mail the author of this application
Found USB 2.0 device 'Patriot Memory USB Device' (13FE:3600)
1 device found
No volume information for drive 0x81
Disk type: Removable, Disk size: 4GB, Sector size: 512 bytes
Cylinders: 486, Tracks per cylinder: 255, Sectors per track: 63
Partition type: SFD (Super Floppy Disk) or unpartitioned
Scanning image...
ISO analysis:
  Image is an ISO9660 image
  Will use '/isolinux/isolinux.cfg' for Syslinux
  Detected Syslinux version: 6.04 (from '/isolinux/isolinux.bin')
Disk image analysis:
  Image has an unknown Master Boot Record
  Image is a bootable disk image
ISO label: 'Rocky-8-4-x86_64-dvd'
  Size: 1.9 GB (Projected)
  Note: File on disk is larger than reported ISO size by 416 KB...
  Has a >64 chars filename
  Uses: Syslinux/Isolinux v6.04
  Uses: EFI
Using image: Rocky-8.4-x86_64-minimal.iso (1.8 GB)
Checking for Rufus updates...
Checking release channel...
No new release version found.
Downloading https://rufus.ie/files/syslinux-6.04/ldlinux.sys
File length: 66.5 KB
Successfully downloaded 'ldlinux.sys'
Download signature is valid ✓
Downloading https://rufus.ie/files/syslinux-6.04/ldlinux.bss
File length: 512 bytes
Successfully downloaded 'ldlinux.bss'
Download signature is valid ✓

Format operation started
Requesting disk access...
Will use 'E:' as volume mountpoint
Deleting partitions:
● Partition 1 (offset: 0, size: 3.7 GB)
Opened \\.\PhysicalDrive1 for exclusive write access
Analyzing existing boot records...
Drive has a Zeroed Master Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Initializing disk...
Partitioning (MBR)...
● Creating Main Data Partition (offset: 1048576, size: 3.7 GB)
Waiting for logical drive to reappear...
[... 1781 lines cut ...]
Extracting: E:\isolinux\libcom32.c32 (176.4 KB)
Extracting: E:\isolinux\libutil.c32 (22.3 KB)
Extracting: E:\isolinux\memtest (178.4 KB)
Extracting: E:\isolinux\splash.png (186 bytes)
Extracting: E:\isolinux\vesamenu.c32 (26.2 KB)
Extracting: E:\isolinux\vmlinuz (9.6 MB)
Extracting: E:\media.repo (86 bytes)
Created: E:\syslinux.cfg → /isolinux/isolinux.cfg
Finalizing, please wait...
Created: E:autorun.inf
Created: E:autorun.ico

Found USB 2.0 device 'Patriot Memory USB Device' (13FE:3600)
Using autorun.inf label for drive E: 'Rocky-8-4-x86_64-dvd'
1 device found
Disk type: Removable, Disk size: 4GB, Sector size: 512 bytes
Cylinders: 486, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x00323A37
Drive has a Syslinux Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 3.7 GB (3998220288 bytes)
  Start Sector: 2048, Boot: Yes
```

pbatard, 2021-09-17:
No surprise there. This is a consequence of a regression from Red Hat in Anaconda, that ultimately [forced us to add exceptions to enforce DD mode for distros that are based on RH 8.2 or later](https://github.com/pbatard/rufus/commit/9c8fa409959a4713f1a9c85043962e6cb4cc465b).

I tried to raise the issue in the CentOS forums, since they were affected, but they aggressively dismissed it as _"This is not a Red Hat issue, this is a Rufus issue"_, without realizing that this needed to be reported upstream to avoid more Red Hat derivatives to be affected... as we are seeing now.

To be honest, at this stage, I am starting to get __REALLY__ fed up of all the distros who equate ISOHybrid with DD only, and appear to have no clue that, there are plenty of people out there who don't want to use a third party application like Rufus, Etcher or even dd to create bootable media for booting a UEFI system when (provided that there doesn't exist a file large than 4 GB on the ISO, which I'm not aware any prominent Linux distro to have) they should just be able to create a FAT32 partition and extract all the content there for it too boot. This, in essence, is ISO mode. And even if it may seem counter intuitive, there are clear advantages to using ISO mode for Windows users, as [highlighted here](https://github.com/pbatard/rufus/wiki/FAQ#Why_doesnt_Rufus_recommend_DD_mode_over_ISO_mode_for_ISOHybrid_images_Surely_DD_is_better).

Debian understand this.
Arch and Ubuntu understand this.
Other popular distro understand this.
Some of them make sure they do test this method of creating bootable media on UEFI system to ensure that it works.
And yeah, it does mean extra work from distro maintainers, but when the idea is that it should improve the overall user experience and provide users with a choice of how they can install the OS, it should be a very easy decision.

Unfortunately, Red Hat and its derivative, as well as some other distros like Manjaro, appear to have taken the approach that they should minimize the amount of work required from their maintainers, so they see ISO mode (which should really be called _"UEFI standard file copy boot"_ as something that they should not have to bother with, and will push back **very strongly** at the idea that anything but DD copy should be used for media creation even if, again, the end result is that users who wish to install their distros will be, at the very least, inconvenienced (due to having less methods they can use to create boot media) and at worst, as I could point to many, may reddit threads where that happens, under the impression that the media they created in DD mode must be faulty when they can only see a small ESP out of it in Windows, and therefore, not attempt to boot the distro at all!

Now, if you can give me the ISO label scheme that you plan to use for your images, I can try to add another regexp to set it as `DD-only` in the next version Rufus (which, again, is less than ideal because it __DOES__ lead to Windows users ditching the distro altogether).

Or, as you should do to prevent yet another instance of a RH derivative needing to figure out why the media creation mode recommended by Rufus does not work, you should liaise with Red Hat and try to make them understand that:
1. No, ISOHybrid does not equate DD
2. They __REALLY__ should taste the media creation path that consist of extracting the ISO content to a FAT32 partition and booting on a UEFI system (which again, is __not__ Rufus specific and which any user on Linux, Mac, Windows and so on can usually accomplish without the need to install a specific tool or even to use command line).
3. They __ARE__ going to lose users if they continue to brush this issue as something that they don't need to concern themselves with, or something that it's up to third party utilities like Rufus to take care of.

pbatard, 2021-09-17:
Heck, right at the time I was writing this post, there was [another reddit user posting about their confusion of only seeing an ESP from a Manjaro ISOHybrid that was written in DD mode](https://old.reddit.com/r/linuxquestions/comments/pq23xe/im_trying_to_do_a_bootable_manjaro_kde_plasma_usb/).

So, yeah, this is a real issue, regardless of whether Red Hat and its derivatives want to consider it as something they need to address.

marcosfrm, 2021-09-17:
From

https://github.com/rhinstaller/anaconda/pull/3529

I **replaced** `inst.stage2=` with `inst.repo=` in the GRUB menu and it worked. Installed just fine in ISO mode (no internet available). It should work with the full DVD too, which also has packages in the install media. Not sure if it will work with network install media.

[image]

Docs:

https://github.com/rhinstaller/anaconda/blob/master/docs/boot-options.rst#installation-source

pbatard, 2021-09-17:
Thanks for the update. As you will see from my post in rhinstaller/anaconda#3529, I still see the issue as something that Red Hat should look into, if they do care about their users especially as it's a bit worrying that one of the most popular Linux distros out there seems to not be aware of one common means of creating a Linux installation media for UEFI systems... There has got to be a better way than force people to edit their config files when creating media at the file system level...

closed 2021-10-03 by pbatard

pbatard, 2021-10-03:
The newly released Rufus 3.16 BETA, which can be downloaded [here](https://github.com/pbatard/rufus/releases) has the `inst.stage2` &rarr; `inst.repo` workaround. Please test it and let me know if you are still seeing the issue.

marcosfrm, 2021-10-03:
Rocky 8.4 is ok!

But the change broke CentOS 7:

[image]

("ISO file" does not work)

If I switch back to `inst.stage2` it works. Hence the workaround is only for 8.x versions I think (and Fedora?).

CentOS 7 log:

```
Rufus x86 v3.16.1829 (Beta)
Windows version: Windows 10 Pro, 64-bit (Build 19043.1237)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.06
System locale ID: 0x0416 (pt-BR)
Will use default UI locale 0x0416
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'pt-BR'
NOTE: The Portuguese Brazilian (Português do Brasil) translation requires an update, but the current translator hasn't submitted one. Because of this, some messages will only be displayed in English.
If you think you can help update this translation, please e-mail the author of this application
Notice: Windows VDS is unavailable
Found USB 2.0 device 'Patriot Memory USB Device' (13FE:3600)
Using autorun.inf label for drive E: 'Rocky-8-4-x86_64-dvd'
1 device found
Disk type: Removable, Disk size: 4GB, Sector size: 512 bytes
Cylinders: 486, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x0014E83B
Drive has a Syslinux Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 3.7 GB (3998220288 bytes)
  Start Sector: 2048, Boot: Yes
Scanning image...
ISO analysis:
  Image is an ISO9660 image
  Will use '/isolinux/isolinux.cfg' for Syslinux
  Detected Syslinux version: 4.05/0x5bd8f633 (from '/isolinux/isolinux.bin')
Disk image analysis:
  Image has an unknown Master Boot Record
  Image is a bootable disk image
ISO label: 'CentOS 7 x86_64'
  Size: 1 GB (Projected)
  Note: File on disk is larger than reported ISO size by 314 KB...
  Has a >64 chars filename
  Uses: Syslinux/Isolinux v4.05
  Uses: EFI
Using image: CentOS-7-x86_64-Minimal-2009.iso (973 MB)
Found USB 2.0 device 'Patriot Memory USB Device' (13FE:3600)
Using autorun.inf label for drive E: 'Rocky-8-4-x86_64-dvd'
1 device found
Disk type: Removable, Disk size: 4GB, Sector size: 512 bytes
Cylinders: 486, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x0014E83B
Drive has a Syslinux Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 3.7 GB (3998220288 bytes)
  Start Sector: 2048, Boot: Yes

Format operation started
Requesting disk access...
Will use 'E:' as volume mountpoint
Warning: Could not delete partition(s): VDS is not available
Opened \\.\PhysicalDrive1 for shared write access
Requesting logical volume handle...
[... 523 lines cut ...]
Extracting: E:\repodata\repomd.xml.asc (811 bytes)
Extracting: E:\repodata\TRANS.TBL (3.3 KB)
Extracting: E:\RPM-GPG-KEY-CentOS-7 (1.7 KB)
Extracting: E:\RPM-GPG-KEY-CentOS-Testing-7 (1.7 KB)
Extracting: E:\TRANS.TBL (2.8 KB)
Extracting: E:\.discinfo (29 bytes)
Extracting: E:\.treeinfo (354 bytes)
Created: E:\syslinux.cfg → /isolinux/isolinux.cfg
Finalizing, please wait...
Created: E:autorun.inf
Created: E:autorun.ico

Found USB 2.0 device 'Patriot Memory USB Device' (13FE:3600)
Using autorun.inf label for drive E: 'CentOS 7 x86_64'
1 device found
Disk type: Removable, Disk size: 4GB, Sector size: 512 bytes
Cylinders: 486, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x00020545
Drive has a Syslinux Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 3.7 GB (3998220288 bytes)
  Start Sector: 2048, Boot: Yes
```

pbatard, 2021-10-04:
I kind of feared that this would be the case, and that's the precise reason why the fix would be better if it came from Red Hat, because now I have to figure out distros that don't work with `inst.stage2`.

At this stage, I'm not sure I'm going to bother for this release, as I have more pressing matters to deal with, and I can always refer RH derivative users to older Rufus versions. Realistically, because it is effectively a regression, making sure that `inst.stage2` does work when extracting media to USB should be Red Hat's job, not Rufus'...

marcosfrm, 2021-10-09:
> 
> Once RHEL 8.x branch gets a fix, we can narrow down the regex replacing `.*` with `-[0-5].*` (fixed in RHEL 8.6 for example).
> 

`-[0-5]-.*` will be better, because RHEL often gets minor version above 9. Then 8-10 would incorrectly match.

commit c09ede16 (2021-10-03): [iso] add kernel options update for Red Hat derivatives

* Since version 8.2, and rhinstaller/anaconda@a7661019546ec1d8b0935f9cb0f151015f2e1d95,
  Red Hat derivatives have changed their CD-ROM detection policy which leads to
  the installation source not being found when writing the media in ISO mode.
* Replace 'inst.stage2' by 'inst.repo' in the kernel options.
* Closes #1777 (See also rhinstaller/anaconda#rhinstaller/anaconda#3529).
* Note that this reverts part of 9c8fa409959a4713f1a9c85043962e6cb4cc465b.