pbatard/rufus #1779: [0x00000057] Error While Formating - Only on Windows 11

xatornet, 2021-09-18:
Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
I cannot format any disk using Rufus when using Windows 11. I've used it on the same system while using Windows 10 without any problem. Also, I can manually format the device using diskpart and Windows without any problem. I've tried:

-Using several USB devices, none of them work while using Win 11.
-Using every single USB port, both from the PC Case and motherboard. None won't allow formating using Rufus on Win 11.
-Using Rufus on Win 10 with the same and diferent PC works properly, using the same devices to format.

Here's the capture of the error, and below there's the log file:

[image]
 

Log
---
Rufus x64 v3.15.1812 (AppStore version)
Windows version: Windows 10 64-bit (Build 22458.1000)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.06
System locale ID: 0x0C0A (es-ES)
Will use default UI locale 0x0C0A
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'es-ES'
Found UAS (USB 3.0) device 'KINGSTON  SM2280S3G2120G SCSI Disk Device' (174C:235C)
1 device found
Disk type: FIXED, Disk size: 120 GB, Sector size: 512 bytes
Cylinders: 14593, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 2
Disk GUID: {9DA39CD2-FEA2-42D5-ADC6-B50859B684CF}
Max parts: 128, Start Offset: 17408, Usable = 120034089472 bytes
Partition 1:
  Type: Microsoft System Reserved Partition
  Name: 'Microsoft reserved partition'
  ID: {16C67BD7-B5F3-41BB-908F-2D8B5492CBFA}
  Size: 16.0 MB (16759808 bytes)
  Start Sector: 34, Attributes: 0x0000000000000000
Partition 2:
  Type: Microsoft Basic Data Partition
  Name: 'Basic data partition'
  ID: {4DCB3B86-3F83-4E7C-BFC5-B74B7314B62D}
  Size: 111.8 GB (120015814656 bytes)
  Start Sector: 32768, Attributes: 0x0000000000000000

Format operation started
Requesting disk access...
Will use 'X:' as volume mountpoint
Could not load VDS Service: [0x80070005] Access is denied.
Opened \\.\PhysicalDrive4 for exclusive write access
Requesting logical volume handle...
I/O boundary checks disabled
Analyzing existing boot records...
Drive has a Windows 7 Master Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 34 sectors
Initializing disk...
Partitioning (MBR)...
● Creating Main Data Partition (offset: 1048576, size: 111.8 GB)
Closing existing volume...
Waiting for logical drive to reappear...
Formatting (NTFS)...
Using cluster size: 4096 bytes
Quick format was selected
Creating file system...
Error while formatting
[0x00000057] The parameter is incorrect. - Retrying...
[0x00000057] The parameter is incorrect. - Retrying...
[0x00000057] The parameter is incorrect. - Retrying...
[0x00000057] The parameter is incorrect. - Retrying...
Format error: Undetermined error while formatting.

Found UAS (USB 3.0) device 'KINGSTON  SM2280S3G2120G SCSI Disk Device' (174C:235C)
1 device found
No volume information for drive 0x84
Disk type: FIXED, Disk size: 120 GB, Sector size: 512 bytes
Cylinders: 14593, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x00BE1408
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: NTFS/exFAT/UDF (0x07)
  Size: 111.8 GB (120033075200 bytes)
  Start Sector: 2048, Boot: No

pbatard, 2021-09-18:
Thanks for the report.

> `Could not load VDS Service: [0x80070005] Access is denied.`

This is a problem with the Windows Store version of Rufus running on Windows 11, and the source of the issue.

It looks like, starting with Windows 11, Microsoft rescinded some permissions that they used to provide Windows Store apps, including the ability to access the VDS service or the ability to call on the Windows formatting APIs.

In other words, Windows 11 restricts the ability of store installed applications like Rufus to perform system operations like formatting a drive.

If you are running Windows 11, you should use the non-Store version of Rufus, which you can download from the [official site](https://rufus.ie) as you will find that it works much better (since Microsoft has not yet applied the same restrictions to non-store applications):

```
Rufus x86 v3.15.1812
Windows version: Windows 10 64-bit (Build 22454.1000)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.06
System locale ID: 0x0809 (en-GB)
NOTE: No Windows Language Pack is installed for Spanish (Español) on this system.
This means that some controls may still be displayed using the system locale.
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'es-ES'
Found UAS (USB 3.0) device 'Mushkin Ventura Ultra SCSI Disk Device' (174C:55AA)
Removing C: from the list: This is the system disk!
Found USB 3.0 device 'SanDisk Extreme USB Device' (0781:558B)
Using autorun.inf label for drive D: '64 GB'
1 device found
Disk type: Removable, Disk size: 64 GB, Sector size: 512 bytes
Cylinders: 7628, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x00000000
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 58.4 GB (62745739264 bytes)
  Start Sector: 2048, Boot: No

Format operation started
Requesting disk access...
Will use 'D:' as volume mountpoint
Deleting partitions:
● Partition 1 (offset: 1048576, size: 58.4 GB)
Opened \\.\PhysicalDrive0 for exclusive write access
Analyzing existing boot records...
Drive has a Zeroed Master Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 2176 sectors
Initializing disk...
Partitioning (MBR)...
● Creating Main Data Partition (offset: 1048576, size: 58.4 GB)
Waiting for logical drive to reappear...
Formatting (Large FAT32)...
Opened \\?\Volume{d21cda0a-1889-11ec-b22f-1c697a0fd916} for exclusive write access
Size : 58.4 GB 122550272 sectors
Cluster size 32768 bytes, 512 bytes per sector
Volume ID is 16f6:2b3c
32 Reserved sectors, 14957 sectors per FAT, 2 FATs
1914380 Total clusters
1914379 Free clusters
Clearing out 30010 sectors for reserved sectors, FATs and root cluster...
Initializing reserved sectors and FATs...
FAT #0 sector at address: 32
FAT #1 sector at address: 14989
Writing Partition Boot Record...
Using Standard FAT32 partition boot record
Confirmed new volume has a primary FAT32 boot sector
Setting primary FAT32 boot sector for boot...
Confirmed new volume has a secondary FAT32 boot sector
Setting secondary FAT32 boot sector for boot...
Setting label...
Format completed.
Writing Master Boot Record...
Partition is already FAT32 LBA...
Using Zeroed MBR
Found volume \\?\Volume{d21cda0a-1889-11ec-b22f-1c697a0fd916}\
Created: D:\autorun.inf
Created: D:\autorun.ico
Successfully remounted \\?\Volume{d21cda0a-1889-11ec-b22f-1c697a0fd916}\ as D:

Found UAS (USB 3.0) device 'Mushkin Ventura Ultra SCSI Disk Device' (174C:55AA)
Removing C: from the list: This is the system disk!
Found USB 3.0 device 'SanDisk Extreme USB Device' (0781:558B)
Using autorun.inf label for drive D: '64 GB'
1 device found
Disk type: Removable, Disk size: 64 GB, Sector size: 512 bytes
Cylinders: 7628, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x00000000
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 58.4 GB (62745739264 bytes)
  Start Sector: 2048, Boot: No
```

No wonder Microsoft calls Windows 11 more secure than Windows 10, when they have restricted things so much that advanced applications can no longer accomplish legitimate tasks...

I'll see if I can log a ticket with Microsoft's Windows Store support, but I get the feeling that they're just going to say that the new security restrictions are here to stay and that Windows Store apps should limit themselves to simply doing things in their little corner and not try to format a drive...

xatornet, 2021-09-18:
Hi there, 

Thanks for the quick response.  I've tried the standalone version, and although the error code switches to 0x00000005, it's not working either.

I've read the FAQ, and I have Windows Defender fully disabled from regedit, so there's no Manage Controlled folder access running. Here's the capture and the log:

[image]

Rufus x86 v3.15.1812
Windows version: Windows 10 64-bit (Build 22458.1000)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.06
System locale ID: 0x0C0A (es-ES)
Will use default UI locale 0x0C0A
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'es-ES'
Found UAS (USB 3.0) device 'KINGSTON  SM2280S3G2120G SCSI Disk Device' (174C:235C)
1 device found
Disk type: FIXED, Disk size: 120 GB, Sector size: 512 bytes
Cylinders: 14593, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 2
Disk GUID: {45F68214-A5E4-4431-B422-3D9DF80C3CEF}
Max parts: 128, Start Offset: 17408, Usable = 120034089472 bytes
Partition 1:
  Type: Microsoft System Reserved Partition
  Name: 'Microsoft reserved partition'
  ID: {B4FA17A4-48C4-4B02-AA5A-158AD745DD8F}
  Size: 16.0 MB (16759808 bytes)
  Start Sector: 34, Attributes: 0x0000000000000000
Partition 2:
  Type: Microsoft Basic Data Partition
  Name: 'Basic data partition'
  ID: {321D1A59-0884-4196-B5AB-9D9900316D32}
  Size: 111.8 GB (120015814656 bytes)
  Start Sector: 32768, Attributes: 0x0000000000000000
0 devices found
Found UAS (USB 3.0) device 'KINGSTON  SM2280S3G2120G SCSI Disk Device' (174C:235C)
No logical drive found (unpartitioned?)
1 device found
No volume information for drive 0x84
Disk type: FIXED, Disk size: 120 GB, Sector size: 512 bytes
Cylinders: 14593, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 2
Disk GUID: {45F68214-A5E4-4431-B422-3D9DF80C3CEF}
Max parts: 128, Start Offset: 17408, Usable = 120034089472 bytes
Partition 1:
  Type: Microsoft System Reserved Partition
  Name: 'Microsoft reserved partition'
  ID: {B4FA17A4-48C4-4B02-AA5A-158AD745DD8F}
  Size: 16.0 MB (16759808 bytes)
  Start Sector: 34, Attributes: 0x0000000000000000
Partition 2:
  Type: Microsoft Basic Data Partition
  Name: 'Basic data partition'
  ID: {321D1A59-0884-4196-B5AB-9D9900316D32}
  Size: 111.8 GB (120015814656 bytes)
  Start Sector: 32768, Attributes: 0x0000000000000000
Found UAS (USB 3.0) device 'KINGSTON  SM2280S3G2120G SCSI Disk Device' (174C:235C)
1 device found
Disk type: FIXED, Disk size: 120 GB, Sector size: 512 bytes
Cylinders: 14593, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 2
Disk GUID: {45F68214-A5E4-4431-B422-3D9DF80C3CEF}
Max parts: 128, Start Offset: 17408, Usable = 120034089472 bytes
Partition 1:
  Type: Microsoft System Reserved Partition
  Name: 'Microsoft reserved partition'
  ID: {B4FA17A4-48C4-4B02-AA5A-158AD745DD8F}
  Size: 16.0 MB (16759808 bytes)
  Start Sector: 34, Attributes: 0x0000000000000000
Partition 2:
  Type: Microsoft Basic Data Partition
  Name: 'Basic data partition'
  ID: {321D1A59-0884-4196-B5AB-9D9900316D32}
  Size: 111.8 GB (120015814656 bytes)
  Start Sector: 32768, Attributes: 0x0000000000000000
Found UAS (USB 3.0) device 'KINGSTON  SM2280S3G2120G SCSI Disk Device' (174C:235C)
1 device found
Disk type: FIXED, Disk size: 120 GB, Sector size: 512 bytes
Cylinders: 14593, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 2
Disk GUID: {45F68214-A5E4-4431-B422-3D9DF80C3CEF}
Max parts: 128, Start Offset: 17408, Usable = 120034089472 bytes
Partition 1:
  Type: Microsoft System Reserved Partition
  Name: 'Microsoft reserved partition'
  ID: {B4FA17A4-48C4-4B02-AA5A-158AD745DD8F}
  Size: 16.0 MB (16759808 bytes)
  Start Sector: 34, Attributes: 0x0000000000000000
Partition 2:
  Type: Microsoft Basic Data Partition
  Name: 'Basic data partition'
  ID: {321D1A59-0884-4196-B5AB-9D9900316D32}
  Size: 111.8 GB (120015814656 bytes)
  Start Sector: 32768, Attributes: 0x0000000000000000

Format operation started
Requesting disk access...
Will use 'X:' as volume mountpoint
Deleting partitions:
● Partition 1 (offset: 17408, size: 16.0 MB)
● Partition 2 (offset: 16777216, size: 111.8 GB)
Opened \\.\PhysicalDrive4 for exclusive write access
Analyzing existing boot records...
Drive has a Zeroed Master Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Initializing disk...
Partitioning (GPT)...
● Creating Main Data Partition (offset: 1048576, size: 111.8 GB)
Waiting for logical drive to reappear...
Formatting (NTFS)...
Using cluster size: 4096 bytes
Quick format was selected
Creating file system...
Error while formatting
[0x00000005] Access is denied. - Retrying...
[0x00000005] Access is denied. - Retrying...
[0x00000005] Access is denied. - Retrying...
[0x00000005] Access is denied. - Retrying...
Format error: Undetermined error while formatting.

Found UAS (USB 3.0) device 'KINGSTON  SM2280S3G2120G SCSI Disk Device' (174C:235C)
1 device found
No volume information for drive 0x84
Disk type: FIXED, Disk size: 120 GB, Sector size: 512 bytes
Cylinders: 14593, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 1
Disk GUID: {980267C9-F2F3-473B-B40D-DC88A0019CA6}
Max parts: 128, Start Offset: 17408, Usable = 120034089472 bytes
Partition 1:
  Type: Microsoft Basic Data Partition
  Name: 'Main Data Partition'
  ID: {B6862E8D-F931-46F1-828C-C13878516C96}
  Size: 111.8 GB (120033058304 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
Saved C:\Users\Alex\Desktop\rufus.log

xatornet, 2021-09-18:
UPDATE:

I've read the FAQ further and, it seems that cheat mode "ALT+ , " fixes the problem, here's the log of a succesfull formatting.

Rufus x86 v3.15.1812
Windows version: Windows 10 64-bit (Build 22458.1000)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.06
System locale ID: 0x0C0A (es-ES)
Will use default UI locale 0x0C0A
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'es-ES'
Found UAS (USB 3.0) device 'KINGSTON  SM2280S3G2120G SCSI Disk Device' (174C:235C)
1 device found
Disk type: FIXED, Disk size: 120 GB, Sector size: 512 bytes
Cylinders: 14593, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 1
Disk GUID: {8E1F4929-F644-422A-98D2-D44F62615534}
Max parts: 128, Start Offset: 17408, Usable = 120034089472 bytes
Partition 1:
  Type: Microsoft Basic Data Partition
  Name: 'Main Data Partition'
  ID: {68FE6874-CB5C-4FA7-BEB8-21605D26DCAF}
  Size: 111.8 GB (120033058304 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000

Format operation started
Requesting disk access...
Will use 'G:' as volume mountpoint
Deleting partitions:
● Partition 1 (offset: 1048576, size: 111.8 GB)
Opened \\.\PhysicalDrive4 for shared write access
Analyzing existing boot records...
Drive has a Zeroed Master Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 2176 sectors
Initializing disk...
Partitioning (GPT)...
● Creating Main Data Partition (offset: 1048576, size: 111.8 GB)
Waiting for logical drive to reappear...
Formatting (Large FAT32)...
Opened \\?\Volume{c344bd3b-0457-4299-8cde-6ec6b157b27f} for exclusive write access
Size : 111.8 GB 234439567 sectors
Cluster size 32768 bytes, 512 bytes per sector
Volume ID is 191a:1e57
32 Reserved sectors, 28612 sectors per FAT, 2 FATs
3662223 Total clusters
3662222 Free clusters
Clearing out 57320 sectors for reserved sectors, FATs and root cluster...
Initializing reserved sectors and FATs...
FAT #0 sector at address: 32
FAT #1 sector at address: 28644
Writing Partition Boot Record...
Using Standard FAT32 partition boot record
Confirmed new volume has a primary FAT32 boot sector
Setting primary FAT32 boot sector for boot...
Confirmed new volume has a secondary FAT32 boot sector
Setting secondary FAT32 boot sector for boot...
Setting label...
Format completed.
Found volume \\?\Volume{c344bd3b-0457-4299-8cde-6ec6b157b27f}\
Successfully remounted \\?\Volume{c344bd3b-0457-4299-8cde-6ec6b157b27f}\ as G:

Found UAS (USB 3.0) device 'KINGSTON  SM2280S3G2120G SCSI Disk Device' (174C:235C)
1 device found
Disk type: FIXED, Disk size: 120 GB, Sector size: 512 bytes
Cylinders: 14593, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 1
Disk GUID: {B3CC44B1-20D3-4B36-9142-232CD256A5AA}
Max parts: 128, Start Offset: 17408, Usable = 120034089472 bytes
Partition 1:
  Type: Microsoft Basic Data Partition
  Name: 'Main Data Partition'
  ID: {C344BD3B-0457-4299-8CDE-6EC6B157B27F}
  Size: 111.8 GB (120033058304 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000

pbatard, 2021-09-18:
Interestingly, I am no longer able to replicate the issue on my test Windows 11 machine, using the Windows Store version (be it with or without exclusive drive locking, i.e. <kbd>Alt</kbd>-<kbd>,</kbd> enabled or disabled).

The only thing that happened since my last test is that I installed the 22454 &rarr; to 22458 build update and rebooted. But, considering that you were already running 22458, we can rule out the update for this change of behaviour.

Did you reboot your machine since you installed the Rufus Windows Store app?

I have not seen the `Access denied` issues on my test machine (which is a vanilla Windows 11 with Defender enabled), be it with the Windows Store or regular version, so it's difficult to comment on why you have been getting those errors. One thing I am seeing is that you were trying to format a drive that had an MSR in the test that failed, but not in the test that worked, and I know that Microsoft can be an ass about letting third party applications access Windows system partitions such as MSR's or ESP's (I've had to add workarounds in Rufus to try to work around that behaviour), so, if I was to have a suspect for `Access denied` it would be this (which disabling exclusive access can indeed sometimes help with).

I'm going to test some more, because it's very puzzling that, after update and reboot, the Windows Store version suddenly started to work for me whereas I did not change anything on the system. And of course, I'm going to play with drives that contain MSR's to see what happens...

xatornet, 2021-09-18:
Hi there,

I've rebooted to test this out, but I cannot format without using the Alt-, hotkey. In my case, it does not matter if I'm formatting from MBR to GPT or else. It won't work either way.  Here's log after rebooting (failed formatting):

Rufus x64 v3.15.1812 (AppStore version)
Windows version: Windows 10 64-bit (Build 22458.1000)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.06
System locale ID: 0x0C0A (es-ES)
Will use default UI locale 0x0C0A
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'es-ES'
Found UAS (USB 3.0) device 'SanDisk SDSSDP256G SCSI Disk Device' (2109:0715)
Device eliminated because it was detected as a Hard Drive (score 18 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives under 'advanced drive properties'
0 devices found
Found UAS (USB 3.0) device 'SanDisk SDSSDP256G SCSI Disk Device' (2109:0715)
1 device found
Disk type: FIXED, Disk size: 256 GB, Sector size: 512 bytes
Cylinders: 31130, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 2
Disk GUID: {14195FD4-87BB-4DFA-B10B-4D36EEEF92CD}
Max parts: 128, Start Offset: 17408, Usable = 256060480000 bytes
Partition 1:
  Type: Microsoft System Reserved Partition
  Name: 'Microsoft reserved partition'
  ID: {0817BAB2-B065-4083-816D-EDFE2F31F4AE}
  Size: 16.0 MB (16759808 bytes)
  Start Sector: 34, Attributes: 0x0000000000000000
Partition 2:
  Type: Microsoft Basic Data Partition
  Name: 'Basic data partition'
  ID: {B2429F58-BFCB-48FF-B1BF-0B767EEB2713}
  Size: 238.5 GB (256042336256 bytes)
  Start Sector: 32768, Attributes: 0x0000000000000000

Format operation started
Requesting disk access...
Will use 'G:' as volume mountpoint
Could not load VDS Service: [0x80070005] Access is denied.
Opened \\.\PhysicalDrive4 for exclusive write access
Requesting logical volume handle...
I/O boundary checks disabled
Analyzing existing boot records...
Drive has a Windows 7 Master Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 34 sectors
Initializing disk...
Partitioning (MBR)...
● Creating Main Data Partition (offset: 1048576, size: 238.5 GB)
Closing existing volume...
Waiting for logical drive to reappear...
Formatting (NTFS)...
Using cluster size: 4096 bytes
Quick format was selected
Creating file system...
Error while formatting
[0x00000057] The parameter is incorrect. - Retrying...
[0x00000057] The parameter is incorrect. - Retrying...
[0x00000057] The parameter is incorrect. - Retrying...
[0x00000057] The parameter is incorrect. - Retrying...
Format error: Undetermined error while formatting.
Re-mounted volume as G: after error

Found UAS (USB 3.0) device 'SanDisk SDSSDP256G SCSI Disk Device' (2109:0715)
1 device found
No volume information for drive 0x84
Disk type: FIXED, Disk size: 256 GB, Sector size: 512 bytes
Cylinders: 31130, Tracks per cylinder: 255, Sectors per track: 63
[... 52 lines cut ...]
● Creating Main Data Partition (offset: 1048576, size: 238.5 GB)
Closing existing volume...
Waiting for logical drive to reappear...
Formatting (NTFS)...
Using cluster size: 4096 bytes
Quick format was selected
Creating file system...
Format completed.
Writing Master Boot Record...
Using Zeroed MBR
Found volume \\?\Volume{00060ac0-0000-0000-0000-100000000000}\
Disabling file indexing...
Successfully remounted \\?\Volume{00060ac0-0000-0000-0000-100000000000}\ as G:

Found UAS (USB 3.0) device 'SanDisk SDSSDP256G SCSI Disk Device' (2109:0715)
1 device found
Disk type: FIXED, Disk size: 256 GB, Sector size: 512 bytes
Cylinders: 31130, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0xEF50BC64
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: NTFS (0x07)
  Size: 238.5 GB (256059465728 bytes)
  Start Sector: 2048, Boot: No

pbatard, 2021-09-19:
Well, I tested on a different Windows 11 machine (this time not a Windows To Go one, but a system that went through a full install) and, as long as the USB I was trying to format didn't have an MSR, I saw no issue using the Windows Store version (apart from the notice about not being able to access `VDS`).

I did see `[0x00000057]` and `[0x00000005]` errors when trying to format a drive that had an MSR though, but those issues went away after zeroing a bit of the drive with <kbd>Alt</kbd>-<kbd>Z</kbd>.

So my current conclusion is that this looks mostly like Windows being an ass about preventing access to system partitions (which isn't exactly new), and not a systemic issue that everyone will face the minute they try to format a drive using Rufus on Windows 11.

I'll see if I can add new workarounds against this annoyance, but I think I'll wait until the formal release of Windows 11 to see how many folks are actually impacted...

xatornet, 2021-09-19:
I've tried on different system too, and both Alt-z and Alt-, work. I'll keep using Alt-, since zeroing the drive takes long, even using the fast mode.It's very weird. 

By the way, I forgot to congratulate you for the amazing work you've been doing. This tool is a MUST. Thank you.

NGRaziel, 2021-09-26:
I'm on Windows 11 dev 22463.1000 and I'm seeing the same behavior, Alt+, fixes it.

pbatard, 2021-09-26:
Thanks for the report. Considering that I am still not able to replicate it, I am curious to know what edition of Windows 11 you guys have installed. Was it Home? Pro? Enterprise? Something else? I'm thinking that the edition may also be a factor in this, especially if you you are running Windows 11 Home edition, as Microsoft seems to restrict things a bit more on Home...

xatornet, 2021-09-26:
In my case, Windows 11 Pro, right now build 22463.1000 from Dev Channel.

pbatard, 2021-09-27:
@xatornet, can you confirm that you are only seeing the error when the USB you are trying to reformat contains an MSR? Or do you also see it when there's only a single regular data partition?

JohnLGalt, 2021-09-27:
I've also experienced this issue, previously on 2200.12x, then after cleanly installing again to the DEV build, I'm on 22458 and cannot upgrade t o22463, nor was I able to make the installer drive via Rufus.

The info here helps a lot, and I'm going to try to both diskpart clean as well as quick format the drive and see if this works to create the installer from 22463 .ISO.

I will report back if the error persists or not, and during which of those conditions.

pbatard, 2021-09-27:
@JohnLGalt, can you please provide a full log from Rufus of your error? This will help me devise the type of workaround I should apply in the next version of Rufus.

JohnLGalt, 2021-09-27:
Sure thing!  Except I'm not sure how to obtain said logs from portable version.

As soon as I figure that out, I'll grab one without modifying the current USB, and attempt to use Rufus so it grabs full logs.

pbatard, 2021-09-27:
Click the small log button (left of `START`) or <kbd>Ctrl</kbd>-<kbd>L</kbd>

JohnLGalt, 2021-09-27:
@pbatard Log when trying to use Rufus on UUP Dump created .ISO for Win11 build 22463:

`2 devices found
Could not get Disk Extents: [0x00000001] Incorrect function.
Could not get Disk Extents: [0x00000001] Incorrect function.
Could not get Disk Extents: [0x00000001] Incorrect function.
Disk type: Removable, Disk size: 32 GB, Sector size: 512 bytes
Cylinders: 3772, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 2
Disk ID: 0xB46B9863
Drive has a Windows 7 Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 2.0 GB (2097152000 bytes)
  Start Sector: 2048, Boot: Yes
Partition 2:
  Type: FAT32 (0x07)
  Size: 26.9 GB (28931260416 bytes)
  Start Sector: 4098048, Boot: No
Scanning image...
ISO analysis:
  Image is an UDF image
Disk image analysis:
  Image does not have a Boot Marker
ISO label: 'CCSA_X64FRE_EN-US_DV5'
  Size: 3.4 GB (Projected)
  Uses: EFI
  Uses: Bootmgr (BIOS and UEFI)
  Uses: Install.esd (version 0.14.0+)
Using image: 22463.1000.210917-1503.RS_PRERELEASE_CLIENTMULTI_X64FRE_EN-US.ISO (3.4 GB)

Format operation started
Requesting disk access...
Will use 'L:' as volume mountpoint
Deleting partitions:
● Partition 1 (offset: 1048576, size: 2.0 GB)
● Partition 2 (offset: 2098200576, size: 26.9 GB)
Could not get Disk Extents: [0x00000001] Incorrect function.
Could not get Disk Extents: [0x00000001] Incorrect function.
Could not get Disk Extents: [0x00000001] Incorrect function.
Opened \\.\PhysicalDrive5 for exclusive write access
Analyzing existing boot records...
Drive has a Windows 7 Master Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Initializing disk...
Partitioning (GPT)...
● Creating Main Data Partition (offset: 1048576, size: 28.9 GB)
● Creating UEFI:NTFS Partition (offset: 31028353024, size: 512 KB)
Writing UEFI:NTFS data...
Waiting for logical drive to reappear...
Formatting (NTFS)...
Using cluster size: 4096 bytes
Quick format was selected
Creating file system...
Error while formatting
[0x00000057] The parameter is incorrect. - Retrying...
[0x00000057] The parameter is incorrect. - Retrying...
[0x00000057] The parameter is incorrect. - Retrying...
[0x00000057] The parameter is incorrect. - Retrying...
Format error: Undetermined error while formatting.

Found USB 3.0 device 'SanDisk Ultra USB Device' (0781:5581)
Found USB 2.0 device 'PNY USB 3.0 FD USB Device' (154B:00ED)
NOTE: This device is a USB 3.0 device operating at lower speed...
2 devices found
Could not get Disk Extents: [0x00000001] Incorrect function.
Could not get Disk Extents: [0x00000001] Incorrect function.
Could not get Disk Extents: [0x00000001] Incorrect function.
No volume information for drive 0x85
Disk type: Removable, Disk size: 32 GB, Sector size: 512 bytes
Cylinders: 3772, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 2
Disk GUID: {04DC6195-D3D0-4462-86B2-9F74EC751681}
Max parts: 128, Start Offset: 17408, Usable = 31029426688 bytes
Partition 1:
  Type: Microsoft Basic Data Partition
  Name: 'Main Data Partition'
  ID: {B29C9EE9-B7CF-47E4-A125-B357A068BE15}
  Size: 28.9 GB (31027304448 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
Partition 2 (UEFI:NTFS):
  Type: Microsoft Basic Data Partition
  Name: 'UEFI:NTFS'
  ID: {27332DFF-12CE-4A10-857B-B663344BBD5F}
  Size: 512 KB (524288 bytes)
  Start Sector: 60602252, Attributes: 0x9000000000000000
`

pbatard, 2021-09-27:
Can you please not truncate the log? I'm afraid I need to see the whole thing. Thanks.

JohnLGalt, 2021-09-27:
From start to finish you mean?  I think I cleared it before running, I assume that is what you mean?

JohnLGalt, 2021-09-27:
@pbatard 

```
Rufus x86 v3.15.1812 (Portable)
Windows version: Windows 10 64-bit (Build 22458.1000)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.06
System locale ID: 0x0409 (en-US)
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found USB 3.0 device 'SanDisk Ultra USB Device' (0781:5581)
Found USB 2.0 device 'PNY USB 3.0 FD USB Device' (154B:00ED)
NOTE: This device is a USB 3.0 device operating at lower speed...
2 devices found
Could not get Disk Extents: [0x00000001] Incorrect function.
Could not get Disk Extents: [0x00000001] Incorrect function.
Could not get Disk Extents: [0x00000001] Incorrect function.
Disk type: Removable, Disk size: 32 GB, Sector size: 512 bytes
Cylinders: 3738, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x0002ABF5
Drive has a Windows 7 Master Boot Record
Partition 1:
  Type: NTFS (0x07)
  Size: 28.6 GB (30751588352 bytes)
  Start Sector: 2048, Boot: Yes
2 devices found
Could not get Disk Extents: [0x00000001] Incorrect function.
Could not get Disk Extents: [0x00000001] Incorrect function.
Could not get Disk Extents: [0x00000001] Incorrect function.
No volume information for drive 0x85
Disk type: Removable, Disk size: 32 GB, Sector size: 512 bytes
Cylinders: 3772, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 2
Disk GUID: {04DC6195-D3D0-4462-86B2-9F74EC751681}
Max parts: 128, Start Offset: 17408, Usable = 31029426688 bytes
Partition 1:
  Type: Microsoft Basic Data Partition
  Name: 'Main Data Partition'
  ID: {B29C9EE9-B7CF-47E4-A125-B357A068BE15}
  Size: 28.9 GB (31027304448 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
Partition 2 (UEFI:NTFS):
  Type: Microsoft Basic Data Partition
  Name: 'UEFI:NTFS'
  ID: {27332DFF-12CE-4A10-857B-B663344BBD5F}
  Size: 512 KB (524288 bytes)
  Start Sector: 60602252, Attributes: 0x9000000000000000
Scanning image...
ISO analysis:
  Image is an UDF image
Disk image analysis:
  Image does not have a Boot Marker
ISO label: 'CCSA_X64FRE_EN-US_DV5'
  Size: 3.4 GB (Projected)
  Uses: EFI
  Uses: Bootmgr (BIOS and UEFI)
  Uses: Install.esd (version 0.14.0+)
Using image: 22463.1000.210917-1503.RS_PRERELEASE_CLIENTMULTI_X64FRE_EN-US.ISO (3.4 GB)

Format operation started
Requesting disk access...
Will use 'K:' as volume mountpoint
Deleting partitions:
● Partition 1 (offset: 1048576, size: 28.9 GB)
● Partition 2 (offset: 31028353024, size: 512 KB)
Could not get Disk Extents: [0x00000001] Incorrect function.
Could not get Disk Extents: [0x00000001] Incorrect function.
Could not get Disk Extents: [0x00000001] Incorrect function.
Opened \\.\PhysicalDrive5 for exclusive write access
Analyzing existing boot records...
Drive has a Zeroed Master Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Initializing disk...
Partitioning (MBR)...
● Creating Main Data Partition (offset: 1048576, size: 28.9 GB)
● Creating UEFI:NTFS Partition (offset: 31028385280, size: 512 KB)
Writing UEFI:NTFS data...
Waiting for logical drive to reappear...
Formatting (NTFS)...
Using cluster size: 4096 bytes
Quick format was selected
Creating file system...
Error while formatting
[0x00000057] The parameter is incorrect. - Retrying...
[0x00000057] The parameter is incorrect. - Retrying...
[0x00000057] The parameter is incorrect. - Retrying...
[0x00000057] The parameter is incorrect. - Retrying...
Format error: Undetermined error while formatting.
Re-mounted volume as K: after error

Found USB 3.0 device 'SanDisk Ultra USB Device' (0781:5581)
Found USB 2.0 device 'PNY USB 3.0 FD USB Device' (154B:00ED)
NOTE: This device is a USB 3.0 device operating at lower speed...
2 devices found
Could not get Disk Extents: [0x00000001] Incorrect function.
Could not get Disk Extents: [0x00000001] Incorrect function.
Could not get Disk Extents: [0x00000001] Incorrect function.
No volume information for drive 0x85
Disk type: Removable, Disk size: 32 GB, Sector size: 512 bytes
Cylinders: 3772, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 2
Disk ID: 0x033A5746
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: NTFS/exFAT/UDF (0x07)
  Size: 28.9 GB (31027336704 bytes)
  Start Sector: 2048, Boot: Yes
Partition 2 (UEFI:NTFS):
  Type: EFI FAT (0xef)
  Size: 512 KB (524288 bytes)
  Start Sector: 60602315, Boot: No
Found USB 3.0 device 'SanDisk Ultra USB Device' (0781:5581)
Found USB 2.0 device 'PNY USB 3.0 FD USB Device' (154B:00ED)
NOTE: This device is a USB 3.0 device operating at lower speed...
2 devices found
Could not get Disk Extents: [0x00000001] Incorrect function.
Could not get Disk Extents: [0x00000001] Incorrect function.
Could not get Disk Extents: [0x00000001] Incorrect function.
No volume information for drive 0x85
Disk type: Removable, Disk size: 32 GB, Sector size: 512 bytes
Cylinders: 3772, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 2
Disk ID: 0x033A5746
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: NTFS/exFAT/UDF (0x07)
  Size: 28.9 GB (31027336704 bytes)
  Start Sector: 2048, Boot: Yes
Partition 2 (UEFI:NTFS):
  Type: EFI FAT (0xef)
  Size: 512 KB (524288 bytes)
  Start Sector: 60602315, Boot: No

```

pbatard, 2021-09-27:
Thanks a lot. That's what I wanted to see.

CornishRattler, 2021-09-27:
I am unable to format a usb on the latest and previous dev builds but not on beta builds

kmai-universe, 2021-09-28:
I can confirm this issue. However i can't get it to format if i hit Alt+ either. Running Rufus 3.15 and latest Windows 11 Pro Dev build.

xatornet, 2021-09-28:
> @xatornet, can you confirm that you are only seeing the error when the USB you are trying to reformat contains an MSR? Or do you also see it when there's only a single regular data partition?

Sorry for the late response. The issue is very weird and here's why:

I've tried formating a GPT pendrive to GPT again, to rule out the issue and check if it happens not only on MBR to GPT formatting. And it correctly and succesfully did it, as it is showed on the log:

```
Rufus x86 v3.15.1812
Windows version: Windows 10 64-bit (Build 22463.1000)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.06
System locale ID: 0x0C0A (es-ES)
Will use default UI locale 0x0C0A
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'es-ES'
Found USB 3.0 device 'Samsung Flash Drive USB Device' (090C:1000)
Found UAS (USB 3.1) device 'SanDisk SDSSDP256G SCSI Disk Device' (2109:0715)
Device eliminated because it was detected as a Hard Drive (score 18 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives under 'advanced drive properties'
1 device found
Disk type: Removable, Disk size: 32 GB, Sector size: 512 bytes
Cylinders: 3899, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 1
Disk GUID: {3ADEDEA5-9EB6-40F4-93CD-BA8B99C5FBD2}
Max parts: 128, Start Offset: 17408, Usable = 32078002688 bytes
Partition 1:
  Type: Microsoft Basic Data Partition
  Name: 'Main Data Partition'
  ID: {BECC584A-D29C-4486-8BC8-67E049758A3C}
  Size: 29.9 GB (32076971520 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000

Format operation started
Requesting disk access...
Will use 'I:' as volume mountpoint
Deleting partitions:
● Partition 2 (offset: 1048576, size: 29.9 GB)
Opened \\.\PhysicalDrive5 for exclusive write access
Analyzing existing boot records...
Drive has a Zeroed Master Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Initializing disk...
Partitioning (GPT)...
● Creating Main Data Partition (offset: 1048576, size: 29.9 GB)
Waiting for logical drive to reappear...
Formatting (NTFS)...
Using cluster size: 4096 bytes
Quick format was selected
Creating file system...
Format completed.
Found volume \\?\Volume{9c66a0bb-7b14-4dc8-a2b5-9f63bb7e0579}\
Disabling file indexing...
Successfully remounted \\?\Volume{9c66a0bb-7b14-4dc8-a2b5-9f63bb7e0579}\ as I:

Found USB 3.0 device 'Samsung Flash Drive USB Device' (090C:1000)
Found UAS (USB 3.1) device 'SanDisk SDSSDP256G SCSI Disk Device' (2109:0715)
Device eliminated because it was detected as a Hard Drive (score 18 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives under 'advanced drive properties'
1 device found
Disk type: Removable, Disk size: 32 GB, Sector size: 512 bytes
Cylinders: 3899, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 1
Disk GUID: {9D182DEB-EA40-4F20-B9E6-9FC768D87E28}
Max parts: 128, Start Offset: 17408, Usable = 32078002688 bytes
Partition 1:
  Type: Microsoft Basic Data Partition
  Name: 'Main Data Partition'
[... 79 lines cut ...]
[0x00000057] The parameter is incorrect. - Retrying...
[0x00000057] The parameter is incorrect. - Retrying...
[0x00000057] The parameter is incorrect. - Retrying...
[0x00000057] The parameter is incorrect. - Retrying...
Format error: Undetermined error while formatting.
Re-mounted volume as H: after error

Found USB 3.0 device 'Samsung Flash Drive USB Device' (090C:1000)
Found UAS (USB 3.1) device 'SanDisk SDSSDP256G SCSI Disk Device' (2109:0715)
2 devices found
No volume information for drive 0x86
Disk type: FIXED, Disk size: 256 GB, Sector size: 512 bytes
Cylinders: 31130, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 1
Disk GUID: {2893542F-7C72-4607-8552-6A7E54597080}
Max parts: 128, Start Offset: 17408, Usable = 256060480000 bytes
Partition 1:
  Type: Microsoft Basic Data Partition
  Name: 'Main Data Partition'
  ID: {88F38A31-E41F-45F9-8DE9-970A6B01D1EC}
  Size: 238.5 GB (256059448832 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
```

So I think that the issue has more to do with the kind of drive we use, more than if it's MBR or GPT. But I'm no expert on this :-)

xatornet, 2021-09-28:
> I can confirm this issue. However i can't get it to format if i hit Alt+ either. Running Rufus 3.15 and latest Windows 11 Pro Dev build.

It's "Alt + , " or Alt plus comma. It should say at the bottom something like this:

[image]

kmai-universe, 2021-09-28:
> > I can confirm this issue. However i can't get it to format if i hit Alt+ either. Running Rufus 3.15 and latest Windows 11 Pro Dev build.
> 
> It's "Alt + , " or Alt plus comma. It should say at the bottom something like this:
> 
> [image]

Ah ok, i'll try that, thanks

CornishRattler, 2021-09-28:
It gets a little weirder, format fails in dev build but is fine on beta builds. So flashed an iso in beta and then went back to dev. It worked fine when overwriting a previous successful flash so I wiped and formatted with diskpart and tried again in the dev build......again no go, format error. So summary:
1. Format error in Dev build after diskpart
2. Same stick flashed fine after going to beta
3. Back to Dev and it flashed fine overwriting what I had done in beta
4. Formatted the stick using diskpart in Dev
5. Tried to flash the same iso and got the same format error.

pbatard, 2021-09-28:
For the record, I installed the latest dev build (`22463.1000`) and I am not seeing the `[0x00000057] The parameter is incorrect. - Retrying...` issue outside of the Windows Store version. And I can actually explain that one on account that, the Windows Store version can't access the VDS services, which results in Rufus being unable to clean the drive before partitioning, which is done precisely so that Windows doesn't try to lock partitions such as MSRs or ESPs that exist on the drive. And that's why zeroing the drive (which is a different but much more costly method of cleaning up the partitions) works, because then Windows sees that there's no longer anything of interest on the drive, and lets us format it.

And, similarly, that's also why disable exclusive drive locking works too, because then we are simply telling Windows _"You wanna keep a lock on these partitions even though we're going to delete them? Fine..."_

I am however not seeing `[0x00000057]` with the non Store version of Rufus on `22463.1000`, or `Could not get Disk Extents: [0x00000001] Incorrect function`.

What I am seeing on the other hand are `[0x00000005] Access is denied...` errors when creating a drive that contains an ESP (e.g, for UEFI:NTFS) or MSR (e.g. for Windows To Go) partition, which I can also explain by Windows 11 probably being __a lot__ more proactive than Windows 10 when trying to keep a lock on system partitions when it sees one, and thus managing to drive Rufus out of being able to claim an exclusive lock.

As such I am planning to add the following workarounds:
1. If Rufus is unable to clear partitions, as is currently the case with the Store version, it will automatically disable exclusive drive locking (i.e. same as <kbd>Alt</kbd>-<kbd>,</kbd>) so that we can proceed with the formatting even as Windows is poking the drive.
2. If Rufus is creating ESP or MSRs on Windows 11, it will also disable exclusive drive locking, so that we stop being kicked out by Windows already having gained its on lock by the time we try to claim our exclusive one...

I believe the above should be enough to get us going on Windows 11, especially as it looks to me like some of the other issues being mentioned in this thread are probably not directly related to using Windows 11...

I'll try to produce an ALPHA with these proposed changes when I have a chance, so that you guys can test and report (unfortunately, I can't produce a Windows Store ALPHA, so Windows Store testing will have to wait for the actual 3.16 release). I'll keep you posted.

JohnLGalt, 2021-09-28:
> 
> 
> For the record, I installed the latest dev build (`22463.1000`) and I am not seeing the `[0x00000057] The parameter is incorrect. - Retrying...` issue outside of the Windows Store version. And I can actually explain that one on account that, the Windows Store version can't access the VDS services, which results in Rufus being unable to clean the drive before partitioning, which is done precisely so that Windows doesn't try to lock partitions such as MSRs or ESPs that exist on the drive. And that's why zeroing the drive (which is a different but much more costly method of cleaning up the partitions) works, because then Windows sees that there's no longer anything of interest on the drive, and lets us format it.
> 
> And, similarly, that's also why disable exclusive drive locking works too, because then we are simply telling Windows _"You wanna keep a lock on these partitions even though we're going to delete them? Fine..."_
> 
> I am however not seeing `[0x00000057]` with the non Store version of Rufus on `22463.1000`, or `Could not get Disk Extents: [0x00000001] Incorrect function`.
> 
> What I am seeing on the other hand are `[0x00000005] Access is denied...` errors when creating a drive that contains an ESP (e.g, for UEFI:NTFS) or MSR (e.g. for Windows To Go) partition, which I can also explain by Windows 11 probably being **a lot** more proactive than Windows 10 when trying to keep a lock on system partitions when it sees one, and thus managing to drive Rufus out of being able to claim an exclusive lock.
> 
> As such I am planning to add the following workarounds:
> 
>     1. If Rufus is unable to clear partitions, as is currently the case with the Store version, it will automatically disable exclusive drive locking (i.e. same as Alt-,) so that we can proceed with the formatting even as Windows is poking the drive.
> 
>     2. If Rufus is creating ESP or MSRs on Windows 11, it will also disable exclusive drive locking, so that we stop being kicked out by Windows already having gained its on lock by the time we try to claim our exclusive one...
> 
> 
> I believe the above should be enough to get us going on Windows 11, especially as it looks to me like some of the other issues being mentioned in this thread are probably not directly related to using Windows 11...
> 
> I'll try to produce an ALPHA with these proposed changes when I have a chance, so that you guys can test and report (unfortunately, I can't produce a Windows Store ALPHA, so Windows Store testing will have to wait for the actual 3.16 release). I'll keep you posted.

Just so you know, though, I'm trying to build a USB installer of 22463.100 from .ISO while running 22458.100 Dev build, and I'm only using the portable version.

The whole point was to build the USB installer *because* 22463.100 will not install via Windows Update, so I had to use UUP Dump to generate an .ISO for 22463.100 and put it on USB to try an in-place upgrade from 22458.100 --> 22463.100.

I'll give all the manual workarounds one more shot, and if they still do not work (and I watch for your pre-release builds always, too, so if that drops soon  I'll try it as well), and hopefully I can get this resolved.  I had an issue moving from Beta preview release to DEV 22454.100 release, and just cleanly installed Windows not a month ago, and simply don't have time to cleanly install *again*, and with WU not taking the upgrade, it's either 1) in-place upgrade, or 2) wait for next release and just skip 22463 entirely.

pbatard, 2021-09-28:
Please try https://files.akeo.ie/test/rufus-3.16_ALPHA.exe and let me know if it works better. Thanks!

pbatard, 2021-09-28:
@xatornet, is that also the case with <kbd>Alt</kbd>-<kbd>,</kbd>?

pbatard, 2021-09-28:
Also, you may want to try <kbd>Alt<kbd>-<kbd>V</kbd> to try to use VDS for formatting (since you aren't using the Windows Store version), to see if that works.

xatornet, 2021-09-28:
> @xatornet, is that also the case with Alt-,?

No, the log shows the behaviour without any secret mode.

If I apply Alt-, it works and formats it correctly. I'll paste a log with Alt-, mode enabled here:

```
Rufus x86 v3.16.1826 (Alpha)
Windows version: Windows 11 Pro, 64-bit (Build 22463.1000)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.06
System locale ID: 0x0C0A (es-ES)
Detected installed Windows Language Pack for 0x0409 (English (English))
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found UAS (USB 3.1) device 'SanDisk SDSSDP256G SCSI Disk Device' (2109:0715)
Device eliminated because it was detected as a Hard Drive (score 18 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives under 'advanced drive properties'
0 devices found
Found UAS (USB 3.1) device 'SanDisk SDSSDP256G SCSI Disk Device' (2109:0715)
1 device found
Disk type: FIXED, Disk size: 256 GB, Sector size: 512 bytes
Cylinders: 31130, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 1
Disk GUID: {DFFC0E03-93BD-460D-AA73-42B4CB8C309D}
Max parts: 128, Start Offset: 17408, Usable = 256060480000 bytes
Partition 1:
  Type: Microsoft Basic Data Partition
  Name: 'Main Data Partition'
  ID: {DBA96922-172B-4EBB-A66C-78B3553106A9}
  Size: 238.5 GB (256059448832 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000

Format operation started
Requesting disk access...
Will use 'H:' as volume mountpoint
Deleting partitions:
● Partition 1 (offset: 1048576, size: 238.5 GB)
Opened \\.\PhysicalDrive5 for shared write access
Analyzing existing boot records...
Drive has a Zeroed Master Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Initializing disk...
Partitioning (GPT)...
● Creating Main Data Partition (offset: 1048576, size: 238.5 GB)
Waiting for logical drive to reappear...
Formatting (NTFS)...
Using cluster size: 4096 bytes
Quick format was selected
Creating file system...
Format completed.
Found volume \\?\Volume{1de3f8b6-ba76-4846-9aa1-1a7d2d3b0235}\
Disabling file indexing...
Successfully remounted \\?\Volume{1de3f8b6-ba76-4846-9aa1-1a7d2d3b0235}\ as H:

Found UAS (USB 3.1) device 'SanDisk SDSSDP256G SCSI Disk Device' (2109:0715)
1 device found
Disk type: FIXED, Disk size: 256 GB, Sector size: 512 bytes
Cylinders: 31130, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 1
Disk GUID: {C2857A12-F37C-445F-B275-61EF32DD060D}
Max parts: 128, Start Offset: 17408, Usable = 256060480000 bytes
Partition 1:
  Type: Microsoft Basic Data Partition
  Name: 'Main Data Partition'
  ID: {1DE3F8B6-BA76-4846-9AA1-1A7D2D3B0235}
  Size: 238.5 GB (256059448832 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
```

xatornet, 2021-09-28:
> Also, you may want to try Alt-V to try to use VDS for formatting (since you aren't using the Windows Store version), to see if that works.

I'll give it a go, and post the log in a minute

CornishRattler, 2021-09-28:
3.16 Alpha worked without issue for me, cannot now replicate my earlier format problems.

pbatard, 2021-09-28:
Thanks for the reports. I'll think about it some more to decide what I should do with the release. It's quite annoying that the Store version doesn't work with VDS, else I would have made the switch to using VDS by default...

Note that, unlike <kbd>Alt</kbd>-<kbd>,</kbd>, <kbd>Alt</kbd>-<kbd>V</kbd> is a persistent option, meaning that it'll be preserved next time you reopen Rufus.

adiantek, 2021-09-28:
Creating standard install Windows failed. My CPU is AMD64 and my image is ARM64. Windows To GO created successfully (but unable to boot on Surface Pro X).
```
Rufus x86 v3.16.1826 (Alpha) (Portable)
Windows version: Windows 11 Edu, 64-bit (Build 22463.1000)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.06
System locale ID: 0x0409 (en-US)
NOTE: No Windows Language Pack is installed for Polish (Polski) on this system.
This means that some controls may still be displayed using the system locale.
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'pl-PL'
Notice: The ISO download feature has been deactivated because 'Check for updates' is disabled in your settings.
Found UAS (USB 3.0) device 'TS128GMT E110S SCSI Disk Device' (152D:0583)
Device eliminated because it was detected as a Hard Drive (score 13 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives under 'advanced drive properties'
0 devices found
Found UAS (USB 3.0) device 'TS128GMT E110S SCSI Disk Device' (152D:0583)
1 device found
Disk type: FIXED, Disk size: 128 GB, Sector size: 512 bytes
Cylinders: 15566, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 3
Disk GUID: {9D1B948E-71B5-40BB-B266-9812DBD41444}
Max parts: 128, Start Offset: 17408, Usable = 128035641856 bytes
Partition 1:
  Type: EFI System Partition
  Name: 'EFI System Partition'
  ID: {EF6DE117-6AA4-471B-9F08-A580599EC235}
  Size: 260 MB (272629760 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
Partition 2:
  Type: Microsoft System Reserved Partition
  Name: 'Microsoft Reserved Partition'
  ID: {03C4DD8A-EFC1-49C1-BE9D-2B7F1E8DD35D}
  Size: 128 MB (134217728 bytes)
  Start Sector: 534528, Attributes: 0x0000000000000000
Partition 3:
  Type: Microsoft Basic Data Partition
  Name: 'Main Data Partition'
  ID: {4F54689B-00A6-4A1D-8D7E-0FDC098FE1A2}
  Size: 118.9 GB (127627763200 bytes)
  Start Sector: 796672, Attributes: 0x0000000000000000
Scanning image...
ISO analysis:
  Image is a UDF image
Disk image analysis:
  Image does not have a Boot Marker
ISO label: 'CCRA_A64FRE_EN-US_DV5'
  Size: 3.2 GB (Projected)
  Uses: EFI
  Uses: Bootmgr (UEFI only)
  Uses: Install.esd (version 0.14.0+)
Using image: 19041.1.191206-1406.VB_RELEASE_CLIENTCORE_OEMRET_A64FRE_EN-US.ISO (3.2 GB)

Format operation started
Requesting disk access...
Will use 'D:' as volume mountpoint
Deleting partitions:
● Partition 1 (offset: 1048576, size: 260 MB)
● Partition 2 (offset: 273678336, size: 128 MB)
● Partition 3 (offset: 407896064, size: 118.9 GB)
Opened \\.\PhysicalDrive2 for exclusive write access
Analyzing existing boot records...
Drive has a Zeroed Master Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 2176 sectors
Initializing disk...
Partitioning (GPT)...
● Creating Main Data Partition (offset: 1048576, size: 119.2 GB)
Waiting for logical drive to reappear...
Formatting (Large FAT32)...
Opened \\?\Volume{8391d232-3924-4f96-ae2c-f0403d472be9} for exclusive write access
Size : 119.2 GB 250067599 sectors
Cluster size 32768 bytes, 512 bytes per sector
Volume ID is 1d1a:4059
32 Reserved sectors, 30519 sectors per FAT, 2 FATs
3906352 Total clusters
3906351 Free clusters
Clearing out 61134 sectors for reserved sectors, FATs and root cluster...
Write error [0x00000005] Access is denied.
Retrying in 5 seconds...
Write error [0x00000005] Access is denied.
Retrying in 5 seconds...
Write error [0x00000005] Access is denied.
Retrying in 5 seconds...
Write error [0x00000005] Access is denied.
write_sectors: Write error [0x00000005] Access is denied.
  StartSector: 0x00000000, nSectors: 0x80, SectorSize: 0x200
Error clearing reserved sectors
Format error: Write error.
Re-mounted volume as D: after error

Found UAS (USB 3.0) device 'TS128GMT E110S SCSI Disk Device' (152D:0583)
1 device found
No volume information for drive 0x82
Disk type: FIXED, Disk size: 128 GB, Sector size: 512 bytes
Cylinders: 15566, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 1
Disk GUID: {11D67E8A-CDEE-4600-83D1-35F4A8FA20BD}
Max parts: 128, Start Offset: 17408, Usable = 128035641856 bytes
Partition 1:
  Type: Microsoft Basic Data Partition
  Name: 'Main Data Partition'
  ID: {8391D232-3924-4F96-AE2C-F0403D472BE9}
  Size: 119.2 GB (128034610688 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
```

pbatard, 2021-09-28:
@adiantek, does <kbd>Alt</kbd>-<kbd>V</kbd> or <kbd>Alt</kbd>-<kbd>,</kbd> work?

adiantek, 2021-09-28:
> @adiantek, does Alt-V or Alt-, work?

Yes, I need to alt-v or alt-, or both and it works. Thanks

NSbuilder, 2021-10-01:
> Thanks for the reports. I'll think about it some more to decide what I should do with the release. It's quite annoying that the Store version doesn't work with VDS, else I would have made the switch to using VDS by default...
> 
> Note that, unlike Alt-,, Alt-V is a persistent option, meaning that it'll be preserved next time you reopen Rufus.

By the way, you will probably want to restore rufus update functionality in the windows 11 store as this can be done by the developer:
https://www.tomshardware.com/news/windows-11-store-restrictions-win32-apps

pbatard, 2021-10-03:
The new Rufus 3.16 BETA (available [here](https://github.com/pbatard/rufus/releases)) should work better with Windows 11. Please test it and let me know if you are still seeing issues.

pbatard, 2021-10-04:
NB, I had to re-upload the BETA, since I had leftover code in there that shouldn't have been present. If you are testing on Windows 11, please make sure your BETA says `Rufus 3.16.1830 (Beta)` on the app title bar.

biship, 2021-10-06:
I can confirm. On Windows 11, 3.16 beta now works (I was having the same error as OP).

xatornet, 2021-10-06:
> The new Rufus 3.16 BETA (available [here](https://github.com/pbatard/rufus/releases)) should work better with Windows 11. Please test it and let me know if you are still seeing issues.

I can also confirm that it works on the latest Windows 11 dev build. Amazing work :-)

spix18, 2021-10-06:
beta fix it for but after closing explorer.exe because its said the disk is being used by it once i closed that window ''This PC'' and tried again , its worked 
[image]
i hope its finishes all the way

JohnLGalt, 2021-10-09:
Awesome news.  Since I have the regular install, I'll be throwing the various Insider builds builds into VMs, but I have a couple of physical machines that I plan to continue to test on, so I'll be watching for any more issues - I suspect, though, that I won't find any, now that I'm not actively running the Insiders builds on my desktop....

NSbuilder, 2021-10-10:
> The new BETA is the result of wanting to allow users to have the option to create Windows 11 installation media that doesn't require TPM 2.0/Secure Boot/8GB+ RAM for installation.

Does it uses AveYo works ( https://gist.github.com/AveYo/c74dc774a8fb81a332b5d65613187b15#gistcomment-3921908 ) which doesn't require tpm at all?

Or microsoft official guide: https://www.pcmag.com/news/microsoft-offers-tpm-20-bypass-to-install-windows-11-on-unsupported-pcs

which requires tpm 1.2 at least?

ilike2burnthing, 2021-10-10:
What about `BypassCPUCheck`?

pbatard, 2021-10-11:
I haven't seen any evidence that this is a valid key. Do you have a source for that one?

ilike2burnthing, 2021-10-13:
Well that was disappointing. Got my hands on an old laptop from 2012/13 with an i7 3630QM, and all that was needed to bypass the requirements page was `BypassTPMCheck`.

Tried `BypassCPUCheck` on its own (in case the two were interchangeable), but no luck.

pbatard, 2021-10-13:
With the release of Rufus 3.16, I am going to close this issue, so that any remaining problems can be tracked against issues opened against the 3.16 release.