pbatard/rufus #1781: Win to go cannot be detected upon booting

graceshuaizhang, 2021-09-20:
﻿Hello,

I created win 10 to go on a usb disk and hard drive. I tried to boot from them on my macbook pro 2015, but neither can be detected when I press option. Tried another macbook pro 2019 and same issue. T

he installation process was smooth and no error occurred. Yet when booted into OSX, I cannot select win as the booting disk.

Anyone knows how to solve this? Thanks!

JonnyTech, 2021-09-21:
> The installation process was smooth and no error occurred.

Without a full log of Rufus' operation, as required, we have no idea of what exactly you have created.

pbatard, 2021-09-23:
AFAIK, MacBooks (or anything Apple) are tricky to boot Windows on, because Windows aims at being booted on a __standard__ BIOS or UEFI PC, and Apple computers use nonstandard EFI.

This is not a Rufus issue, so I will kindly ask that you post it in a dedicated forum, rather than the issue tracker for Rufus. Thank you.

closed 2021-09-23 by pbatard