pbatard/rufus #1782: command.com not found

edrahn01, 2021-10-02:
I burned FreeDOS to a USB and the USB installation won't run. I get can't find Command Interrupter and it asks me to type one in. I tried downloading command.com but it appears it is already on the USB stick. I however only see LOCALE, and favicon. What am I doing wrong.

Thanks,
Ed

pbatard, 2021-10-02:
> and the USB installation won't run

There is no installation for FreeDOS as provided by Rufus. Either you boot to DOS or you don't.

> I get can't find Command Interrupter and it asks me to type one in

You most likely added some extra files to the USB.
What happens if you create a plain FreeDOS bootable drive in Rufus? If that boots, then whatever happens after you modify its content is outside of what this issue tracker can provide support for.

>  I tried downloading command.com but it appears it is already on the USB stick.

It is. But it also has the __hidden__ attribute which means that, while present, and unless you change the defaults, Windows File Explorer will not show it, just as it will not show any files with the hidden attribute.

> I however only see LOCALE, and favicon

That's expected, as most DOS files are set to hidden so that users don't delete them by mistake.
Try to check _Hidden items_ in the File Explorer bar to see these files:

[image]

> What am I doing wrong.

Hard to tell, since you don't provide enough details about what you did exactly and what you are trying to accomplish. You also chose not to follow the check list, so I can't even see if there were any possible hints as to what you are doing wrong from within Rufus...

pbatard, 2021-10-09:
No update &rarr' Closing

closed 2021-10-09 by pbatard