pbatard/rufus #1786: Command line arguments for silent install without any prompts

kdpuvvadi, 2021-10-09:
﻿

Checklist
---------
- [ ] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [ ] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [ ] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [ ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
<Please describe your issue here>

Trying to install rufus silently with winget and testing it in the sandbox. Validations fails every time because while installing rufus prompts for auto check update and launches itself. 

Log
---
```
<FULL LOG>
```

pbatard, 2021-10-09:
I'd like to understand this a little better.

Especially, how comes the validation of a full GUI application does not appear to be able to deal with GUI prompts?

Surely, a validation process for an application that isn't a commandline one should be designed to handle GUI?
Or, if the issue is that winget somehow expects applications to come with a non GUI installer, then I'd say that that's a bit of a limiting view of how some Windows applications are designed, and winget should be updated to handle applications, such as Rufus (and others) that __don't__ need and installer.

At any rate, I believe there exist a few solutions for you, that aren't going to require adding a custom commandline option that's designed pretty much exclusively for winget usage:
1. Decide on whether you want to enable/disable the check for update and add an additional entry to your winget package that creates the registry key `HKLU\Software\Akeo Consulting\Rufus\UpdateCheckInterval`. This sets the interval between update checks (in seconds) or disables updates altogether if set to `0`.
2. Add a `rufus.app` empty file in the same directory where you have Rufus. This disables the check for update altogether (and removes the "update settings" button in Rufus) as it makes the executable run in appstore mode, which is pretty much the only difference there is between appstore and regular.
3. Make sure your executable is named `rufus.exe` as it should enable the check for update by default without prompting per [this](https://github.com/pbatard/rufus/wiki/FAQ#Rufus_connects_to_the_internet_but_I_never_allowed_it_to__why).

kdpuvvadi, 2021-10-10:
In my opinion, application just installed don't need to check for an update as soon as it installed.  Rufus installer just creating shortcut on start menu to the download location. I don't think that's much better than portable version.

pbatard, 2021-10-10:
> In my opinion, application just installed don't need to check for an update as soon as it installed.

You're still talking in terms of applications that come with an installer (i.e. as if the executable that is running is a `setup.exe` or something that had just been downloaded). 

That is __NOT__ the case of Rufus.

The executable that people download is the __installed__ application. People may therefore be running a months old or even years old executable, and we have no way of telling whether it was just downloaded from the internet.

Therefore, we __MUST__ run the check for update as soon as the application starts, because we really want people who use an older version to get bugfixes and new features.

> Rufus installer just creating shortcut on start menu to the download location.

Again there is no Rufus installer.

> I don't think that's much better than portable version.

The portable and regular version are the same binary and the only difference of behaviour is that one creates/works with a `.ini` file and the other doesn't: https://github.com/pbatard/rufus/wiki/FAQ#What_is_the_difference_between_the_portable_and_the_regular_version

I think you need to stop thinking in terms of what __other__ applications, that come with a formal setup do, and concentrate on what Rufus actually does.