pbatard/rufus #1794: Rufus full-formats my usb device before writing KNOPPIX_V9.1DVD-2021-01-25-EN.iso

as-muncher, 2021-10-14:
﻿

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [ ] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [ ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
Rufus full-formats my usb device when writing KNOPPIX_V9.1DVD-2021-01-25-EN.iso.  I think this is not necessary!

Log
---
```
<FULL LOG>
```

pbatard, 2021-10-14:
I need to see your log before I can answer your question.

Why? Because I can't tell if the image you use is ISOHybrid or not and it will take me a while to download KNOPPIX to find out.

The log is __NOT__ optional.

as-muncher, 2021-10-14:
I'm working on it now.

as-muncher, 2021-10-14:
After using Rufus3.15p to successfully write the image, I tried to use Rofus3.16p to write the image, which then worked.  It must have something to do with Rufus not being able to do something with a Manjaro usb.  Perhaps I had a different image originally on the usb, which then made rufus full-format it when it tried to write knoppix9.  I'll see if I can replicate it.

as-muncher, 2021-10-15:
This is the log when I write manjaro-kde-21.1.2-210904-linux513.iso then try to write KNOPPIX_V9.1DVD-2021-01-25-EN.iso.  
I guess you may have to change the title of this Issue.  It's a failure to partition the device.

Rufus x86 v3.16.1836 (Portable)
Windows version: Windows 10 (Unknown Edition), 64-bit (Build 19043.1288)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.06
System locale ID: 0x0809 (en-GB)
Will use default UI locale 0x0809
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found USB 2.0 device 'Verbatim USB External SSD USB Device' (18A5:0246)
NOTE: This device is a USB 3.0 device operating at lower speed...
Device eliminated because it was detected as a Hard Drive (score 18 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives under 'advanced drive properties'
Found USB 3.0 device 'Philips USB Flash Drive USB Device' (13FE:6300)
1 device found
Disk type: Removable, Disk size: 16 GB, Sector size: 512 bytes
Cylinders: 1927, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 2
Disk ID: 0x6D716F2F
Drive has an unknown Master Boot Record
Partition 2:
  Type: EFI FAT (0xef)
  Size: 17 MB (17825792 bytes)
  Start Sector: 9126432, Boot: No
Partition 3:
  Type: GNU/Linux (0x83)
  Size: 4 MB (4194304 bytes)
  Start Sector: 9161248, Boot: No
Scanning image...
ISO analysis:
  Image is an ISO9660 image
  Will use '/boot/isolinux/isolinux.cfg' for Syslinux
  Detected Syslinux version: 6.03/20150822 (from '/boot/isolinux/isolinux.bin')
Disk image analysis:
  Image has an unknown Master Boot Record
  Image is a bootable disk image
ISO label: 'KNOPPIX_9'
  Size: 4.4 GB (Projected)
  Uses: Syslinux/Isolinux v6.03
  Uses: EFI
  Note: This ISO uses symbolic links, which will not be replicated due to file system
  limitations. Because of this, some features from this image may not work...
Using image: KNOPPIX_V9.1DVD-2021-01-25-EN.iso (4.4 GB)
Will reuse 'ldlinux.sys' and 'ldlinux.bss' from 'C:\software\Rufus\syslinux-6.03/20150822\' for Syslinux installation

Format operation started
Requesting disk access...
Will use 'G:' as volume mountpoint
Opened \\.\PhysicalDrive2 for exclusive write access
Analyzing existing boot records...
Drive has an unknown Master Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Initializing disk...
Partitioning (MBR)...
● Creating Main Data Partition (offset: 1048576, size: 14.8 GB)
Waiting for logical drive to reappear...
Formatting to FAT32 (using IFS)
Using cluster size: 8192 bytes
Quick format was selected
Creating file system...
Format completed.
Writing Master Boot Record...
Partition is already FAT32 LBA...
Set bootable USB partition as 0x80
Using Syslinux MBR
Found volume \\?\Volume{635ef976-2d39-11ec-8a4a-021d2b274cf5}\
[... 146 lines cut ...]
Format operation started
Requesting disk access...
No drive letter was assigned...
Will use 'G:' as volume mountpoint
Opened \\.\PhysicalDrive2 for exclusive write access
Analyzing existing boot records...
Drive has an unknown Master Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Initializing disk...
Partitioning (MBR)...
● Creating Main Data Partition (offset: 1048576, size: 14.8 GB)
Could not set drive layout: [0x00000032] The request is not supported.

Found USB 2.0 device 'Verbatim USB External SSD USB Device' (18A5:0246)
NOTE: This device is a USB 3.0 device operating at lower speed...
Device eliminated because it was detected as a Hard Drive (score 18 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives under 'advanced drive properties'
Found USB 3.0 device 'Philips USB Flash Drive USB Device' (13FE:6300)
1 device found
No volume information for drive 0x82
Disk type: Removable, Disk size: 16 GB, Sector size: 512 bytes
Cylinders: 1927, Tracks per cylinder: 255, Sectors per track: 63
Partition type: SFD (Super Floppy Disk) or unpartitioned

as-muncher, 2021-10-18:
I wrote something but then my computer froze.
I think you did not even read my Issue but rather skimmed over it, and you seemed to make a great big deal about it.  This in no ways was a personal attack against you @pbatard, as if you did not know how to write code or that I could do it better.  Perhaps there was some misunderstanding?  When I mentioned "full format", that does not mean a format.  It means that Rufus was formatting the whole drive, all 16 GB of it, the long way, instead of doing it the regular way, which is a quick format.  As I wrote in previous posts, I tried to duplicate the error, but for some reason couldn't.  I had written a note to myself to post this Issue, because I thought it might be helpful to make your software better, not as a personal attack on you or because I used an exclamation mark.  I didn't find out how to change the title of this Issue until recently.  

For my Windows version, when I open a command prompt, and type "winver", a window pops up that says that I'm using 21H1 (build 19043.1288) and it says "The Windows 10 Home operating system...".  Is there some other command you want me to enter to give you more information?  I'm not sure why it's showing up for you as an unknown version.  And my USB drives seem to work okay in terms of reading and writing files etc, but I ran the USB troubleshooter just in case, which said that the USB controller is in a failed state or is not currently installed.  The USB Controller must be installed or in a working..."  It seems to work, but yeah, Windows is saying there's a problem without being able to fix it.  If you recommend a tool to fix it, I will look into it.

pbatard, 2021-10-19:
Yup, I got that. The best I can do at this stage is wait for somebody else to report the same problem, possibly with some clear stepts on how this can be replicated. But until that happens, I have no choice but to leave that issue aside.

Also, as I mentioned earlier, I expect the `[0x00000032] The request is not supported` (which were an actual bug introduced in 3.16, but then again, the log also makes it clear that this is a real issue) to be fixed in the next version.