pbatard/rufus #1814: RUFUS nagy lépést tenne, ha a következő módosítással törölni tudná a CPU ellenőrzését is. - RUFUS would take a big step if you could delete CPU verification with the next change

anonymusxp, 2021-10-30:
﻿

Checklist
---------
- [ ] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [ ] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [ ] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [ ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
<Please describe your issue here>

Log
---
```
<FULL LOG>
```

anonymusxp, 2021-10-30:
RUFUS would take a big step if you could delete CPU verification with the next change

pbatard, 2021-10-30:
https://github.com/pbatard/rufus/wiki/FAQ#Do_you_plan_to_remove_other_Windows_11_installation_restrictions

closed 2021-10-30 by pbatard

anonymusxp, 2021-10-30:
By checking the CPU, MICROSOFT makes it impossible to install Windows 11 on all machines with unsupported CPUs.
For this reason, I do not understand in the FAQ a: Do you plan to remove other Windows 11 installation restrictions? answer to question: NO! Oh, and whereas some people (on user forums) have suggested the existence of a BypassCPUCheck key, we have seen a ABSOLUTELY NO CONCRETE EVIDENCE that this key actually exists or does anything. BypassCPUCheck excludes the installation of win11 on computers with unsupported CPUs.

pbatard, 2021-10-30:
Please provide evidence that `BypassCPUCheck` actually does anything.

It has been _hinted_ that such a key __may__ exist, but I have yet to see reports of _"I was able to install Windows 11 on this specific CPU when using `BypassCPUCheck`, whereas I couldn't do it without"_.

Especially, I have tested the installation of Windows 11 on PCs where the CPU was not in the list of CPUs that are official supported by Windows 11, and saw no issue, even as I hadn't applied `BypassCPUCheck`. So, unlike all the people who seem to claim that `BypassCPUCheck` is needed, I have direct evidence that it doesn't appear to be.

Or to put it more succinctly:

> By checking the CPU, MICROSOFT makes it impossible to install Windows 11 on all machines with unsupported CPUs

[Citation Needed]

anonymusxp, 2021-10-30:
RUFUS 3.17 has solved the anomalies of a clean installation of Windows 11, but due to the unsupported CPU, version upgrade is not possible by starting from the USB setup.exe file created with RUFUS 3.17 as well as by starting the original ISO setup.exe. Look at the attached screenshots with the same message!
Mydigitallife solved this problem by blocking CPU verification.

pbatard, 2021-10-30:
Mydigitallife is the origin of the _unsubstantiated_ `BypassCPUCheck` claim.

I did [ask for evidence there](https://forums.mydigitallife.net/threads/update-rufus-usb-tool.50859/page-26#post-1699168), and got none. And at any rate, once again, I did install Windows 11 on __unsupported__ CPUs without the `BypassCPUCheck` key.

Also you are talking about attached screenshots, but I can't see any.

So, again, I have to ask, can you please provide verifiable proof that `BypassCPUCheck`  does anything? The fact that some scripts use it is not proof. You must be able to demonstrate that there exists systems out there (by providing their hardware specs) where Windows 11 will not install when `BypassCPUCheck` is not applied, but that will when it is.

Just telling me that you were able to install Windows 11 on an old system with a script that applies `BypassCPUCheck`  is not enough, because it still does not demonstrate that Windows 11 would not have installed without the key.

anonymusxp, 2021-10-30:
Thank you for your answers. At this point I&#39;m getting out.
Rufus was an excellent program, and there will be. The world&#39;s problems are solved by the big ones.
As an end-user, I will not be impatient and I will not be a Keynote in hereinafter.
The point is the success of the end result.