pbatard/rufus #1816: Windows 11 Install Fails, Shows CPU and TPM Check

arrmo, 2021-10-31:
﻿

Checklist
---------
- [X] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [X] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [X] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [X] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [NA] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [X] I also tried one or more of the following:
  - [X] Using a different USB drive.
  - [X] Plugging the USB into a different port.
  - [X] Running Rufus on a different computer.
- [X] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
First off - thanks! I'm somewhat new to Rufus, but it really seems excellent. Much appreciated.

I am trying to run a Windows 11 install, with the CPU and TPM check disabled. I get the error below ... seems the checks are not bypassed? I thought it was perhaps me, selecting the wrong options in Rufus - so I did it a second time, but the same issue. I tried the upgrade with and without the install download checks (i.e. "Not right now"), but no delta. I also manually tried some of the LanSetup registry mods manually, but still no change. I'm more than happy to capture logs, but not sure the Rufus log is really the item needed here?

Just let me know what you need, more than happy to try to help!

Thanks again.

Log
---
Log is not really applicable here, but a pic to show the error,
[image]

pbatard, 2021-10-31:
1. I still want to see your creation log. Because, for one thing, it contains important information about the ISO you are using and other parameters. The log is always needed. Always.
2. Is that the exact error message you got? If so, then your issue is that you are trying to __upgrade__ a running Windows using a media created by Rufus rather than install Windows from scratch from __booting__ the media. Rufus is for creating __bootable__ media, so if you don't plan to actually boot from the media, you shouldn't be using Rufus (because you don't need Rufus to do that: You can just right click on the ISO in Windows, select "Mount" and you will have accomplished the same thing).
3. If you want to upgrade, rather than boot and create a fresh installation of Windows, you will have to use a different method. Rufus is for creating boot media, not upgrade media.

arrmo, 2021-10-31:
Hi,

Thanks for the quick feedback! And the numbering - helps in replying 😆. Some answers,
1) Sure, NP! Attached. Again, just yell if you need anything else, or want me to try something. Really no issue at all.
2) I did try from a USB boot as well - do you want me to try that, capture any errors there? And I can check the Registry settings if you want as well, NP.
3) Yep, as 2 ... I can also run an upgrade if you prefer, booting from the created USB. Would that help?

Thanks again!
[rufus.log](https://github.com/pbatard/rufus/files/7449127/rufus.log)

arrmo, 2021-10-31:
FYI, tried to upgrade from a USB boot, got this,
```
The upgrade option isn't available if you start your computer using the installation media.

Please remove the installation media and restart your computer. After the computer has started normally, insert the installation media, and run Setup.
```

Hmmm ... so the only option is USB boot, and wipe the Windows 10 install (i.e. start clean)?

Thanks!

pbatard, 2021-10-31:
Yes, you need to choose clean install. The goal of Rufus is to create bootable media for clean installation.

From what I see in your log, everything was properly patched for a clean installation of Windows 11, even if you don't have a TPM.

arrmo, 2021-10-31:
OK, makes sense - and my apologies for the sidetrack (and dumb question). I was thinking upgrade was an option. Clean install really isn't, wife won't want her PC down for all to be redone ... LOL.

Thanks again!

closed 2021-10-31 by arrmo

arrmo, 2021-10-31:
BTW, I was following the info [here ](https://www.tomshardware.com/how-to/bypass-windows-11-tpm-requirement)... seems the upgrade part isn't quite right. Just so you know if / when others ask.

Thanks.

marcosfrm, 2021-10-31:
For **upgrades** you need a different registry value:

https://support.microsoft.com/en-us/windows/ways-to-install-windows-11-e0edbbfb-cfc5-4011-868b-2ce77ac7c70e

arrmo, 2021-11-01:
Thanks! Unfortunately, I had tried that one - but it doesn't work, for the reason that,
```
you can create the following registry key values and bypass the check for TPM 2.0 (at least TPM 1.2 is required) and the CPU family and model
```
The Dell laptop I am using has no TPM at all, not even TPM 1.2 😞