pbatard/rufus #1821: Samsung external hd does not appear in rufus version 1.3.2

vc655767, 2021-11-06:
my Samsung HD doesn't appear on rufus

pbatard, 2021-11-06:
Please use the latest version of Rufus.

If you can't use the latest version of Rufus, then it means that you're running Rufus on a Windows platform that I am no longer supporting (such as Windows XP), in which case, you are on your own.

closed 2021-11-06 by pbatard

vc655767, 2021-11-06:
mas quero usar essa versão o meu e o windows 8.1 não sabe de alguma tecla
para apertar e usar pois o meu pendrive aperto ctrl +f e funciona

pbatard, 2021-11-06:
Just like any software developer, I cannot support old versions of my software indefinitely.

If you want to receive support for Rufus, I will kindly ask you that you run the 3.x version, and also, that you do provide the items that were requested from you when opening a support request, such as providing a complete log.

If you can't do that, then I simply have no choice but to decline support.