pbatard/rufus #1824: Store version chrashes when formating a 64GB micro sd card as FAT32

theChaosCoder, 2021-11-12:
﻿

Checklist
---------
- [X] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [X] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.

Issue description
-----------------
**I happens only in the store version of Rufus**. At least since ver 3.16
IDK how to provide a log from the store version since it crashes after clicking on start.
No problems with the "regular" version.

My "Settings".
[image]

pbatard, 2021-11-12:
Please install DebugView from https://docs.microsoft.com/en-us/sysinternals/downloads/debugview.

Rufus sends its log messages there as well, so you can capture the log data even if the application crashes.

Also, even without using DebugView, I'm still interested in your log before the application crash (i.e. what you get right after opening Rufus), especially if you apply the <kbd>Alt</kbd>-<kbd>.</kbd> cheat mode that enables enumeration Debug.

If you do that, you should at least be able to provide something like the following, which may provide important information:
```
Rufus x64 v3.17.1846 (AppStore version)
Windows version: Windows 10 Pro Workstation, 64-bit (Build 19043.1348)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.06
System locale ID: 0x0809 (en-GB)
Will use default UI locale 0x0809
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Notice: Windows VDS is unavailable
Processing Hub 1:
  Hub[1] = '\\?\usb#vid_174c&pid_3074#5&1cc0e58a&0&20#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[059]: USB\VID_0424&PID_5534\6&2C14036A&0&1
Processing Hub 2:
Processing Hub 3:
Processing Hub 4:
  Hub[2] = '\\?\usb#asmediaroot_hub#5&13f8d294&0&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[040]: USB\VID_05E3&PID_0612&ASMEDIAUSBD_HUBSS\00000820
  Found ID[100]: USB\VID_05E3&PID_0610&ASMEDIAUSBD_HUB\00000840
Processing Hub 5:
  Hub[3] = '\\?\usb#vid_05e3&pid_0612&asmediausbd_hubss#00000820#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[113]: USB_ASMEDIA\VID_174C&PID_55AA\000000000025
Processing Hub 6:
  Hub[4] = '\\?\usb#vid_174c&pid_2074#5&1cc0e58a&0&9#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[019]: USB\VID_0424&PID_2134\6&16396D6C&0&1
Processing Hub 7:
Processing Hub 8:
  Hub[5] = '\\?\usb#vid_05e3&pid_0610&asmediausbd_hub#00000840#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[139]: USB\VID_045E&PID_0288&ASMEDIAUSBD_HUB\10000840
Processing Hub 9:
  Hub[6] = '\\?\usb#root_hub30#4&34be82cc&0&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[225]: USB\VID_174C&PID_3074\5&1CC0E58A&0&20
  Found ID[245]: USB\VID_174C&PID_2074\5&1CC0E58A&0&9
  Found ID[119]: USB\VID_0B05&PID_180A\F0038C9991CB
  Found ID[028]: USB\VID_04B3&PID_301A\5&1CC0E58A&0&6
Processing Hub 10:
  Hub[7] = '\\?\usb#vid_04b3&pid_301a#5&1cc0e58a&0&6#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[089]: USB\VID_04B3&PID_301B\6&3522BF70&0&1
  Found ID[182]: USB\VID_046D&PID_C077\6&3522BF70&0&3
Processing Hub 11:
  Hub[8] = '\\?\usb#vid_0424&pid_2134#6&16396d6c&0&1#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[081]: USB\VID_045E&PID_028E\1F3251C
Processing Hub 12:
  Hub[9] = '\\?\usb#vid_045e&pid_0288&asmediausbd_hub#10000840#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[063]: USB\VID_045E&PID_0289\11000040
Processing IDs belonging to 'USBSTOR':
  USB_ASMEDIA\VID_174C&PID_55AA\000000000025
Processing 'SCSI' device:
  Hardware ID: ''
Found non-USB non-removable device 'Asgard AN3 500NVMe-M.2/80' => Eliminated
Processing 'SCSI' device:
  Hardware ID: 'SCSI\DiskWDC_____WD60EMAZ-11LW3B080.0'
Found non-USB non-removable device 'WDC WD60EMAZ-11LW3B0' => Eliminated
Processing 'STORAGE' device:
  Unsupported or disabled by policy
Processing 'SCSI' device:
  Hardware ID: 'SCSI\Disk________ST8000DM004-2CX10001'
Found non-USB non-removable device 'ST8000DM004-2CX188' => Eliminated
Processing 'SCSI' device:
  Hardware ID: ''
Found non-USB non-removable device 'NVME SSD 512GB' => Eliminated
Processing 'USBSTOR' device:
  Hardware ID: 'USBSTOR\DiskMushkin_Ventura_Ultra___0___'
  Matched with ID[113]: USB_ASMEDIA\VID_174C&PID_55AA\000000000025
  Matched with Hub[3]: '\\?\usb#vid_05e3&pid_0612&asmediausbd_hubss#00000820#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
Could not get node connection information for 'USB_ASMEDIA\VID_174C&PID_55AA\000000000025': [0x0000001F] A device attached to the system is not functioning.
Found USB device 'Mushkin Ventura Ultra USB Device' (174C:55AA) [ID]
Using autorun.inf label for drive G: 'CCCOMA_X64FRE_EN-GB_DV9'
1 device found
Windows volumes from this device:
● \\?\Volume{d640ba0e-b195-4539-9278-0e5932cde365}\ @407896064
● \\?\Volume{f88ee577-c19f-4d81-8490-8ef3404ec8be}\ @1048576
Disk type: FIXED, Disk size: 120 GB, Sector size: 512 bytes
Cylinders: 14593, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 3
Disk GUID: {484E3C7A-CA6E-44C6-9324-4B9A2EF07A61}
Max parts: 128, Start Offset: 17408, Usable = 120034089472 bytes
Partition 1:
  Type: EFI System Partition
  Name: 'EFI System Partition'
  ID: {F88EE577-C19F-4D81-8490-8EF3404EC8BE}
  Size: 260 MB (272629760 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
Partition 2:
  Type: Microsoft System Reserved Partition
  Name: 'Microsoft Reserved Partition'
  ID: {D5BFD50D-7DC6-4607-8240-060E0F1E3701}
  Size: 128 MB (134217728 bytes)
  Start Sector: 534528, Attributes: 0x0000000000000000
Partition 3:
  Type: Microsoft Basic Data Partition
  Name: 'Main Data Partition'
  ID: {D640BA0E-B195-4539-9278-0E5932CDE365}
  Size: 111.4 GB (119626210816 bytes)
  Start Sector: 796672, Attributes: 0x0000000000000000
```

So, please provide both the DebugView output from the crash as well as the pre-crash application log.
Note that, if you don't see the line `Rufus x64 v3.17.1846 (AppStore version)` appear in DebugView, you may have to run it as admin and also play with the _Capture_ menu options.

theChaosCoder, 2021-11-12:
Rufus log after start with `alt + .`

```
Rufus x64 v3.17.1846 (AppStore version)
Windows version: Windows 10 Pro, 64-bit (Build 19043.1288)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.06
System locale ID: 0x0407 (de-DE)
Will use default UI locale 0x0407
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'de-DE'
Notice: Windows VDS is unavailable
Processing Hub 1:
  Hub[1] = '\\?\usb#vid_0bda&pid_5412#6&3223911d&0&1#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[207]: USB\VID_046D&PID_C52E\7&2565C421&0&2
Processing Hub 2:
  Hub[2] = '\\?\usb#root_hub30#5&f2e74e8&0&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[074]: USB\VID_2537&PID_1081\0123456789ABCDE
Processing Hub 3:
Processing Hub 4:
Processing Hub 5:
  Hub[3] = '\\?\usb#root_hub30#5&e780179&0&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[057]: USB\VID_046D&PID_C083\187C38503330
  Found ID[184]: USB\VID_0BDA&PID_0412\6&3223911D&0&5
  Found ID[018]: USB\VID_0BDA&PID_5412\6&3223911D&0&1
Processing IDs belonging to 'USBSTOR':
  USB\VID_2537&PID_1081\0123456789ABCDE
Processing 'SCSI' device:
  Hardware ID: ''
Found non-USB non-removable device 'INTEL SSDPEKKW010T8' => Eliminated
Processing 'USBSTOR' device:
  Hardware ID: 'USBSTOR\DiskNORELSYS1081____________0___'
  Matched with ID[074]: USB\VID_2537&PID_1081\0123456789ABCDE
  Matched with Hub[2]: '\\?\usb#root_hub30#5&f2e74e8&0&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
Found USB 3.0 device 'NORELSYS 1081 USB Device' (2537:1081)
Windows volumes from this device:
● \\?\Volume{4f471a68-370e-11ec-9ed7-309c23ad6cbe} @1048576
Processing 'SCSI' device:
  Hardware ID: ''
Found non-USB non-removable device 'PNY CS3030 1TB SSD' => Eliminated
Processing 'SCSI' device:
  Hardware ID: 'SCSI\Disk_________CT1000MX500SSD1M3CR'
Found non-USB non-removable device 'CT1000MX500SSD1' => Eliminated
Processing 'SCSI' device:
  Hardware ID: 'SCSI\DiskSanDisk_____SDSSDH3_2T004110'
Found non-USB non-removable device 'SanDisk SDSSDH3 2T00' => Eliminated
Processing 'SCSI' device:
  Hardware ID: 'SCSI\Disk_________CT2000MX500SSD1M3CR'
Found non-USB non-removable device 'CT2000MX500SSD1' => Eliminated
1 device found
Windows volumes from this device:
● \\?\Volume{4f471a68-370e-11ec-9ed7-309c23ad6cbe}\ @1048576
No volume information for drive 0x86
Disk type: Removable, Disk size: 64 GB, Sector size: 512 bytes
Cylinders: 7764, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x15C22BBE
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 59.5 GB (63863521280 bytes)
  Start Sector: 2048, Boot: No
Unable to send request: The attempt to connect to the server failed.
Found USB 3.0 device 'NORELSYS 1081 USB Device' (2537:1081)
1 device found
No volume information for drive 0x86
Disk type: Removable, Disk size: 64 GB, Sector size: 512 bytes
Cylinders: 7764, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x15C22BBE
[... 206 lines cut ...]
00000197	165.70854187	[9668] Cluster size 32768 bytes, 512 bytes per sector 	
00000198	165.71351624	[9668] Volume ID is 19fe:336d 	
00000199	165.72001648	[9668] 32 Reserved sectors, 15223 sectors per FAT, 2 FATs 	
00000200	165.72145081	[9668] 1948483 Total clusters 	
00000201	165.72372437	[9668] 1948482 Free clusters 	
00000202	165.72601318	[9668] Clearing out 30542 sectors for reserved sectors, FATs and root cluster... 	
00000203	166.16058350	[10968] 18:25:41.497 22   AudioDeviceSession OnStateChanged Inactive *SystemSounds {0.0.0.00000000}.{1edb2e1d-33e2-4078-a280-c16f75f4f850}|#%b{A9EF3FD9-4240-455E-A4D5-F2B3301887B2}|1%b# 	
00000204	166.86791992	[1632] 	
00000205	166.86791992	[1632] *** Status originated: -1073741811	
00000206	166.86791992	[1632] ***   Source File: onecore\com\netfx\windowsbuilt\iso_legacy\base\isolation\hier_hierarchy.cpp, line 230	
00000207	166.86791992	[1632] 	
00000208	166.87394714	[1632] 	
00000209	166.87394714	[1632] *** Status originated: -1073741811	
00000210	166.87394714	[1632] ***   Source File: onecore\com\netfx\windowsbuilt\iso_legacy\base\isolation\hier_hierarchy.cpp, line 230	
00000211	166.87394714	[1632] 	
00000212	166.90809631	[1632] 	
00000213	166.90809631	[1632] *** Status originated: -1073741811	
00000214	166.90809631	[1632] ***   Source File: onecore\com\netfx\windowsbuilt\iso_legacy\base\isolation\hier_hierarchy.cpp, line 230	
00000215	166.90809631	[1632] 	
00000216	166.91104126	[1632] 	
00000217	166.91104126	[1632] *** Status originated: -1073741811	
00000218	166.91104126	[1632] ***   Source File: onecore\com\netfx\windowsbuilt\iso_legacy\base\isolation\hier_hierarchy.cpp, line 230	
00000219	166.91104126	[1632] 	

```

theChaosCoder, 2021-11-12:
Do you also need a log from the working non store version, for comparison?

pbatard, 2021-11-12:
I was about to reply that I can always use more logs, but you already provided it, so thanks!

For the record, I'm not seeing issues on my end using the App Store version to format a 64 GB SD card to Large FAT32 using an USB 3.0 adapter:
```
Rufus x64 v3.17.1846 (AppStore version)
Windows version: Windows 10 Pro Workstation, 64-bit (Build 19043.1348)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.06
System locale ID: 0x0809 (en-GB)
Will use default UI locale 0x0809
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Notice: Windows VDS is unavailable
Found USB 3.0 device 'Lexar microSD RDR USB Device' (05DC:B051)
Using autorun.inf label for drive H: '64 GB'
1 device found
Disk type: Removable, Disk size: 64 GB, Sector size: 512 bytes
Cylinders: 7783, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x00000000
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 59.6 GB (64020807680 bytes)
  Start Sector: 2048, Boot: No

Format operation started
Requesting disk access...
Will use 'H:' as volume mountpoint
Warning: Could not delete partition(s): VDS is not available
Opened \\.\PhysicalDrive9 for shared write access
Requesting logical volume handle...
I/O boundary checks disabled
Analyzing existing boot records...
Drive has a Zeroed Master Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 2048 sectors
Initializing disk...
Partitioning (MBR)...
● Creating Main Data Partition (offset: 1048576, size: 59.6 GB)
Closing existing volume...
Waiting for logical drive to reappear...
Formatting (Large FAT32)...
Opened \\?\Volume{7a6e46ce-43c4-11ec-8cff-f0038c9a2e6f} for exclusive write access
Size : 59.6 GB 125040640 sectors
Cluster size 32768 bytes, 512 bytes per sector
Volume ID is 190c:f53
32 Reserved sectors, 15261 sectors per FAT, 2 FATs
1953282 Total clusters
1953281 Free clusters
Clearing out 30618 sectors for reserved sectors, FATs and root cluster...
Initializing reserved sectors and FATs...
FAT #0 sector at address: 32
FAT #1 sector at address: 15293
Writing Partition Boot Record...
Using Standard FAT32 partition boot record
Confirmed new volume has a primary FAT32 boot sector
Setting primary FAT32 boot sector for boot...
Confirmed new volume has a secondary FAT32 boot sector
Setting secondary FAT32 boot sector for boot...
Setting label...
Format completed.
Writing Master Boot Record...
Partition is already FAT32 LBA...
Using Zeroed MBR
Found volume \\?\Volume{7a6e46ce-43c4-11ec-8cff-f0038c9a2e6f}\
Created: H:\autorun.inf
Created: H:\autorun.ico
Successfully remounted \\?\Volume{7a6e46ce-43c4-11ec-8cff-f0038c9a2e6f}\ as H:

Found USB 3.0 device 'Lexar microSD RDR USB Device' (05DC:B051)
Using autorun.inf label for drive H: '64 GB'
1 device found
Disk type: Removable, Disk size: 64 GB, Sector size: 512 bytes
Cylinders: 7783, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x00000000
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 59.6 GB (64020807680 bytes)
  Start Sector: 2048, Boot: No
```

So, there seems to be an environmental component to this issue that's linked to your system.
All I can say, from your extra log and mine is that the crash happens between `Clearing out ##### sectors for reserved sectors, FATs and root cluster...` and `Initializing reserved sectors and FATs...`, so that means [this specific section of the code](https://github.com/pbatard/rufus/blob/366ce99712fa7239dcec2165ec3dd0890f24161e/src/format_fat32.c#L416-L422) (since we can eliminate a memory allocation issue that would have produced an error). Of course it's also possible that we crash shortly after we start to initialize the reserved sectors and FATs, but that the debug output doesn't get a chance to appear...

But the thing is, all this code does is write some sectors, which, even if we were to pass completely garbage data, I wouldn't expect to crash and there's also nothing specific about the App Store version for that section. As a matter of fact, the 3.17 App Store binary is the same as the standard 64-bit binary, so I'm going to ask you to do a few things:

1. Try the same thing with the `rufus_x64.exe` you can find in [this zip archive](https://github.com/pbatard/rufus/suites/4140327823/artifacts/106121542), since that is the exact same binary as the App Store version, except it's not digitally signed (which means that you may get a message about this exe not being an "official" one on startup, that you can safely ignore).
2. Try using the App Store to format to NTFS.
3. Try formatting a different USB device that is not an SD card connected to your _Norel System_ (`NORELSYS 1081) device, which, at this point, I am still not sure whether it is an internal device, that you just plug an SD card in, or an external USB card reader that you plug to a USB port.

You also seem to imply that this is something that started to happen after the 3.16 release. Does it mean that you have been able to format the same device to FAT32 without errors using 3.15 or earlier?

theChaosCoder, 2021-11-14:
**1.** It crashes also with rufus_x64.exe from the zip file. 
(rufus-3.17p.exe [Version 3.17 (Build 1846)] from the website has no problems. I formatted the card many time with the "website" version and it always worked. I could not trigger a single crash)

**2**. Weird thing happend. Selected format to NTFS, after some time it says "Media is write protected". I happend on both store and portable version. Removing and inserting again my USB-Card reader solved the problem. 

Another weird thing: Formating it to exFat (non bootable) seems impossible with rufus for me (both store and portable version). It waits a long time, then it is either "write protected" or it is currently in use by the explorer process . BUT if I select fat32 (portable ver) it happily formats it without complaints.  NTFS is also "buggy" for me.
Windows itself has also no trouble formating it to exfat or ntfs. 

**3.** I also happens with a different card reader :-/

TL:DR;
- FAT32 OK in portable, crash in store version
- exFAT / NTFS show sometimes "media is write protected" portable+store (but selecting FAT32 = immediately OK [only portable])
- Windows format always OK

Not sure if any of that helps you in any way ^^"

> You also seem to imply that this is something that started to happen after the 3.16 release. Does it mean that you have been able to format the same device to FAT32 without errors using 3.15 or earlier?

A couple of month ago I had this issue with v3.16-store.  The website already had v3.17 and it worked without any issues, so I just assumed that it was fixed in this version. 
This week I tried the latest store version v3.17 and same crash happend immediately. So I again went to the webiste, downloaded the latest version => no crash.

**Is there a way to test an older store version like 3.15?**

pbatard, 2021-11-15:
Thanks a lot for these tests.

It's quite strange that exFAT or NTFS would fail formatting, as we're simply invoking the Windows formatting DLL behind the scenes and not running actual Rufus code to perform the operations.

However, your testing gives us a few more things to try:

1. Can you try the same tests with the `rufus_86.exe` from the [same VS2019.zip archive as before](https://github.com/pbatard/rufus/suites/4140327823/artifacts/106121542) and let me know if it shows the same issue. This will tell us if it's a 64-bit vs 32-bit problem or if it's a MinGW vs MVSC compiler problem, since the version that works for you is the MinGW 32-bit version.
2. You indicate you tried with a different card reader, but have you also tried with a different card? Considering that I'm unable to replicate the issue, and that I am also not getting reports of crashes when formatting SD cards (even if I know that the store version is of course not as widely used as the regular version), it logically follows that there is some hardware/environmental component to this. Possibly, the card might having issues when accessed in a specific manner, that is being triggered when doing so from a 64-bit or MSVC application, which would of course create all kind of problems that would not be seen with other cards.
3. If you have the possibility, the best you could do to help with this issue would be to install Visual Studio Community edition which you can download [here](https://github.com/pbatard/rufus/issues/1597#issuecomment-673501858) since it would tell us more precisely where/why it crashes when formatting to FAT32 (and we know that the Visual Studio compiled version of Rufus should crash for 64-bit since that's the version you just tested). If you are willing to install and test with Visual Studio, this is how you should proceed:
   1. When installing Visual Studio, you should make sure that you select the _Desktop development with C++_ component
   2. When you start Visual Studio, select _Clone a repository_ and enter `https://github.com/pbatard/rufus.git` as the URL
   3. If you are using VS2022 it will ask you to upgrade the project files, which you should accept (I'll update the solution to VS2022 soon but there are a few things that I need to sort out before I can do that)
   4. After the Rufus source has been downloaded and the project created, make sure you select `Debug` and `x64` per the screenshot below:
[image]
   5. Click _Local Windows Debugger_ to build and debug Rufus. Visual Studio will prompt you to restart as administrator when it first launches Rufus, since it's an elevated app, so let it do that (and click _Local Windows Debugger_ again once it has done so).
   6. Try to format your SD card to FAT32. If the application crashes you should see some output like this:
[image]
   7. The part of interest is the _Call Stack_ window (bottom pane) that tells us what was the last line of code that was ran before the crash. In the example above, since I'm triggering a crash through a division by zero, the _Call Stack_ points directly into the Rufus code, but in your case it may point to Windows libraries, in which case you need to move up the _Call Stack_ until you are back to the Rufus source and can see the last line from our source that was invoked.
   8. When you are done invetigating the crash, just press the big red "stop" square at the top.

If you can do any of the above, that would help try to figure out what is going on.

As to your question _"Is there a way to test an older store version like 3.15?"_, I'm afraid I don't know any workaround to download old store applications. But I don't think that's needed. I simply wanted to know if recalled testing store versions that worked, so that I could get an idea if there was a specific version where a potential regression had been introduced, but it's not a big deal if you don't have any data like that.

theChaosCoder, 2021-11-15:
I feel like the micro sd-card is trolling me.... downloaded the rufus source code, made upgrade (I had to run vs2022 as admin, otherwise it wouldn't run rufus in debug mode), etc. ... and suddenly everything works now. Like a well oiled machine, even the store version. Should I be happy now? :-/

I will try again tomorrow and hope for the worst :D

theChaosCoder, 2021-11-16:
Tested today again and still no crashes. 
The only explanation I have is that there was a "bigger" windows update on my machine. So maybe it was caused by windows, maybe not, idk. All I can say is that it works reliably now

If it happens again I will try to debug in VS2020 first.
Will close this issue for now. 

Thank you for your quick response and help!! :-)

closed 2021-11-16 by theChaosCoder

pbatard, 2021-11-16:
Okay, thanks for the info.

As mentioned earlier, it certainly does look like there's an environmental component to the crashes you observe, even if the fact that it crashes with MSVC/x64 and not MinGW/x86 would make one think that it's a specific issue with the MSVC/x64 code...

But if you observe the crashes again in a consistent manner, I'll be interested to know what Visual Studio reports.