pbatard/rufus #1825: UEFI:NTFS still looks like unsigned

tavrez, 2021-11-12:
﻿

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.

Issue description
-----------------
I created a flash with new UEFI:NTFS, but neither in hyper-v or my system(asus laptop with series 7 intel CPU) I can boot it with secure mode enabled, when I disable secure boot everything goes well. Does it requires new models or extra steps?

I also have a question: If I create a MBR usb disk with 2 partitions on it, one being fat with UEFI:NTFS and one being NTFS with proper boot record on it I can boot this in both UEFI and BIOS systems without requiring CSM, but in your documentations and inside rufus app, you say CSM is necessary.

pbatard, 2021-11-12:
Sorry but I need to see the log of how you create the drive.

The log is not optional.

tavrez, 2021-11-12:
I just did that manually,
copy pasted the content of uefi-ntfs.img into a mbr drive(and another time in GPT drive to make sure).
boot process is fine when secure boot is disabled.
I wanted to use rufus on a vhdx but it doesn't show me the UEFI:NTFS option, so I manually copied

tavrez, 2021-11-12:
Ok I wiped partitions on vhdx and successfully used rufus to copy UEFI:NTFS, same result though.
Here is the log file:
```Rufus x86 v3.17.1846 (Portable)
Windows version: Windows 10 Pro, 64-bit (Build 19044.1288)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.06
System locale ID: 0x0409 (en-US)
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Notice: The ISO download feature has been deactivated because 'Check for updates' is disabled in your settings.
Found VHD device 'Microsoft Virtual Disk'
No logical drive found (unpartitioned?)
1 device found
No volume information for drive 0x82
Disk type: FIXED, Disk size: 136 GB, Sector size: 512 bytes
Cylinders: 16578, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 1
Disk GUID: {4F99765C-78FD-4101-ADF2-B21145810339}
Max parts: 128, Start Offset: 17408, Usable = 136365177344 bytes
Partition 1:
  Type: Microsoft System Reserved Partition
  Name: 'Microsoft reserved partition'
  ID: {72B4CD34-F1EE-4DAD-9D72-66518470997D}
  Size: 16.0 MB (16759808 bytes)
  Start Sector: 34, Attributes: 0x0000000000000000
Found VHD device 'Microsoft Virtual Disk'
No logical drive found (unpartitioned?)
1 device found
No volume information for drive 0x82
Disk type: FIXED, Disk size: 136 GB, Sector size: 512 bytes
Cylinders: 16578, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 1
Disk GUID: {4F99765C-78FD-4101-ADF2-B21145810339}
Max parts: 128, Start Offset: 17408, Usable = 136365177344 bytes
Partition 1:
  Type: Microsoft System Reserved Partition
  Name: 'Microsoft reserved partition'
  ID: {72B4CD34-F1EE-4DAD-9D72-66518470997D}
  Size: 16.0 MB (16759808 bytes)
  Start Sector: 34, Attributes: 0x0000000000000000

Format operation started
Requesting disk access...
No drive letter was assigned...
Will use 'F:' as volume mountpoint
Opened \\.\PhysicalDrive2 for exclusive write access
Analyzing existing boot records...
Drive has a Zeroed Master Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Initializing disk...
Partitioning (GPT)...
● Creating Main Data Partition (offset: 1048576, size: 127.0 GB)
● Creating UEFI:NTFS Partition (offset: 136364127232, size: 1 MB)
Writing UEFI:NTFS data...
Waiting for logical drive to reappear...
Closed Windows format prompt
Formatting to NTFS (using IFS)
Using cluster size: 4096 bytes
Quick format was selected
Creating file system...
Format completed.
Writing Master Boot Record...
Using Rufus protective MBR
Writing protective message SBR
Found volume \\?\Volume{d31ae11d-b6b7-4291-93ea-827ec5c71a85}\
Disabling file indexing...
Successfully remounted \\?\Volume{d31ae11d-b6b7-4291-93ea-827ec5c71a85}\ as F:
Finalizing, please wait...

Found VHD device 'Microsoft Virtual Disk'
1 device found
Disk type: FIXED, Disk size: 136 GB, Sector size: 512 bytes
Cylinders: 16578, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 2
Disk GUID: {B45511D8-FFA2-4F85-82CF-782B26470DFA}
Max parts: 128, Start Offset: 17408, Usable = 136365177344 bytes
Partition 1:
  Type: Microsoft Basic Data Partition
  Name: 'Main Data Partition'
  ID: {D31AE11D-B6B7-4291-93EA-827EC5C71A85}
  Size: 127.0 GB (136363078656 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
Partition 2 (UEFI:NTFS):
  Type: Microsoft Basic Data Partition
  Name: 'UEFI:NTFS'
  ID: {67580085-CE73-4BB8-8E89-BBED8C030884}
  Size: 1 MB (1048576 bytes)
  Start Sector: 266336186, Attributes: 0x9000000000000000
0 devices found
0 devices found
```

pbatard, 2021-11-12:
> I just did that manually,

Then this is not a Rufus issue. Please create a UEFI:NTFS drive in Rufus using the standalone UEFI:NTFS option you get in advanced mode. This is the only way I will be able to support you, as I'm not going to assume anything about whether you were able to create your UEFI:NTFS drive properly.

Note however that current UEFI:NTFS binaries are properly signed. And, __yes, we did test them in a Secure Boot environment__ (I mean, if we did screw this up and embedded unsigned image, there's little chance that you would be the first person to report it). So if they don't work on your system, either you are trying to boot on ARM-32bit (only the ARM32 binaries are unsigned) or you're doing something that breaks Secure Boot and that I have little interest in investigating.

pbatard, 2021-11-12:
> Ok I wiped partitions on vhdx and successfully used rufus to copy UEFI:NTFS, same result though.

Please test on real hardware with a real USB. I cannot vouch of how Virtual Machines handle Secure Boot.

Also, why did you cut the most important line from the log, which is the Rufus version you are using?

pbatard, 2021-11-12:
I'm going to close this issue, because it looks to me like you are deliberately withholding important data.

You are using an old Rufus version. Only 3.17 is Secure Boot compatible. And you would have been notified that you weren't using the latest if you had enabled the check for update, which is disabled.

Please stop wasting my time.

closed 2021-11-12 by pbatard