pbatard/rufus #1830: add an option to keep win11 as is without checking

Jpmmes, 2021-11-23:
﻿

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [x] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [x] I also tried one or more of the following:
  - [x] Using a different USB drive.
  - [x] Plugging the USB into a different port.
  - [x] Running Rufus on a different computer.
- [x] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
<Please describe your issue here>
I want that the .iso is written without any modification
in my case the TPM GPT checkig is already ignored
Log
---
```
<FULL LOG>
```

pbatard, 2021-11-23:
Already there.

There are 2 options: "__Standard__ Windows installation (TPM 2.0 + Secure Boot)" which uses the __UNMODIFIED__ windows installation process, and therefore __requires__ a computer to have TPM 2.0 as well a be Secure Boot compatible. And then there's the option that removes the TPM 2.0 and/or Secure Boot requirement.

I think you are getting confused by the fact that the two options mention TPM and Secure Boot, but that does not mean that Rufus does anything to the image with the first one, on the contrary.

closed 2021-11-23 by pbatard

Jpmmes, 2021-11-23:
Thanks

But how can you explain that run under VmWare the standard .iso is OK 

and not when installed thru a USB created by Rufus ?

pbatard, 2021-11-23:
> But how can you explain that run under VmWare the standard .iso is OK and not when installed thru a USB created by Rufus ?

Environment.

I can guarantee that, if you install in UEFI mode, the content is 100% identical to the one from the original ISO, so if it doesn't work, your environment is the issue, not Rufus.

Jpmmes, 2021-11-23:
Thanks again,

In fact I generate a MBR with Rufus so does that explain why the .iso is OK under VmWare ?

pbatard, 2021-11-23:
> In fact I generate a MBR with Rufus so does that explain why the .iso is OK under VmWare ?

The MBR is pretty standard, and I can't recall hearing of any failures associated with using MBRs written by Rufus.

Again, I can't vouch for whatever the environment you are trying to boot your drive from is deciding to do.
However, you may also want to read [this entry from the FAQ](https://github.com/pbatard/rufus/wiki/FAQ#Help_I_created_a_Windows_installation_drive_but_it_doesnt_boot) as if you're trying to boot a Windows installation drive that was created for BIOS/Legacy system on a UEFI-based computer, it will not work without enabling CSM. And the reverse, of trying to boot a UEFI-only Windows installation drive on a BIOS computer will also not work.

You __MUST__ know whether the type of system you are trying to boot on is BIOS based or UEFI based, and either use a drive that was created for the appropriate _Target system_ in Rufus, or use the <kbd>Alt</kbd>-<kbd>E</kbd> cheat mode, if you want a dual BIOS+UEFI bootable drive.

Jpmmes, 2021-11-23:
Thanks again and again.

In fact the MBR is taken in effect but the win11 installation detect that it is not compatible with TPM /GPT

The same iso is OK under VMWare

Win11 . 132 .iso

So I still don’t understand who is responsible …

 

You MUST know whether the type of system you are trying to boot on is BIOS based or UEFI based, and either use a drive that was created for the appropriate Target system in Rufus, or use the Alt-E cheat mode, if you want a dual BIOS+UEFI bootable drive.

pbatard, 2021-11-23:
The MBR is completely ignored by UEFI compliant systems.

Especially, if your idea is that having an MBR prevents UEFI boot, then that is wrong.
Per UEFI specs, a computer booting in UEFI mode will completely ignore any MBR botoloader, whether it exists or not, and look instead for the UEFI bootloaders residing in a compatible file system. Oh and also per specs, UEFI is designed to support __either__ MBR or GPT partition scheme, so even using MBR for partitions is not an issue.

Therefore, if you have a UEFI systems or OS that seems to complain about the presence of a MBR while booting in UEFI mode, then it is __NOT__ UEFI compliant, because it shouldn't ever look at the MBR bootloader.

Jpmmes, 2021-11-23:
Thanks for all these explanations,

Your support is awesome.

I still miss to solve my issue

Anyway

Cheers