pbatard/rufus #1831: memorize options GPT/MBR

Jpmmes, 2021-11-23:
﻿

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [x] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [x] I also tried one or more of the following:
  - [x] Using a different USB drive.
  - [x] Plugging the USB into a different port.
  - [x] Running Rufus on a different computer.
- [x] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
<Please describe your issue here>
Please memorize the previous options selected for GPT/MBR, Win11 stuff, ...
as Ctrl-T
Log
---
```
<FULL LOG>
```

pbatard, 2021-11-23:
Not gonna happen, because:

1. Rufus is not designed for Power Users who happen to create bootable drive after bootable drive, but for regular users who usually only need to create a bootable drive once, for a single installation.
2. The options are dependent on the ISO selected, so it would be quite a mess to try to preserve them between images, with the end result that it would be very confusing for users, as we might have to remove stuff that they'd expect to see, and then they would most likely complain that the options __they expected__ were not preserved.
3. See aslo [this](https://github.com/pbatard/rufus/wiki/FAQ#Why_cant_you_just_add_the_feature_I_want).

closed 2021-11-23 by pbatard

Jpmmes, 2021-11-23:
Thanks

I was just expecting the same behavior as the Ctrl-T memorization

AnyWay thanks for your answer