pbatard/rufus #1832: VOLMGR.sys BSOD during creating Windows To Go installation

Siddhant-M3, 2021-11-23:
﻿

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [ ] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [ ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.
Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
i tried creating Windows To Go install, but it BSODded when partitioning GPT. 
here is how i did it:
open rufus (latest version) , select iso file (Windows 10 1809 ISO)
select "Windows To Go" instead of "Standard Windows Installation" , Partition scheme : GPT (i enabled Dual BIOS-UEFI mode)
click on "Start" , click Ok during warning, and then when it says "Partitioning GPT" , i get a volmgr.sys BSOD
sorry, i couldn't copy log because of BSOD, and i didn't remember what the log was

also, i am not using SanDisk Cruzer Blade , or any other SanDisk , HP , Kingston , or Toshiba flash drive
i am using a Samsung MicroSD Card (16GB) inserted into MicroSD Reader

also, sorry if this information was insufficient

Log
---
```
<FULL LOG>
```

pbatard, 2021-11-23:
Well, for one thing, if you want BIOS+UEFI mode to work, then you shouldn't select GPT for partitioning, because GPT is UEFI only. So, even if you did enable dual BIOS+UEFI, by selecting GPT, then you are removing BIOS compatibility altogether.

Second, I am unable to replicate this crash using Windows 10 1809 (though if the crash occurs during partitioning, it's unlikely that the ISO has any impact on it), so I'm going to need a bit more info.

1. Can you at least provide the full log from Rufus from before you try to format the drive. Preferably, you should also enable debug enumeration with <kbd>Alt</kbd>-<kbd>.</kbd> so that I can get more information.
2. What happens if you create a UEFI:NTFS drive with GPT? You will see the UEFI:NTFS option appear in the _Boot selection_ dropdown after you click _Show advanced drive properties_.

Siddhant-M3, 2021-11-23:
i don't know why applying the .wim image is slow, i never got this slow write speed on Class 10 MicroSD Card
the write speed is usually 9-11 Mb/s when i am copying some file from C: drive to SD Card
[image]

pbatard, 2021-11-23:
Because sequential access speed (which is what manufacturers will tend to quote when claiming _"speed up to ..."_) is very different from random access speed, and while applying the image, as well as when running the OS, you may find that the random access speed of a drive with a decent sequential access, is actually disastrous.

Using an SD card will give you a very, __very__ poor Windows To Go experience, because SD cards are not designed for sequential access. For a decent Windows To Go experience, you really want to use an SSD drive in a USB 3.x enclosure, or a consumer flash drive that is targeted at random access.

As a rule of thumb, if it takes more than 15 minutes to create a Windows To Go drive, the device you use is subpar, and your  experience with running Windows will be equally subpar...

See also https://github.com/pbatard/rufus/wiki/FAQ#Why_is_the_Windows_To_Go_media_creation_process_so_slow

You can't just use any drive or SD card you have lying around, and expect a decent Windows To Go experience.

Siddhant-M3, 2021-11-24:
ok

Siddhant-M3, 2021-11-27:
i have question:
while creating bootable media, during "Copying ISO Files" , does Rufus use Sequential or Random Write?

pbatard, 2021-11-27:
That depends whether you're creating a Windows To Go or a regular installation drive.

If using Windows To Go, then Rufus applies an image, which requires a lot of random writes. This is __NOT__ something that can be changed, as Microsoft are the people who designed the structure of the Windows image format. And again, it doesn't matter, since, even if you were to use sequential write to create a Windows To Go image, using a different image format, you would still get dreadful performance if your drive is slow on random I/O because Windows needs to perform a lot of random I/O when running.

When creating an installer drive on the other hand (i.e. one we're not running Windows from), we're simply copying the files sequentially, so there Rufus uses sequential writes.

I would also still like you to answer the two questions I asked you earlier, otherwise I will just close this issue:
1. Can you at least provide the full log from Rufus from before you try to format the drive. Preferably, you should also enable debug enumeration with <kbd>Alt</kbd>-<kbd>.</kbd> so that I can get more information.
2. What happens if you create a UEFI:NTFS drive with GPT? You will see the UEFI:NTFS option appear in the _Boot selection_ dropdown after you click _Show advanced drive properties_.

Siddhant-M3, 2021-11-27:
the BSODs don't happen regularly , you can close this issue.

closed 2021-11-27 by pbatard