pbatard/rufus #1834: [Request]Ability to back up the current USB memory as a .usb file (disk image).

SysVR, 2021-11-27:
Under "Device" in [Drive Properties]
Add a backup function for the device.

It is a function to compress the entire drive and save it as a * .usb file.
This makes it easy to back up a USB memory that has multiple partitions.

Restoration can be done by loading the * .usb file from [Select].

pbatard, 2021-11-27:
The feature you want is [already there](https://github.com/pbatard/rufus/wiki/FAQ#Saving_an_existing_drive_to_VHD):

[image]

And there is __no__functional difference between an uncompressed VHD and a flat `.usb` image, as you can use both with the `dd` command if you wish.

closed 2021-11-27 by pbatard

SysVR, 2021-11-27:
There was insufficient confirmation.
I'm sorry.