pbatard/rufus #1836: [Enhancement] Support creating pure NTFS drives

PeashyScript, 2021-12-04:
﻿

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got 
Issue description
-----------------
I read and followed your instructions at https://github.com/pbatard/ffs to create FFS drivers from the `ntfs_x64.efi` and `ext2_x64.efi` files, and used UEFITool (https://github.com/LongSoft/UEFITool) to insert them into a firmware image and flash my bios with it. Now that my bios is capable of reading EXT2 and NTFS (I tried, and it actually does work) I was wondering if it would be possible to add a feature to make a pure NTFS/EXT4 usb for windows and linux respectively.

pbatard, 2021-12-04:
UEFI:NTFS does **not** impact the drive in any way, since it only creates a very small partition at the very end of the drive that's no larger than 1 MB. So there really is no point creating "pure" NTFS drives for people who are irked at the idea of having a small secondary partition. Furthermore, if you are annoyed by this, you can just format a drive to NTFS yourself (using Windows or using Rufus as non-bootable) and extract the Windows files from the ISO yourself.

Allowing the creation of Windows drives that can not be booted (by the many, many people who wouldn't have support for NTFS in their UEFI firmware) is simply a NO_GO situation, and I'd much rather annoy __you__, than annoy tons of people who may end up creating a non UEFI:NTFS drive that they won't be able to boot, and then have to spend time and effort figuring why the options they selected prevented them from booting.

In other words, I'm afraid that the request for "pure" NTFT falls in points 3 & 4 of [this relevant FAQ entry](https://github.com/pbatard/rufus/wiki/FAQ#Why_cant_you_just_add_the_feature_I_want).

As to ext4 (which by the way is not supported, only ext2/ext3 are supported, ext4 isn't), considering that Windows cannot read/write natively to such partitions, and that users would therefore not be able to see any of the file written to the drive, and be left convinced that Rufus didn't work (as evidenced by [all the links you'll find at the beginning of this other FAQ entry](https://github.com/pbatard/rufus/wiki/FAQ#Why_doesnt_Rufus_recommend_DD_mode_over_ISO_mode_for_ISOHybrid_images_Surely_DD_is_better)), this is also not a feature I am willing to consider, on account that it too would result in a lot of user confusion with regards to the state of their bootable drives, which is what I am trying to avoid.

As such, I'm afraid I have no choice but to reject this request.

closed 2021-12-04 by pbatard

PeashyScript, 2021-12-04:
Thank you for having the decency to at least respond and provide a reason. I respect your judgement.

marlop352, 2022-02-23:
Even if it's hidden behind a "Power key" or "Cheat mode" so that it doesn't impact normal users is the answer also no?
(unrelated, thanks for your effort on making secure-boot work)