pbatard/rufus #1839: Showing the Download ISO drop-down only if update checks are enabled is confusing and lacks transparency

chylex, 2021-12-12:
Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
I would like to reopen #1668 which has been automatically closed due to inactivity, and ask if there are any plans to improve the UI.

While the FAQ is eventually how I found *why* there was no drop-down, I spent quite a while going on a wild goose chase after seeing screenshots of the drop-down existing, trying different versions of Rufus on 3 computers, and only found the FAQ when I went to create an issue here. Well, I would still like to open an issue because this is too confusing.

To answer your points from this comment: https://github.com/pbatard/rufus/issues/1668#issuecomment-751716567

> I don't think I can just go with a short "You need to enable check for updates if you want to use this feature" popup (which is not a bad suggestion), because for the sake of transparency, I would need to inform people that the Download feature will download an run a remote PowerShell script

There is no indication in the app - including the Update Policy section - that using the Download ISO option will download and run a remote PowerShell script. The only way I know is by reading the FAQ. Right now, I don't consider the app to be transparent about what it's doing when you use this feature.

> folks who choose to disable the check for updates did so precisely because they don't want the application they run to do these kind of things

I chose to disable the update check, because I'm downloading a portable version of Rufus exactly when I need it, and deleting it afterwards. I'm always downloading the latest version, therefore the update check has no use for me.

I would argue that security-minded people, who wouldn't want Rufus to download and run a remote script, would likely want to know about updates in case of security issues. As it stands, Rufus is basically saying "Enabling update checks means I'm ok with Rufus downloading and running remote PowerShell scripts", which I think sounds quite ridiculous.

Since the original thread has already had several suggestions proposed, I'm not going to propose any new suggestions, but I would like to ask if you're planning to implement or accept a PR for any of them, and make the situation less confusing and more transparent. Thanks.

Log
---
```
Rufus x86 v3.17.1846 (Portable)
Windows version: Windows 11 Pro, 64-bit (Build 22000.194)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.06
System locale ID: 0x0409 (en-US)
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Notice: The ISO download feature has been deactivated because 'Check for updates' is disabled in your settings.
0 devices found
```

pbatard, 2021-12-12:
> I would like to ask if you're planning to implement or accept a PR for any of them

Not at this stage, sorry, because it seems clear to me that whatever I do with this, unless I __annoy__ Rufus users with a lengthy explanation about what the ISO download feature does on startup, some people will never be satisfied.

Please understand that the current situation is a __compromise__ between the needs/wants of the many and the needs/wants of the few, and that, despite what some have been trying to argue, there is no silver bullet.

Furthermore, as your log point out, if you do create an issue to complain about this, then you'll get to see the following message:
```
Notice: The ISO download feature has been deactivated because 'Check for updates' is disabled in your settings.
```

I am considering that this should provide enough data to put people wondering about the non appearance of the ISO download button on the right track. And the issue check list also directs to the FAQ, which has a specific entry about this.

You need to understand that this is far from the only item that people have complained to me should be explicitly clarified in the UI, without users having to look for it. But if I were to agree with all these suggestions of what the UI should and should not do to alert users to this or that specific behaviour, then I'd have to pretty much force people to read through the complete FAQ before Rufus displays its UI, to satisfy everyone's claim, and that is simply not a workable situation.

So, I have no choice but to compromise and disappoint some people like yourself, who will argue that Rufus can and should do better, because accumulating all these "betters" would lead to a convoluted UI with what the majority of users will consider as annoyances.

I very much understand that this is not the answer you were expecting for. But this is my stance as the developer of the app.

closed 2021-12-12 by pbatard

chylex, 2021-12-12:
Not even the addition to the Update Policy you suggested in the last comment?

> and add an entry about the PowerShell script download in the Update policy and settings

pbatard, 2021-12-12:
Please don't cherry pick the parts you like and remove the parts you don't when you quote someone.

> Possibly, if I were to do this properly, I might try to keep a short message and add a "Please check the 'Update policy and settings' section for more details" in the dialog, and add an entry about the PowerShell script download in the Update policy and settings, __but I see this as a lot of work for little benefit, so I don't see myself doing something like this anytime soon.__

chylex, 2021-12-12:
I didn't feel that part was necessary to copy, because (1) it's not part of the description of the suggestion, (2) since you specifically included "anytime soon", the point of reopening the discussion was to find out whether your stance has changed since you posted that reply a year ago.