pbatard/rufus #1842: Drive letter assignation is disabled (in Rufus v3.15 and later) when creating USB flash drive with a single Large FAT32 partition.

FuriousKiwi, 2021-12-21:
﻿

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
Hello!
We are using Rufus to create bootable USB flash drive with a custom Debian 9 image. This image contains a single LFAT-32 partition, with MBR, bootable both in BIOS and UEFI-CSM modes.
Since Rufus v3.15 and later we have noticed that the drive letter assignment is no longer performed after writing.
As I can see in the changelog for Rufus v3.15:
 - Prevent drive letter assignation to the UEFI:NTFS partition [pbatard/uefi-ntfs#25]

Judging by the changes in aeae66e97192a2eaa5d0a06d11a11549ab74280a, it should only take place when working with UEFI:NTFS bootloader, and since we are just DD'ing our img file to USB flash drive, this behavior is unexpected. Could you please give any information on how Rufus works in our case and how it decides whether to assign a letter or not. Thank you!

Log (Rufus 3.14)
---
We can see that the USB flash drive is remounted after writing is completed.
```
Rufus x86 v3.14.1788
Windows version: Windows 10 64-bit (Build 19042.1415)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.04
System locale ID: 0x0419 (ru-RU)
Detected installed Windows Language Pack for 0x0409 (English (English))
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found USB 3.0 device 'JetFlash Transcend 64GB USB Device' (8564:1000)
1 device found
No volume information for drive 0x81
Disk type: Removable, Disk size: 60 GB, Sector size: 512 bytes
Cylinders: 7352, Tracks per cylinder: 255, Sectors per track: 63
Partition type: SFD (Super Floppy Disk) or unpartitioned
Scanning image...
ISO analysis:
  'E:\rufus_test\live_upflash.img' doesn't look like an ISO image
Disk image analysis:
  Image has an unknown Master Boot Record
  Image is a bootable disk image
Using image: live_upflash.img (7.8 GB)

Format operation started
Requesting disk access...
Will use 'D:' as volume mountpoint
Deleting partitions:
● Partition 1 (offset: 0, size: 56.3 GB)
Opened \\.\PhysicalDrive1 for exclusive write access
Requesting logical volume handle...
+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
Remounted \\?\Volume{f4e468e9-6238-11ec-9137-3497f69dc536}\ as D:

Found USB 3.0 device 'JetFlash Transcend 64GB USB Device' (8564:1000)
1 device found
Disk type: Removable, Disk size: 60 GB, Sector size: 512 bytes
Cylinders: 7352, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x000F0300
Drive has an unknown Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 7.8 GB (8387559424 bytes)
  Start Sector: 2048, Boot: No
```

Log (Rufus 3.15)
---
No remount here.
```
Rufus x86 v3.15.1812
Windows version: Windows 10 64-bit (Build 19042.1415)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.06
System locale ID: 0x0419 (ru-RU)
Detected installed Windows Language Pack for 0x0409 (English (English))
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found USB 3.0 device 'JetFlash Transcend 64GB USB Device' (8564:1000)
1 device found
No volume information for drive 0x81
Disk type: Removable, Disk size: 60 GB, Sector size: 512 bytes
Cylinders: 7352, Tracks per cylinder: 255, Sectors per track: 63
Partition type: SFD (Super Floppy Disk) or unpartitioned
Scanning image...
ISO analysis:
  'E:\rufus_test\live_upflash.img' doesn't look like an ISO image
Disk image analysis:
  Image has an unknown Master Boot Record
  Image is a bootable disk image
Using image: live_upflash.img (7.8 GB)

Format operation started
Requesting disk access...
Will use 'D:' as volume mountpoint
Deleting partitions:
● Partition 1 (offset: 0, size: 56.3 GB)
Opened \\.\PhysicalDrive1 for exclusive write access
Requesting logical volume handle...
+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++

Found USB 3.0 device 'JetFlash Transcend 64GB USB Device' (8564:1000)
1 device found
Disk type: Removable, Disk size: 60 GB, Sector size: 512 bytes
Cylinders: 7352, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x000F0300
Drive has an unknown Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 7.8 GB (8387559424 bytes)
  Start Sector: 2048, Boot: No
```

[image]

pbatard, 2021-12-21:
Actually, the relevant ChangeLog entry from 3.15 is:
```
Prevent the creation of 'System Volume Information' on ESPs written in DD mode
```

You are writing your image in DD mode, and when doing so and trying to assign a drive letter, Windows always creates a `System Volume Information\` folder on the partitions it can mount such as ESPs.

The problem then is that distributions like Ubuntu and others, that check the image content, will report an error because they map the whole ESP to a file, that they validate the checksum of, and the minute you create a `System Volume Information\`  on that partition, even as it is a completely benign change that is triggered by Windows only, users are led to believe that Rufus is "corrupting" the data that is being written, which is of course problematic.

Therefore, we had to make the decision of no longer assigning drive letters to ISOs that are written in DD mode, in a attempt to prevent validation failures (which current Ubuntu performs by default on startup, and is therefore not something we can afford to dismiss as a a problem that would only affect a limited number of users) and which we expect to apply to more than ESPs in the long run, hence the reason why we're not limiting this to only images where we detect an ESPs (which would be tricky anyway, as all the UEFI systems I know can use a FAT partition as ESP, even if it doesn't officially declare itself as an ESP, meaning that we should expect partitions without an explicit ESP type to be used as ESPs, and figuring out a criteria to determine whether a partition should be assumed to be an ESP or not would quickly become a headache).

I'm afraid that I have no plans to change this behaviour at this stage, as doing so would result in a lot of complaints from Ubuntu users, and I'd rather inconvenience a small number of users than have Ubuntu users mistakenly believe that Rufus is not writing Ubuntu images properly in DD mode.

I may consider a cheat mode to enable remount if a large enough number of people complain, but considering that you are the first person to do so since the 3.15 release, it doesn't look to me like a lot of users are seeing it as a big problem, especially as a lot of images written in DD mode would use file systems that Windows can't mount anyway...

As such, I will close this issue.

closed 2021-12-21 by pbatard

9001, 2022-02-05:
I was about to open an issue regarding this behavior, so count this comment as a +1 :)

My usecase is a bootable FAT32 filesystem which is intended to be edited, just like a normal flashdrive. Users write a template image to a flashdrive, then immediately mount it and make changes (copy additional files, modify existing files). The image is dual-BIOS/UEFI bootable, so I am suggesting that windows-people use Rufus to preserve the BIOS boot stuff.

Currently, my workaround is telling users to "unplug the flashdrive once rufus has finished writing to it, close rufus, run the command `mountvol /r` and then reinsert it". This is probably not the best workaround, but it does avoid the issue (and I'm not much of a windows guy, so don't really know where to dig).

Assuming that no option is added to disable this feature, is there a better workaround than my current suggestion?

FuriousKiwi, 2022-02-05:
@9001 Hey :) Just in case if you need another workaround with letter assigning here is what we use:
After flashing the image with Rufus go to This PC -> Manage -> Disk Management. Select the USB Flash drive and right click on the partition, in the context menu select "Change Drive Letter and Paths". This will allow you to assign the letter to that partition on your USB Flash drive and it will appear in the Windows explorer.
This method seems a bit faster than unplugging the flash drive, running cmd and plugging it back in.

9001, 2022-02-07:
Actually -- turns out there is another utility, [USBimager](https://bztsrc.gitlab.io/usbimager/), which is a perfect fit for my usecase. And, like Rufus, it can also decompress gzip on the fly :>

That said, I'll still swear by Rufus for most other things, for example every time I need a WindowsToGo disk :+1: thanks a lot for keeping that feature alive btw!

vpeter4, 2022-03-18:
> Therefore, we had to make the decision of no longer assigning drive letters to images that are written in DD mode

Seems this tool will become useless for me. If automatic labeling doesn't work then maybe some button could be added? If I can assign letter from _Disk Management_ manually then I'm sure it can be done in code.

pbatard, 2022-03-18:
> If automatic labeling doesn't work then maybe some button could be added?

That's too expensive a change for a very small minority of users (I cannot stress that enough, because, I know it seems very counter intuitive than that it should be the case: UI changes are __VERY__ costly). The most I _might_ do is add a cheat mode, but even that is debatable, as I'm not seeing that many people complaining about the new behaviour, and I expect that those who do can sort drive letter assignation on their own.

vpeter4, 2022-03-18:
> I expect that those who do can sort drive letter assignation on their own

Why would I do this step manually if other programs can do that automatically for me. In my opinion this is step back in functionality.

> as I'm not seeing that many people complaining about the new behaviour

Until I did some deep tests I didn't even know for this new functionality. Now that I know I will not use program anymore.

And btw: this issue happens with writing CoreELEC disk images (and very probably LibreELEC images too).

pbatard, 2022-03-18:
> In my opinion this is step back in functionality.

Then complain to Ubuntu that their handling of ESP is a step back in functionality since, as soon as you assign a letter to it in Windows, their image validation check will fail.

>  this issue happens with writing CoreELEC disk images (and very probably LibreELEC images too).

Can you elaborate on the reason why you need to have a letter assigned to these images in Windows, if their purpose is to plug them on another machine? I'm actually genuinely curious about that: Why do you need to access these images from Windows after DD write has been completed? Are you adding or modifying files from these images? If so, what are these files?

vpeter4, 2022-03-18:
> Then complain to Ubuntu

Ubuntu images doesn't concern me.

> Why do you need to access these images from Windows after DD write has been completed? Are you adding or modifying files from these images? If so, what are these files?

CoreELEC disk image consists of 2 partitions: one 512 MB FAT32 and 32 MB EXT4. After image is written manual step is needed by copying one file: https://wiki.coreelec.org/coreelec:dtb

I don't see how this problem with CoreELEC is related to ESP (Efi System Partition). Should FAT partition be made by some special command? Is this wrong? https://github.com/CoreELEC/CoreELEC/blob/coreelec-20/scripts/mkimage#L80-L82

pbatard, 2022-03-18:
> Ubuntu images doesn't concern me.

It does, since the behaviour you complain about was triggered by a change in Ubuntu that results in the Ubuntu validation check failing if a drive letter is assigned to the ESP. If they hadn't made that change, I wouldn't have had to change the behaviour of Rufus. So the actual source of your complaint should be directed at Ubuntu, since I can't have a default in Rufus that results in Ubuntu users complaining that their media validation will fail and I'm afraid that there are just too many Ubuntu derivatives to be able to identify with certainty whether an image is one of them and should not be assigned a drive letter, which is why prompted me to use a blanket "no more drive letter assignation for DD images".

Note that at the time, I briefly looked at only removing drive letter assignation for images that had a formal ESP, but decided it was not worth the effort.

> I don't see how this problem with CoreELEC is related to ESP (Efi System Partition)

Can you tell me the exact CoreELEC image you downloaded? If that FAT32 partition is not an ESP, I may revisit the "only disable drive letter assignation for ESPs", though I have to point out that I may still decide that it's too much effort for too little benefit.

At any rate, I'm afraid that making sure that Ubuntu users don't run into unexpected behaviour when they write an image in DD mode comes before inconveniencing CoreElec users, and that the __only__ reason we had to change the default Rufus behaviour is because Ubuntu started mapping their ESP to a `.img` that they then validate the checksum of during boot, which will fail as soon as Windows assigns a letter to it, which, IMO, was a decision that had more to do with simplifying the life of Ununtu distro maintainers at the cost of creating issues down the line for Ubuntu users... So, again, if you are unhappy that Rufus had to change its default behaviour, you should complain to Ubuntu distro maintainers, because their decision made it very difficult for downstream utilities like Rufus to satisfy both Ubuntu and non Ubuntu users.

vpeter4, 2022-03-18:
Image used: https://github.com/CoreELEC/CoreELEC/releases/download/19.4-Matrix/CoreELEC-Amlogic-ng.arm-19.4-Matrix-Odroid_C4.img.gz

As I wrote above I can use alternative program and no changes are needed. I'm only sentimental because I was using Rufus for years. Other users are using for example balenaEtcher, ... and we all prefer something. It's personal choice.

pbatard, 2022-03-18:
> Image used: https://github.com/CoreELEC/CoreELEC/releases/download/19.4-Matrix/CoreELEC-Amlogic-ng.arm-19.4-Matrix-Odroid_C4.img.gz

Thanks. I've just tested it and I can see that the FAT partition is not an ESP, so, again, now that I have evidence of mainstream images like this, I __may__ revisit the idea of only disabling driver letter assignation to ESP partitions. I can't promise anything though.

vpeter4, 2022-03-18:
Fair enough. And at least the problem is now more clear to me.