pbatard/rufus #1843: Regression: rufus 3.17 can't read already opened images, while rufus 2.18 can

positron96, 2021-12-22:
﻿

Checklist
---------
- [ v ] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [ v ] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [ v ] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [ v ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Issue description
-----------------

So, I have an .img file that I want to write to an SD card. I launch my go-to app rufus (3.17p), but it says that it cannot read that file (something about the image being non-bootable). Ok, I've created that img myself, so maybe the file is somehow broken? But my file is mounted into my Ubuntu VM (on Virtual Box), so I can check if the partitions and everything is correct, and they are, the image looks totally fine. Fortunately, I keep by old rufus 2.18p, which opens this file without any problems, and writes it to SD card, making a fully working bootable SD card in dd mode (for raspberry pi). 
But then I look into logs of rufus 3.17 and see that it could not even open the file, not to mention scan it. So, I close the VM where the image was mounted, and rufus 3.17 now opens it no problem!

So, this is a report of:
* a regression: rufus 3 (3.17) does not like opened (locked?) files, while rufus 2 (2.18) works fine.
* an improvement: when image file cannot be opened, provide more meaningful message about it so the user can close the file or the application that is using the file.

The log contains both failed attempt to open the file and a successful one.

Log
---
```
Rufus x86 v3.17.1846 (Portable)
Windows version: Windows 10 Pro, 64-bit (Build 19043.1415)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.06
System locale ID: 0x0409 (en-US)
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found USB 3.0 device 'USB DISK 3.0 USB Device' (13FE:6300)
1 device found
Disk type: Removable, Disk size: 16 GB, Sector size: 512 bytes
Cylinders: 1927, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 1
Disk GUID: {D23458C8-79E0-477C-99EF-7FA5355EC8BF}
Max parts: 128, Start Offset: 17408, Usable = 15854434816 bytes
Partition 1:
  Type: Microsoft Basic Data Partition
  Name: 'Main Data Partition'
  ID: {6099FEA2-A0D9-47BB-83C1-234D53AD4AF0}
  Size: 14.8 GB (15853403648 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
Scanning image...
ISO analysis:
  'D:\DATA\Downloads\soft\OS images\sd-202112.img' doesn't look like an ISO image
Disk image analysis:
  Could not open image 'D:\DATA\Downloads\soft\OS images\sd-202112.img'
Disk type: Removable, Disk size: 16 GB, Sector size: 512 bytes
Cylinders: 1927, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 1
Disk GUID: {D23458C8-79E0-477C-99EF-7FA5355EC8BF}
Max parts: 128, Start Offset: 17408, Usable = 15854434816 bytes
Partition 1:
  Type: Microsoft Basic Data Partition
  Name: 'Main Data Partition'
  ID: {6099FEA2-A0D9-47BB-83C1-234D53AD4AF0}
  Size: 14.8 GB (15853403648 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
Failed to scan image
Found USB 3.0 device 'USB DISK 3.0 USB Device' (13FE:6300)
Found USB 2.0 device 'TS-RDF5 SD  Transcend USB Device' (8564:4000)
NOTE: This device is a USB 3.0 device operating at lower speed...
2 devices found
Disk type: Removable, Disk size: 16 GB, Sector size: 512 bytes
Cylinders: 1927, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 1
Disk GUID: {D23458C8-79E0-477C-99EF-7FA5355EC8BF}
Max parts: 128, Start Offset: 17408, Usable = 15854434816 bytes
Partition 1:
  Type: Microsoft Basic Data Partition
  Name: 'Main Data Partition'
  ID: {6099FEA2-A0D9-47BB-83C1-234D53AD4AF0}
  Size: 14.8 GB (15853403648 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
Found USB 3.0 device 'USB DISK 3.0 USB Device' (13FE:6300)
Found USB 2.0 device 'TS-RDF5 SD  Transcend USB Device' (8564:4000)
NOTE: This device is a USB 3.0 device operating at lower speed...
2 devices found
Disk type: Removable, Disk size: 16 GB, Sector size: 512 bytes
Cylinders: 1927, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 1
Disk GUID: {D23458C8-79E0-477C-99EF-7FA5355EC8BF}
Max parts: 128, Start Offset: 17408, Usable = 15854434816 bytes
Partition 1:
  Type: Microsoft Basic Data Partition
  Name: 'Main Data Partition'
  ID: {6099FEA2-A0D9-47BB-83C1-234D53AD4AF0}
  Size: 14.8 GB (15853403648 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
Found USB 3.0 device 'USB DISK 3.0 USB Device' (13FE:6300)
Found USB 2.0 device 'TS-RDF5 SD  Transcend USB Device' (8564:4000)
NOTE: This device is a USB 3.0 device operating at lower speed...
2 devices found
Disk type: Removable, Disk size: 16 GB, Sector size: 512 bytes
Cylinders: 1927, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 1
Disk GUID: {D23458C8-79E0-477C-99EF-7FA5355EC8BF}
Max parts: 128, Start Offset: 17408, Usable = 15854434816 bytes
Partition 1:
  Type: Microsoft Basic Data Partition
  Name: 'Main Data Partition'
  ID: {6099FEA2-A0D9-47BB-83C1-234D53AD4AF0}
  Size: 14.8 GB (15853403648 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
Scanning image...
ISO analysis:
  'D:\DATA\Downloads\soft\OS images\sd-202112.img' doesn't look like an ISO image
Disk image analysis:
  Image has an unknown Master Boot Record
  Image is a bootable disk image
Using image: sd-202112.img (6.1 GB)

```

pbatard, 2021-12-22:
I think the issue is that we changed the way we open images for reading by validating that we can open a file with the "file share read" attribute (i.e. we pass the `FILE_SHARE_READ` attribute [here](https://github.com/pbatard/rufus/blob/master/src/vhd.c#L342-L343), which is how one should proceed to read data that might already be opened by another application.

The reason we do that is that some applications may alter the content and in doing so, may want to tell other applications that they shouldn't open the data for reading because it may be modified at any time (and I guess that if you are mounting the image in Virtual Box, one can indeed expect that Virtual Box may alter the content without notice while another application is reading it). The end result of this would be exceedingly difficult to troubleshoot errors that have to do with the content being read being modified as it is written to USB (e.g. a file allocation table might be written that states a file is located at a specific address but then the VM alters the data and updates the file for a new one that starts at an address range that may overlap the original address resulting in corrupted content).

As such, for safety reason, we want to make sure that we let the OS (and the other application that may have the file open) that we plan to read the data while the file is being shared, so that, if the other application thinks that it is likely to alter the file, it would have opened it in a manner that will make trying to open it with the  `FILE_SHARE_READ` attribute fail).

So I guess this is what happens here: Virtual Box cannot guarantee that the image you mounted isn't going to be altered, therefore it opened it with attributes that make it incompatible for read file sharing, therefore when Rufus 3.x tries to open it and asks the OS if it should be okay to be read while it is being shared by 2 applications, the OS states that it's not and opening the image fails.

Of course, one could try to revert opening it without asking for read file share, but I hope the explanation above is enough to convince you that this might not be for the best, and why this behaviour was changed.

Thus, if you want to use an image that is being shared and that may be altered by an application while Rufus attempts to read it (even if from your point of view you estimate that it will not be altered) I will ask you to that you make sure that the image is unmounted from the other application first, as I don't think I want to go back to opening images nilly willy and hope that, if they are being opened elsewhere, their content will not be altered...

positron96, 2021-12-22:
Thank you for prompt response!

That sounds reasonable and I totally understand that using a file that can change on-the-fly can be disastrous. 

I'd be fine with it and wouldn't have created any issues in the first place if the error message was not so misleading:
[image]

It implies that the image is wrong and unsupported, while the image is actually fine. It may lead the user to spend time fixing the problem that does not exist (like I did). Hence I suggest that if rufus cannot open a file, it provides some more meaningful message (e.g. "cannot open file", "file locked" or something like that).

pbatard, 2021-12-22:
I can look into that, yes.

commit a47072c3 (2022-01-07): [ui] produce a more explicit error message when an image can't be opened/read

* Closes #1843
* Also publish Coverity artifacts since upload to Coverity seems to be broken atm.
closed 2022-01-07 by pbatard