pbatard/rufus #1844: Load failure:[3] unsupported

DarrenSong, 2021-12-24:
i make an win10 installation usb drive with UEFI mode by rufus.  but i boot it in my pc,it failed.
the detail see the picture
[image]

pbatard, 2021-12-24:
Can't help you until you post the __full__ log of how you created the drive with Rufus.

This was asked explicitly in the mandatory check list that you chose to ignore...

DarrenSong, 2021-12-24:
thanks a lot for your reply.
below is the full log in the process of creating drive.
Rufus x86 v3.17.1846 (Portable)
Windows version: Windows 10 Pro, 64-bit (Build 19043.1415)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.06
System locale ID: 0x0804 (zh-CN)
Will use default UI locale 0x0804
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'zh-CN'
Found USB 2.0 device 'USB 2.0 USB Flash DISK USB Device' (090C:1000)
1 device found
Disk type: Removable, Disk size: 16 GB, Sector size: 512 bytes
Cylinders: 1970, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 2
Disk GUID: {26A97183-F053-4DB6-9020-70F416BEF3BD}
Max parts: 128, Start Offset: 17408, Usable = 16206756352 bytes
Partition 1:
  Type: Microsoft Basic Data Partition
  Name: 'Main Data Partition'
  ID: {B008A8D4-A290-4433-BABC-4A0ED791FC70}
  Size: 15.1 GB (16204640256 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
Partition 2 (UEFI:NTFS):
  Type: Microsoft Basic Data Partition
  Name: 'UEFI:NTFS'
  ID: {587B38BC-E692-4478-9AB4-E41AD7EA6932}
  Size: 1 MB (1048576 bytes)
  Start Sector: 31651736, Attributes: 0x9000000000000000
Scanning image...
ISO analysis:
  Image is a UDF image
Disk image analysis:
  Image does not have a Boot Marker
ISO label: 'CPBA_X64FRE_ZH-CN_DV9'
  Detected: Windows 10 ISO (Build 19041.1)
  Size: 4.8 GB (Projected)
  Has a >4GB file
  Uses: EFI
  Uses: Bootmgr (BIOS and UEFI)
  Uses: Install.wim (version 0.13.1)

DarrenSong, 2021-12-24:
see the attachment.
[rufus.log](https://github.com/pbatard/rufus/files/7774325/rufus.log)

DarrenSong, 2021-12-24:
Hi pbatard ,
thanks for your friendly answer/
i try to make usb drive with win10 latest version just now, i can boot sussfully in my psc.
maybe the ISO file i created usb drive with last time is not correct

pbatard, 2021-12-24:
> maybe the ISO file i created usb drive with last time is not correct

That's a possibility, which is why the check list asked you to validate the SHA-1 to confirm that the ISO wasn't corrupted.
It's also possible that your USB drive is defective, which is why the check list also asks you to perform a bad blocks check.

At any rate, if it works now, I will close the issue.

closed 2021-12-24 by pbatard