pbatard/rufus #1846: Error saying not eought space and a 64 GB flash drive

Alakazella, 2022-01-01:
﻿

Checklist
---------
- [x ] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x ] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x ] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [x ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [x ] I also tried one or more of the following:
  - [x ] Using a different USB drive.
  - [x ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
Error [0x00000070] There's is not enough space on the disk ( I am using a 64 GB ubs flash drive) I tried windows 10 for a boot disk and got a error. Formatted the drive and tried windows 8.1 and got this error. 

Log
---
```
<FDisk type: Removable, Disk size: 64 GB, Sector size: 512 bytes
Cylinders: 7532, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x044F2E6D
Drive has a Syslinux Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 57.7 GB (61957210112 bytes)
  Start Sector: 2048, Boot: Yes
Checking for Rufus updates...
Checking release channel...
No new release version found.
Checking beta channel...
No new beta version found.
Downloading https://github.com/pbatard/Fido/releases/download/v1.27/Fido.ps1.lzma
File length: 15.8 KB
Successfully downloaded 'Fido.ps1.lzma'
Download signature is valid ✓
Script signature is valid ✓
Please Wait...
Querying https://www.microsoft.com/en-US/software-download/
Querying https://www.microsoft.com/en-US/api/controls/contentinclude/html?pageId=a8f8f489-4c7f-463a-9ca6-5cff94d8d041&host=www.microsoft.com&segments=software-download,windows8ISO&query=&action=getskuinformationbyproductedition&sessionId=a3cb4c66-37d3-4bff-9d9c-fc21030c96ff&productEditionId=52&sdVersion=2
Querying https://www.microsoft.com/en-US/api/controls/contentinclude/html?pageId=a224afab-2097-4dfa-a2ba-463eb191a285&host=www.microsoft.com&segments=software-download,windows8ISO&query=&action=GetProductDownloadLinksBySku&sessionId=a3cb4c66-37d3-4bff-9d9c-fc21030c96ff&skuId=WN7-00614&language=English International&sdVersion=2
Exited download script with code: 0
Downloading https://software-download.microsoft.com/pr/Win8.1_EnglishInternational_x64.iso?t=584c4a86-174d-4e08-a6db-37e63c5193c0&e=1641155236&h=3ef01602bd78ad44ea3e8ac1318956de
File length: 4 GB
Error writing file 'Win8.1_EnglishInternational_x64.iso': [0x00000070] There is not enough space on the disk.
>```

pbatard, 2022-01-01:
Your error has to do with saving the ISO and not creating the USB.

I don't know what location you selected to save your ISO, but I'm pretty sure it has less than 4 GB of space. And in case you chose the USB as your destination, __don't__. The ISO should __NOT__ be saved to the USB because that's what Rufus needs to read to create the bootable USB after it's done repartitioning and formatting it. Therefore, if you save the ISO to USB, Rufus won't be able to use it, because it will be deleted before it can be used.

In other words, please make sure that when you download Windows, you save the ISO outside of the USB drive you plan to use with Rufus, and that whatever drive you select has more than  4 GB of free space.

Alakazella, 2022-01-02:
I was trying to make to make a windows create boot tool. To reinstall windows, since my computer crashed. Some other site said to use Rufus over using Windows creation tool. The instructions aren't the best on the site. If I don't pick Disk or ISO what do I pick for boot-able media?

pbatard, 2022-01-02:
For creating a bootable media, Rufus needs a source Windows installation ISO to exist on your computer (outside of the USB).

So you would usually have downloaded that ISO (e.g. from https://www.microsoft.com/en-us/software-download) __before__ you run Rufus, so that you can then "write" the ISO to USB, which will be completely erased in the process and filled with the ISO content afterwards (which is the precise reason why you cannot store that source ISO on the USB, because we need to be able to access it to copy the files it contains after the media has been erased).

Now, for people who may not already have a Windows ISO, Rufus does have a feature, which is the one you were using when you encountered your issue, to download the ISO directly from Microsoft, just as if you had visited https://www.microsoft.com/en-us/software-download and selected the version of Windows you want. But the principle is still the same: That ISO must be downloaded and reside on your computer (outside of the USB) before Rufus can do its work.

So, there are 2 cases:
1. Either you already downloaded the Windows ISO for the version of Windows you are trying to install, without using Rufus, and then you just have to press `SELECT` in Rufus and navigate to that ISO to select it, which again, must exist outside of the USB on one of your computer's internal disks, or else Rufus won't be able to do its job.
2. Either you have not already downloaded the Windows ISO for the version of Windows you are trying to install, in which case you can do so in Rufus, by using the `DOWNLOAD` button. But there again, after you are done selecting the ISO, and press `Download` in the Windows version and language selection dialog, you must make sure that you select a location on your computer that has enough free space to store more than 4 GB of data, and that isn't the USB you want to use with Rufus since all content from it will be erase. Then, once the download is complete, the ISO will be automatically selected, as if you had just navigated to it using the `SELECT` button like in 1., and you can press `START` to begin the media creation process.

In other words, in each case, you need to pick an ISO for the source media, as Rufus won't be able to install Windows without one, and it all depends on whether you __already__ have that ISO somewhere on your computer before you start Rufus (in which case you just select your ISO with the `SELECT` button and then press `START`) or you don't already have that ISO downloaded, in which case you use the `DOWNLOAD` button, select the ISO you want to download in the dialog, wait for it to download, and then press `START`. Oh, and if using the second option, once you have downloaded the ISO in Rufus (and can remember where you put it) you can just use `SELECT` if you want to use Rufus with it again, as you obviously only need to download it once.

I hope that makes things a bit clearer for you on how you should proceed, the most important things being that you shouldn't have the ISO file on the USB, because the USB will be erased, and you need to make sure that, if/when you download the ISO outside of the USB, you have enough space to save it.

pbatard, 2022-01-04:
In the absence of other data, I'm going to consider that your original issue was that you tried to download the Windows ISO onto a USB drive that had less than 4 GB of space, and therefore, I will consider it resolved.

closed 2022-01-04 by pbatard