pbatard/rufus #1847: Rufus is not able to write

Advik-B, 2022-01-04:
﻿

Checklist
---------
- [x ] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x ] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x ] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [ x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ x] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ x] Plugging the USB into a different port.
  - [ x] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------

Log
---
```

Format operation started
Requesting disk access...
Will use 'H:' as volume mountpoint
Warning: Could not delete partition(s): [0x800701B1] A device which does not exist was specified.
Opened \\.\PhysicalDrive2 for shared write access
Requesting logical volume handle...
Could not get Disk Extents: [0x000001B1] A device which does not exist was specified.
No logical drive found (unpartitioned?)
Could not unmount drive: [0x00000006] The handle is invalid.
Trying to continue regardless...
Analyzing existing boot records...
read_sectors: Read error [0x00000037] The specified network resource or device is no longer available.
  Read: 0, Expected: 512
  StartSector: 0x00000000, nSectors: 0x1, SectorSize: 0x200
Drive does not have a Boot Marker
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Write error [0x00000037] The specified network resource or device is no longer available.
Retrying in 5 seconds...
Write error [0x00000037] The specified network resource or device is no longer available.
Retrying in 5 seconds...
Write error [0x00000037] The specified network resource or device is no longer available.
Retrying in 5 seconds...
Write error [0x00000037] The specified network resource or device is no longer available.
Could not reset partitions
Re-mounted volume as H: after error

Found USB 2.0 device 'SanDisk Ultra USB Device' (0781:5590)
NOTE: This device is a USB 3.0 device operating at lower speed...
1 device found
No volume information for drive 0x82
Disk type: Removable, Disk size: 32 GB, Sector size: 512 bytes
Cylinders: 3781, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x02216F87
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 29.0 GB (31103909888 bytes)
  Start Sector: 2048, Boot: No
Found USB 2.0 device 'SanDisk Ultra USB Device' (0781:5590)
NOTE: This device is a USB 3.0 device operating at lower speed...
1 device found
No volume information for drive 0x82
Disk type: Removable, Disk size: 32 GB, Sector size: 512 bytes
Cylinders: 3781, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x02216F87
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 29.0 GB (31103909888 bytes)
  Start Sector: 2048, Boot: No

```

pbatard, 2022-01-04:
1. You didn't copy the full log, so I am missing a lot of important data that I need to try to help you.
2. Could I please see a log with the output from a bad blocks check? You indicate that you did run one, so I assumed it passed. Therefore I would be very interested in seeing a log with the output from your bad blocks check.
3. Nonetheless, all the errors I see seem to point to a hardware issue. What happens when you try to format your drive in Windows (i.e. not using Rufus)?

Advik-B, 2022-01-04:
Hello sorry for wasting your time
I tried it on another PC and it worked
I think something is wring with my windows-11

closed 2022-01-04 by Advik-B