pbatard/rufus #1848: RUFUS  Y USB

aldoeri, 2022-01-08:
CHINGUEN A SU PUTA MADRE PINCHE PROGRAMA QUE SOLO CHINGA USB YA HICELA PRUEBA CON 2MEMORIAS DE DISTINTAS COMPAÑIAS CHINGUEN A SU MADRE PUTO PROGRAMA DE MIERDA

aldoeri, 2022-01-08:
FUCK YOU

pbatard, 2022-01-08:
Once you have calmed down, maybe you'll want to read [this](https://github.com/pbatard/rufus/wiki/FAQ#Help_Rufus_damaged_my_flash_drive). Or maybe even [this](https://github.com/pbatard/rufus/wiki/FAQ#Help_how_do_I_restore_my_USB_to_how_it_was_before_I_created_a_bootable_drive).

But then again, since your goto action for an application that you got entirely for free is to insult its developer because you are (wrongly) assuming that the situation you are experiencing can only be binary (i.e. the developer _must_ have done something wrong and there can be no other possible explanation besides this), I'll go on a limb and assume that it'll be difficult to try to reason with you. Therefore I will ban you and lock this thread.

closed 2022-01-08 by pbatard