pbatard/rufus #1850: Jay

JazzyJazzjazzy, 2022-01-11:
Good day,

I have an issue where my device requires the use of a Version 1.0 USB Pen drive 

Requirement is also that it be formatted in FAT using a Windows 7 computer and not a Windows 10 computer.

My question is WHAT IS THE DIFFERENCE in the formatting styles of Windows 7 versus Windows 10?

If there is a difference in using Windows 7 versus Windows 10 formatting, then can I use RUFUS to emulate the formatting style of Windows 7 on a Windows 10 computer?

Does Windows 7 make the USB drive bootable by Default when formatting Using FAT? 

Using  RUFUS, would I make the device BOOTABLE?  

How would I proceed to emulate the creation of the drive on Windows 10 to look and behave exactly as if it had been formated in FAT on a Windows 7 machine?

Thank you.

pbatard, 2022-01-13:
> Requirement is also that it be formatted in FAT using a Windows 7 computer and not a Windows 10 computer.

Sorry, but this sounds like pure bullshit. I'm pretty confident that, since the FAT32 file system has been fairly well established for years prior to the release of Windows 7 and Windows 8, and there's really very little that can be modified without breaking the whole thing, there's zero difference between creating a FAT32 file system in Windows 7 and Windows 10.

> My question is WHAT IS THE DIFFERENCE in the formatting styles of Windows 7 versus Windows 10?

Most likely none. But it's not me you should ask, it's Microsoft.

> Does Windows 7 make the USB drive bootable by Default when formatting Using FAT?

No.

> Using RUFUS, would I make the device BOOTABLE?

Yes, but you need to use a bootable source (such as an ISO or a dd image). Formatting to FAT alone will not make the drive bootable.

> How would I proceed to emulate the creation of the drive on Windows 10 to look and behave exactly as if it had been formated in FAT on a Windows 7 machine?

I think the question you really need to ask yourself is: _"On what grounds do I believe that I need to format to FAT using a Windows 7 system instead of a Windows 10 one?"_ Because, most likely, you have been jumping to one conclusion (when I formatted from Windows 7 it worked, but not from Windows 10) and __assuming__ that it was due to the FAT filesystem, when it's most likely due to another parameter you haven't thought of yet.

In other words, unless you have verifiable evidence that Windows 10 and Windows 7 introduce differences when creating FAT file systems, I'm really not interested in answering a question that, in essence, has nothing to do with Rufus (since Rufus merely uses the Windows formatting calls to create a FAT file system anyway)...

closed 2022-01-13 by pbatard

JazzyJazzjazzy, 2022-01-13:
Good day Sir,

Thank you for your kind reply. I thank you for your efforts in developing and maintaining RUFUS.

I offer you a more complete explanation in order that you do not think "... this sounds like pure bullshit.".

I am a Gulfstream jet pilot. The "device" in question is a *Honeywell DL-950 data loader used to update our navigational database once a month. The aircraft is from the 1980's and has older equipment onboard.

The newer aircraft can use USB 2.0, 3.0, and 3.1 versions but this unit uses version 1.0 only due to its age. Yes, we can buy an updated unit that is more capable, but the owner is reluctant to spend the $60,000 required for the updated unit.

If you can believe it, some aircraft are still using FLOPPY DISKS to update their databases.

None the less, your answers have been very useful to me. Thank you, Sir, and All the Best!

Respectfully,
J.

*https://www.seaerospace.com/sales/product/Honeywell/DL-950/G7226-01

[Honeywell.pdf](https://github.com/pbatard/rufus/files/7859763/Honeywell.pdf)