pbatard/rufus #1853: Something happened both my new and old usb and they aren't being shown anymore

wasi-master, 2022-01-13:
Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [ ] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [ ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [x] Using a different USB drive.
  - [x] Plugging the USB into a different port.
  - [x] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
So I have checked other issues but nowhere did I find a solution on how to fix the issues.

So In both cases I tried making my usb a bootable drive and it went to the deleting partitions part succesfully. but then it showed an error (I forgot it sadly) and after that I can't seem to find the drives in windows explorer.

Although it is being shown in disk manager but I can't do anything to it
[image]
Every option is greyed out other than properties
[image]
Even clicking properties doesn't work
[image]

I read everywhere rufus doesn't kill drives but in my case I am just dumbfounded on what happened. when it first happened I thought the drive was faulty since I read that somewhere and so I went out and bought a new one thinking it would work. but alas, the new one also had the same issue and the drive isn't showing in windows explorer

Now I think rufus didn't really kill the drive since it is shown in disk manager but I belive it deleted the partition, at this point what can I do to re-create the partition? and if the partition isn't/wasn't deleted then why isn't it being shown in Windows Explorer

Currently re-trying raises this error:
[image]

I did record the log for this error but not the other ones

Log
---
The log below is useless to be honest since it was recorded after the first try when the error actually happened. Now trying just shows a error 
```
Rufus x86 v3.17.1846
Windows version: Windows 10 Pro, 64-bit (Build 19044.1415)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.06
System locale ID: 0x0409 (en-US)
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found USB 2.0 device 'Generic Flash Disk USB Device' (058F:6387)
Could not get Disk Extents: [0x00000001] Incorrect function.
No logical drive found (unpartitioned?)
1 device found
Could not get Disk Extents: [0x00000001] Incorrect function.
No volume information for drive 0x83
Disk type: Removable, Disk size: 16 GB, Sector size: 512 bytes
Cylinders: 1988, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 1
Disk GUID: {02A13E6F-1124-4472-B87B-4E5783C461E5}
Max parts: 128, Start Offset: 17408, Usable = 16357751296 bytes
Partition 1:
  Type: Microsoft Basic Data Partition
  Name: 'Basic data partition'
  ID: {F7194811-14FC-4CA8-A902-C0156768F9C5}
  Size: 15.2 GB (16355688448 bytes)
  Start Sector: 2048, Attributes: 0x8000000000000000
1 device found
Could not get Disk Extents: [0x00000001] Incorrect function.
No volume information for drive 0x83
Disk type: Removable, Disk size: 16 GB, Sector size: 512 bytes
Cylinders: 1988, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 1
Disk GUID: {02A13E6F-1124-4472-B87B-4E5783C461E5}
Max parts: 128, Start Offset: 17408, Usable = 16357751296 bytes
Partition 1:
  Type: Microsoft Basic Data Partition
  Name: 'Basic data partition'
  ID: {F7194811-14FC-4CA8-A902-C0156768F9C5}
  Size: 15.2 GB (16355688448 bytes)
  Start Sector: 2048, Attributes: 0x8000000000000000
Scanning image...
ISO analysis:
  Image is a UDF image
Disk image analysis:
  Image does not have a Boot Marker
ISO label: 'CCCOMA_X64FRE_EN-GB_DV9'
  Detected: Windows 11 ISO (Build 22000.194)
  Size: 5.2 GB (Projected)
  Has a >4GB file
  Uses: EFI
  Uses: Bootmgr (BIOS and UEFI)
  Uses: Install.wim (version 0.13.1)
Using image: Win11_EnglishInternational_x64v1.iso (5.2 GB)
Search for conflicting processes was interrupted due to timeout

Format operation started
Requesting disk access...
Could not lock access to \\.\PhysicalDrive3: [0x00000005] Access is denied.
Search for conflicting processes was interrupted due to timeout

Found USB 2.0 device 'Generic Flash Disk USB Device' (058F:6387)
Could not get Disk Extents: [0x00000001] Incorrect function.
No logical drive found (unpartitioned?)
1 device found
Could not get Disk Extents: [0x00000001] Incorrect function.
No volume information for drive 0x83
Disk type: Removable, Disk size: 16 GB, Sector size: 512 bytes
Cylinders: 1988, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 1
Disk GUID: {02A13E6F-1124-4472-B87B-4E5783C461E5}
Max parts: 128, Start Offset: 17408, Usable = 16357751296 bytes
Partition 1:
  Type: Microsoft Basic Data Partition
  Name: 'Basic data partition'
  ID: {F7194811-14FC-4CA8-A902-C0156768F9C5}
  Size: 15.2 GB (16355688448 bytes)
  Start Sector: 2048, Attributes: 0x8000000000000000

```

pbatard, 2022-01-13:
If you purchases a drive by the same vendor, it's possible that you were sold "fake drives" (a.k.a. drives that have been modified to report more capacity than they already have, such as a 4 GB drive that has been altered to pretend it has 64 GB capacity).

Unfortunately these drives are still relatively common, especially with cheap drives, and they tend to fail in pretty much the manner you describe as soon as you get over the actual capacity, because then the flash area that is reserved for the USB functions gets overwritten and the drive becomes unrecoverable.

Still, the fact that you are getting access denied could also mean that there is software on your machine that is preventing Rufus from accessing the drive. This is also fairly common, as you can see from the [list of Software that is incompatible with Rufus](https://github.com/pbatard/rufus/wiki/FAQ#Incompatible_Software), so I will invite you to read [the FAQ entry about `Error [0x00000005] Access is denied`](https://github.com/pbatard/rufus/wiki/FAQ#Error_0x00000005_Access_is_denied) and see if you can figure what software is preventing access.

If you have a Linux machine, I would also invite you to try to repartition and reformat the drive there, as it will tell you for sure if it's due to software running on Windows that is preventing access (in which case you will be able to repartition/reformat from Linux), or if your drive is defective (in which case you won't).

wasi-master, 2022-01-13:
Thanks for the quick response.

The drives weren't by the same vendor and they did have the capacity written on them, I tested it myself. I have seen the links you mentioned previously and I don't have any of them as far as I am aware.

I could try to do it in another pc that is father's and doesn't have anything installed on it so those incompatible softwares wouldn't be an issue. I'll try to do that, although I don't really know how to re-format/re-partition a disk so I'll look up some tutorials

pbatard, 2022-01-13:
> Yeah I did that and the drive works now

Good. Then it means you have software on your system (most likely a security solution like an AntiVirus) that is preventing Rufus from accessing the drive. You will need to find out what that is.

> should I use rufus on it again from my father's pc? or is my drive faulty?

Your drive doesn't look faulty if you could recover it, and I suspect Rufus will work on that other PC, as it probably doesn't have the problematic software installed. So just try it and see what happens.

wasi-master, 2022-01-13:
Rufus seems to work on that other PC. at least it went farther than this one did. It's creating file system now which it failed to do on this PC. I'll just wait

Oh and another thing, I just checked the other pc and it has Kaspersky Antivirus (not listed as incompatible) but it seems like it isn't causing any issues just so you know.

wasi-master, 2022-01-13:
Okay so the bootable media is created. It only took 1 hour 46 minutes 😅

closed 2022-01-13 by wasi-master