pbatard/rufus #1856: Software vulnerabilities detected during code analysis with ESBMC-WR tool

janislley, 2022-01-21:
Hello,

3 potential software vulnerabilities were found in code.
To identify this kind of vulnerabilities I used tool ESBMC-WR: https://github.com/thalestas/esbmc-wr

More about the tool: https://arxiv.org/pdf/2102.02368.pdf

**Expected behavior:**

Our main objective was to check memory safety
properties (e.g., pointer dereference and memory leaks) while
performing the verification code.

**Please, check the logs of analysis:**

Issue 1
-------

[FILE] src/re.c

State 5 file re.c line 269 function re_print thread 0
----------------------------------------------------
Violated property:
file re.c line 269 function re_print
array bounds violated: array `types' upper bound
(signed long int)(pattern + (signed long int)i)->type < 17

Issue 2
-------

[FILE] src/ext2fs/hashmap.c

State 4 file hashmap.c line 51 function ext2fs_hashmap_add thread 0
----------------------------------------------------
Violated property:
file hashmap.c line 51 function ext2fs_hashmap_add
division by zero
h->size != 0

Issue 3
-------

[FILE] src/ext2fs/hashmap.c

State 8 file hashmap.c line 99 function ext2fs_hashmap_free thread 0
----------------------------------------------------
Violated property:
file hashmap.c line 99 function ext2fs_hashmap_free
dereference failure: invalid pointer

pbatard, 2022-01-21:
Thanks for the report.

I am not planning to fix any of these in Rufus on account that these should really be reported against the parent project (https://github.com/kokke/tiny-regex-c and http://e2fsprogs.sourceforge.net/ respectively) and also because:

1. `re_print()` is not used at all by Rufus, and almost certainly never will, since its purpose is to output data using `printf()`.
2. https://git.kernel.org/pub/scm/fs/ext2/e2fsprogs.git/tree/lib/ext2fs/hashmap.c#n52 and https://git.kernel.org/pub/scm/fs/ext2/e2fsprogs.git/tree/lib/ext2fs/hashmap.c#n103 are present in the parent project and should really be fixed there. Once that is done, we will naturally pick up these fixes into Rufus once we update our ext2fs dependency.

I will therefore expect you to report these software vulnerabilities (that, I have to stress out, are not vulnerabilities that an attacker can exploit to compromise a user's system through Rufus) to the parent projects, as fixing them downstream will not accomplish much and will still leave other projects, that are dependent on those, vulnerable.

closed 2022-01-21 by pbatard

janislley, 2022-01-24:
Thank you @pbatard 
We are in contact directly with tyne-regex and ext2fe projects.