pbatard/rufus #1859: Feature request: Add crossplatform RUFUS(Linux/Mac)

avosa, 2022-01-27:


avosa, 2022-01-27:
This will increase usage and availability.

JonnyTech, 2022-01-27:
https://github.com/pbatard/rufus/wiki/FAQ#Do_you_plan_to_port_Rufus_to_LinuxMac_OSSome_other_OS

pbatard, 2022-01-27:
It seems you are under the impression that my time is unlimited and free, and that I somehow have nothing better to do than port Rufus to platforms where I'm going to have to write/reinvent A LOT of stuff that comes naturally from the Windows operating system.

The thing is, I am not looking for new projects to take on, because I already have more than enough stuff lined up, improving the Windows version of Rufus as well as working on many other project, to last __years__, full time.

If you want to see a Linux/MacOS version, then by all means, please invest your __own__ time doing it, as I believe this is the only way you (and all the others who are adamant that I'm going to reap the benefits tenfold from doing it) will understand that it's far from being the sure shot walk in the park you believe it is.

Also, please don't open duplicate issues, or I will have to take action.

closed 2022-01-27 by pbatard

avosa, 2022-01-28:
Gotcha brother. Thanks for helping make sense.