pbatard/rufus #1860: Rufus not appearing on screen

Empress42, 2022-01-28:
﻿

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
After starting the executable, Rufus doesn't show up. 
When starting the application without the rufus.ini file, only the update prompt shows up, else nothing shows up at all. In both cases it shows up in Task Manager. Same results after changing the title of the executable to the non-portable version. 
The logs are made with DebugView. 

Log
---
```
00000001	0.00000000	[13000] *** Rufus init *** 	
00000002	0.00804580	[13000] Binary executable is signed by 'Akeo Consulting' 	
00000003	0.00812500	[13000] Will use settings from INI file 	
00000004	0.05008280	[13000] loc file not found in current directory - embedded one will be used 	
00000005	0.05081030	[13000] localization: extracted data to 'C:\Users\Alex\AppData\Local\Temp\Ruf61CD.tmp' 	
00000006	0.05101810	[13000] localization: found locale 'en-US' 	
00000007	0.05174920	[13000] localization: found locale 'ar-SA' 	
00000008	0.05235460	[13000] embedded.loc(408): the version of this translation is older than the base one and may result in some messages not being properly translated.	
00000009	0.05235460	[13000] If you are the translator, please update your translation with the changes that intervened between v3.5 and v3.14.	
00000010	0.05235460	[13000] See https://github.com/pbatard/rufus/blob/master/res/loc/ChangeLog.txt 	
00000011	0.05261800	[13000] localization: found locale 'bg-BG' 	
00000012	0.05263030	[13000] embedded.loc(762): the version of this translation is older than the base one and may result in some messages not being properly translated.	
00000013	0.05263030	[13000] If you are the translator, please update your translation with the changes that intervened between v3.5 and v3.14.	
00000014	0.05263030	[13000] See https://github.com/pbatard/rufus/blob/master/res/loc/ChangeLog.txt 	
00000015	0.05303750	[13000] localization: found locale 'zh-CN' 	
00000016	0.05350130	[13000] localization: found locale 'zh-TW' 	
00000017	0.05383140	[13000] localization: found locale 'hr-HR' 	
00000018	0.05428410	[13000] localization: found locale 'cs-CZ' 	
00000019	0.05431340	[13000] embedded.loc(2219): the version of this translation is older than the base one and may result in some messages not being properly translated.	
00000020	0.05431340	[13000] If you are the translator, please update your translation with the changes that intervened between v3.5 and v3.14.	
00000021	0.05431340	[13000] See https://github.com/pbatard/rufus/blob/master/res/loc/ChangeLog.txt 	
00000022	0.05466660	[13000] localization: found locale 'da-DK' 	
00000023	0.05500570	[13000] localization: found locale 'nl-NL' 	
00000024	0.05534570	[13000] localization: found locale 'fi-FI' 	
00000025	0.05568770	[13000] localization: found locale 'fr-FR' 	
00000026	0.05604520	[13000] localization: found locale 'de-DE' 	
00000027	0.05637650	[13000] localization: found locale 'el-GR' 	
00000028	0.05649240	[13000] embedded.loc(4368): the version of this translation is older than the base one and may result in some messages not being properly translated.	
00000029	0.05649240	[13000] If you are the translator, please update your translation with the changes that intervened between v3.5 and v3.14.	
00000030	0.05649240	[13000] See https://github.com/pbatard/rufus/blob/master/res/loc/ChangeLog.txt 	
00000031	0.05700900	[13000] localization: found locale 'he-IL' 	
00000032	0.05737250	[13000] localization: found locale 'hu-HU' 	
00000033	0.05739630	[13000] embedded.loc(5089): the version of this translation is older than the base one and may result in some messages not being properly translated.	
00000034	0.05739630	[13000] If you are the translator, please update your translation with the changes that intervened between v3.5 and v3.14.	
00000035	0.05739630	[13000] See https://github.com/pbatard/rufus/blob/master/res/loc/ChangeLog.txt 	
00000036	0.05775510	[13000] localization: found locale 'id-ID' 	
00000037	0.05809770	[13000] localization: found locale 'it-IT' 	
00000038	0.05843300	[13000] localization: found locale 'ja-JP' 	
00000039	0.05880710	[13000] localization: found locale 'ko-KR' 	
00000040	0.05914780	[13000] localization: found locale 'lv-LV' 	
00000041	0.05948900	[13000] localization: found locale 'lt-LT' 	
00000042	0.05982980	[13000] localization: found locale 'ms-MY' 	
00000043	0.05991980	[13000] embedded.loc(7622): the version of this translation is older than the base one and may result in some messages not being properly translated.	
00000044	0.05991980	[13000] If you are the translator, please update your translation with the changes that intervened between v3.5 and v3.14.	
[... 19 lines cut ...]
00000064	0.06306910	[13000] See https://github.com/pbatard/rufus/blob/master/res/loc/ChangeLog.txt 	
00000065	0.06346590	[13000] localization: found locale 'sr-SP' 	
00000066	0.06383800	[13000] localization: found locale 'sk-SK' 	
00000067	0.06421250	[13000] localization: found locale 'sl-SI' 	
00000068	0.06454580	[13000] localization: found locale 'es-ES' 	
00000069	0.06488160	[13000] localization: found locale 'sv-SE' 	
00000070	0.06531490	[13000] localization: found locale 'th-TH' 	
00000071	0.06571510	[13000] localization: found locale 'tr-TR' 	
00000072	0.06605370	[13000] localization: found locale 'uk-UA' 	
00000073	0.06642350	[13000] localization: found locale 'vi-VN' 	
00000074	0.06644710	[13000] embedded.loc(13386): the version of this translation is older than the base one and may result in some messages not being properly translated.	
00000075	0.06644710	[13000] If you are the translator, please update your translation with the changes that intervened between v3.5 and v3.14.	
00000076	0.06644710	[13000] See https://github.com/pbatard/rufus/blob/master/res/loc/ChangeLog.txt 	
00000077	0.06701410	[13000] Note: UAC is active 	
00000078	0.07635270	[13000] localization: initializing default message table 	
00000079	11.85397053	[13000] Rufus x86 v3.17.1846 (Portable) 	
00000080	11.85823631	[13000] Windows version: Windows 10 Pro, 64-bit (Build 19044.1466) 	
00000081	11.85946465	[13000] Syslinux versions: 4.07/2013-07-25, 6.04/pre1 	
00000082	11.86040020	[13000] Grub versions: 0.4.6a, 2.06 	
00000083	11.86120224	[13000] System locale ID: 0x0809 (en-GB) 	
00000084	11.86186028	[13000] Will use default UI locale 0x0809 	
00000085	11.86186028	[13000] SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E 	
00000086	11.86186028	[13000] Localization set to 'en-US' 	

```

pbatard, 2022-01-30:
Are you using [Sophos Antivirus] by any chance](https://github.com/pbatard/rufus/issues/1597#issuecomment-723462063)?

This is a common issue with people using Sophos, because Sophos prevents Rufus from accessing the drive in a timely manner.

Empress42, 2022-01-30:
No, I'm not running any antivirus software and Windows Defender is turned off. 
But I actually managed to get it open a few times and it seems to be an issue with some browsers. The first time it worked was when I closed some Chrome background processes and it suddenly appeared. I tried again with no Chrome in background and it didn't work, but closing Opera GX and it's processes did. The first time Opera GX was open, though. 
I also tried with Edge, with stopped Rufus from appearing, while Firefox didn't. Is there a current issue with chromium-based browsers?
I also made some additional logs with DebugView, and they differ from the first ones. 

```
00000001	0.00000000	[2772] *** Rufus init *** 	
00000002	0.03714580	[2772] Binary executable is signed by 'Akeo Consulting' 	
00000003	0.03734870	[2772] Will use settings from INI file 	
00000004	0.03748250	[2772] found locale 'en-US' 	
00000005	0.08511450	[2772] loc file not found in current directory - embedded one will be used 	
00000006	0.08779880	[2772] localization: extracted data to 'C:\Users\Alex\AppData\Local\Temp\Ruf4122.tmp' 	
00000007	0.08784450	[2772] localization: found locale 'en-US' 	
00000008	0.08866730	[2772] localization: found locale 'ar-SA' 	
00000009	0.08873180	[2772] embedded.loc(408): the version of this translation is older than the base one and may result in some messages not being properly translated.	
00000010	0.08873180	[2772] If you are the translator, please update your translation with the changes that intervened between v3.5 and v3.14.	
00000011	0.08873180	[2772] See https://github.com/pbatard/rufus/blob/master/res/loc/ChangeLog.txt 	
00000012	0.08930790	[2772] localization: found locale 'bg-BG' 	
00000013	0.08937550	[2772] embedded.loc(762): the version of this translation is older than the base one and may result in some messages not being properly translated.	
00000014	0.08937550	[2772] If you are the translator, please update your translation with the changes that intervened between v3.5 and v3.14.	
00000015	0.08937550	[2772] See https://github.com/pbatard/rufus/blob/master/res/loc/ChangeLog.txt 	
00000016	0.08989040	[2772] localization: found locale 'zh-CN' 	
00000017	0.09043250	[2772] localization: found locale 'zh-TW' 	
00000018	0.09099630	[2772] localization: found locale 'hr-HR' 	
00000019	0.09157590	[2772] localization: found locale 'cs-CZ' 	
00000020	0.09163110	[2772] embedded.loc(2219): the version of this translation is older than the base one and may result in some messages not being properly translated.	
00000021	0.09163110	[2772] If you are the translator, please update your translation with the changes that intervened between v3.5 and v3.14.	
00000022	0.09163110	[2772] See https://github.com/pbatard/rufus/blob/master/res/loc/ChangeLog.txt 	
00000023	0.09222940	[2772] localization: found locale 'da-DK' 	
00000024	0.09296800	[2772] localization: found locale 'nl-NL' 	
00000025	0.09355600	[2772] localization: found locale 'fi-FI' 	
00000026	0.09410340	[2772] localization: found locale 'fr-FR' 	
00000027	0.09456200	[2772] localization: found locale 'de-DE' 	
00000028	0.09506300	[2772] localization: found locale 'el-GR' 	
00000029	0.09510560	[2772] embedded.loc(4368): the version of this translation is older than the base one and may result in some messages not being properly translated.	
00000030	0.09510560	[2772] If you are the translator, please update your translation with the changes that intervened between v3.5 and v3.14.	
00000031	0.09510560	[2772] See https://github.com/pbatard/rufus/blob/master/res/loc/ChangeLog.txt 	
00000032	0.09552080	[2772] localization: found locale 'he-IL' 	
00000033	0.09617320	[2772] localization: found locale 'hu-HU' 	
00000034	0.09623600	[2772] embedded.loc(5089): the version of this translation is older than the base one and may result in some messages not being properly translated.	
00000035	0.09623600	[2772] If you are the translator, please update your translation with the changes that intervened between v3.5 and v3.14.	
00000036	0.09623600	[2772] See https://github.com/pbatard/rufus/blob/master/res/loc/ChangeLog.txt 	
00000037	0.09660470	[2772] localization: found locale 'id-ID' 	
00000038	0.09696580	[2772] localization: found locale 'it-IT' 	
00000039	0.09738110	[2772] localization: found locale 'ja-JP' 	
00000040	0.09780510	[2772] localization: found locale 'ko-KR' 	
00000041	0.09821920	[2772] localization: found locale 'lv-LV' 	
00000042	0.09859900	[2772] localization: found locale 'lt-LT' 	
00000043	0.09897730	[2772] localization: found locale 'ms-MY' 	
00000044	0.09902190	[2772] embedded.loc(7622): the version of this translation is older than the base one and may result in some messages not being properly translated.	
00000045	0.09902190	[2772] If you are the translator, please update your translation with the changes that intervened between v3.5 and v3.14.	
00000046	0.09902190	[2772] See https://github.com/pbatard/rufus/blob/master/res/loc/ChangeLog.txt 	
00000047	0.09937420	[2772] localization: found locale 'nb-NO' 	
00000048	0.09976340	[2772] localization: found locale 'fa-IR' 	
00000049	0.09979590	[2772] embedded.loc(8329): the version of this translation is older than the base one and may result in some messages not being properly translated.	
00000050	0.09979590	[2772] If you are the translator, please update your translation with the changes that intervened between v3.5 and v3.14.	
[... cut ...]
00000088	0.30845821	[2772] Notice: The ISO download feature has been deactivated because 'Check for updates' is disabled in your settings. 	
00000089	17.73363686	[2772] Found USB 2.0 device 'SanDisk Cruzer Blade USB Device' (0781:5567) 	
00000090	17.74393845	[2772] 1 device found 	
00000091	17.74627495	[2772] Disk type: Removable, Disk size: 16 GB, Sector size: 512 bytes 	
00000092	17.74666595	[2772] Cylinders: 1945, Tracks per cylinder: 255, Sectors per track: 63 	
00000093	17.74700356	[2772] Partition type: SFD (Super Floppy Disk) or unpartitioned 	

```

pbatard, 2022-01-30:
> Is there a current issue with chromium-based browsers?

Not that I can see, and, as you can expect, if there was a widespread compatibility issue with these browsers, this issue tracker would be full of _"Rufus doesn't work when I have Chrome/Edge"_ running, as that would be a pretty big deal.

So, the fact that other people don't seem to report a similar problem seems to point to a specific issue with your environment.

The first thing I would check is whether you have installed browser extensions that might interfere with USB access. For instance, I know that WebUSB is becoming popular to let browsers access USB devices, and I would expect that, if you have a browser plugin that uses WebUSB, it may somehow prevent access.

Still, as far as your log shows, Rufus was able to initialize everything properly and it should be able to display its UI. As a matter of fact, some of the elements from your log, such as `Rufus x86 v3.17.1846 (Portable)` are only being generated when Windows is initializing the UI (they come from the dialog receiving a `WM_INIT` message, which is basically Windows saying _"Hey, I'm about to display the UI from your app, so if there's anything you want to do before that, I'm just letting you know"_, and there's nothing blocking that I can think of happening during the code that Rufus runs when it receives the `WM_INIT` message).

For reference, this is what I get on my machine, with a blank USB drive similar to yours, after Rufus has displayed the UI, and as you can see, the output is pretty much the same as yours, which means that all the init Rufus needed to do was completed successfully and the rest is just a matter of Windows itself displaying the UI elements.

```
00000001	0.00000000	[1128] *** Rufus init *** 	
00000002	0.01961670	[1128] Binary executable is signed by 'Akeo Consulting' 	
00000003	0.02030300	[1128] Will use settings from registry 	
00000004	0.02056630	[1128] found locale 'en-US' 	
00000005	0.14268680	[1128] loc file not found in current directory - embedded one will be used 	
00000006	0.14548600	[1128] localization: extracted data to 'C:\Users\nil\AppData\Local\Temp\Ruf7FC7.tmp' 	
00000007	0.17507111	[1128] localization: found locale 'en-US' 	
00000008	0.17913760	[1128] localization: found locale 'ar-SA' 	
00000009	0.18234210	[1128] embedded.loc(408): the version of this translation is older than the base one and may result in some messages not being properly translated.	
00000010	0.18234210	[1128] If you are the translator, please update your translation with the changes that intervened between v3.5 and v3.14.	
00000011	0.18234210	[1128] See https://github.com/pbatard/rufus/blob/master/res/loc/ChangeLog.txt 	
00000012	0.19398130	[1128] localization: found locale 'bg-BG' 	
00000013	0.19410460	[1128] embedded.loc(762): the version of this translation is older than the base one and may result in some messages not being properly translated.	
00000014	0.19410460	[1128] If you are the translator, please update your translation with the changes that intervened between v3.5 and v3.14.	
00000015	0.19410460	[1128] See https://github.com/pbatard/rufus/blob/master/res/loc/ChangeLog.txt 	
00000016	0.20437621	[1128] localization: found locale 'zh-CN' 	
00000017	0.21488219	[1128] localization: found locale 'zh-TW' 	
00000018	0.22489570	[1128] localization: found locale 'hr-HR' 	
00000019	0.23236310	[1128] localization: found locale 'cs-CZ' 	
00000020	0.23255810	[1128] embedded.loc(2219): the version of this translation is older than the base one and may result in some messages not being properly translated.	
00000021	0.23255810	[1128] If you are the translator, please update your translation with the changes that intervened between v3.5 and v3.14.	
00000022	0.23255810	[1128] See https://github.com/pbatard/rufus/blob/master/res/loc/ChangeLog.txt 	
00000023	0.23650619	[1128] localization: found locale 'da-DK' 	
00000024	0.24050240	[1128] localization: found locale 'nl-NL' 	
00000025	0.24452110	[1128] localization: found locale 'fi-FI' 	
00000026	0.24886259	[1128] localization: found locale 'fr-FR' 	
00000027	0.25292560	[1128] localization: found locale 'de-DE' 	
00000028	0.25690860	[1128] localization: found locale 'el-GR' 	
00000029	0.25703070	[1128] embedded.loc(4368): the version of this translation is older than the base one and may result in some messages not being properly translated.	
00000030	0.25703070	[1128] If you are the translator, please update your translation with the changes that intervened between v3.5 and v3.14.	
00000031	0.25703070	[1128] See https://github.com/pbatard/rufus/blob/master/res/loc/ChangeLog.txt 	
00000032	0.26104960	[1128] localization: found locale 'he-IL' 	
00000033	0.26495591	[1128] localization: found locale 'hu-HU' 	
00000034	0.26505771	[1128] embedded.loc(5089): the version of this translation is older than the base one and may result in some messages not being properly translated.	
00000035	0.26505771	[1128] If you are the translator, please update your translation with the changes that intervened between v3.5 and v3.14.	
00000036	0.26505771	[1128] See https://github.com/pbatard/rufus/blob/master/res/loc/ChangeLog.txt 	
00000037	0.26843739	[1128] localization: found locale 'id-ID' 	
00000038	0.27191359	[1128] localization: found locale 'it-IT' 	
00000039	0.27535269	[1128] localization: found locale 'ja-JP' 	
00000040	0.27912551	[1128] localization: found locale 'ko-KR' 	
[... cut ...]
So, I think it could possibly be the browser or a browser extension interfering with application display. Or maybe, since Chromium based browsers are very hungry in RAM, it's a resource issue on your machine, with Windows not having enough left to properly display other applications.

At any rate, I'm afraid it's clearly an environmental issue on your side, which you are likely to have to figure out on your own, as I really can't think of anything in Rufus that would prevent the UI from displaying, when running on a vanilla Windows 10 system with enough resources and no conflicting applications (such as Sophos) installed...

pbatard, 2022-02-11:
Since there hasn't been any update, and I've offered the only guidance I can offer (which is to figure out what, in your environment, is different from a vanilla installation of Windows and causing this problem), I'm going to close this issue.

closed 2022-02-11 by pbatard