pbatard/rufus #1861: Rufus cannot locate the image I downloaded, so I cannot select it

kavousan, 2022-01-28:
﻿

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
I downloaded an image from https://sourceforge.net/projects/systemrescuecd/files/sysresccd-x86/9.00/systemrescue-9.00-i686.iso/download

I checked its checksum manually, so the file exists.

I tried to select the image; rufus does not see it in the directory. 

I removed the "ISO Image" filter, still cannot see the file in question, and selected a random file; Rufus correctly complained.

Log
---
```
Rufus x86 v3.17.1846
Windows version: Windows 11 Core, 64-bit (Build 22000.376)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.06
System locale ID: 0x0809 (en-GB)
Will use default UI locale 0x0809
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found USB 2.0 device 'USB Flash Memory USB Device' (0930:6545)
1 device found
Disk type: Removable, Disk size: 4.1GB, Sector size: 512 bytes
Cylinders: 501, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0xF435B2EE
Drive has a Windows 95B/98/98SE/ME Master Boot Record
Partition 1:
  Type: FAT32 (0x0b)
  Size: 3.8 GB (4126130176 bytes)
  Start Sector: 32, Boot: Yes
Scanning image...
ISO analysis:
  'C:\Users\Admin\Desktop\mission christmas slide.pptx' doesn't look like an ISO image
Disk image analysis:
  Image does not have a Boot Marker
Disk type: Removable, Disk size: 4.1GB, Sector size: 512 bytes
Cylinders: 501, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0xF435B2EE
Drive has a Windows 95B/98/98SE/ME Master Boot Record
Partition 1:
  Type: FAT32 (0x0b)
  Size: 3.8 GB (4126130176 bytes)
  Start Sector: 32, Boot: Yes
Failed to scan image

```

JonnyTech, 2022-01-28:
> C:\Users\Admin\Desktop\mission christmas slide.pptx

This is **not** a Windows installer ISO image file.

kavousan, 2022-01-28:
I see on email a notification, but I cannot see it here.

Was it withdrawn?

pbatard, 2022-01-30:
> I see on email a notification, but I cannot see it here.
>
> Was it withdrawn?

Yeah, that's what I hate about GitHub, because instead of leaving a _"This comment was deleted by its user"_ so that folks reading the thread can understand that someone once replied with something that may have some had some significant impact on the discussion, even if it has since been deleted, you're just being left with absolutely __nothing__ to indicate that someone once replied.

This is very problematic because I've had a few issues where not having any idea that there once was a comment in between replies can give a completely wrong idea to people reading a thread. Often-times, the fact that there isn't even a notice of a missing comment will make whoever replied afterwards look like jerks, because there will be no hint that maybe the reason a conversation got heated was because of an inflammatory comment that was since deleted... I don't have a problem with people choosing to delete a comment if they see it as no longer appropriate (on the contrary, this is definitely something you want), but GitHub should leave an explicit notice that there was a deleted comment, instead of making it look like nothing ever happened.

Anyway, with regards to your issue, please bear in mind that Rufus runs elevated, and therefore there's no guarantee that a file that may be accessible to your (non-elevated) current user will still be accessible to the (elevated) admin. Even as one would expect the Administrator to always have access to any files, there are situations where, because of permissions, this may be the case.

The first thing I'd try then is open Notepad as Administrator, and see if you can see your ISO when trying to open it in Notepad, because my guess is that elevated Notepad will probably display the same behaviour as Rufus, since both applications use the common system dialogs to access files.

There's nothing fancy with regards to what Rufus does when selecting a file, in terms of trying to filter out specific media. Especially, the `*.*` filter should give you access to ALL the files that the current elevated user should have access to, and it's really Windows, and not us, that does the cherry-picking with regards to what it actually will list in that dialog. 

For the record, I see no issue seeing and opening `systemrescue-9.00-i686.iso` on my side:
```
Rufus x86 v3.17.1846
Windows version: Windows 10 Pro Workstation, 64-bit (Build 19044.1466)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.06
System locale ID: 0x0809 (en-GB)
Will use default UI locale 0x0809
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
0 devices found
Scanning image...
ISO analysis:
  Image is an ISO9660 image
  Will use '/isolinux/isolinux.cfg' for Syslinux
  Detected Syslinux version: 6.04/pre2-11-gbf6db5b4 (from '/isolinux/isolinux.bin')
Disk image analysis:
  Image has an unknown Master Boot Record
  Image is a bootable disk image
ISO label: 'RESCUE900'
  Size: 800.1 MB (Projected)
  Uses: Syslinux/Isolinux v6.04
  Uses: EFI
Using image: systemrescue-9.00-i686.iso (801 MB)
```

kavousan, 2022-01-30:
Many thanks for responding. I logged in as admin, copied the ISO image into its filesystem and succeeded. 

I am surprised that I did not find a previous report; either everyone knows about this, so I apologise, or everyone runs as admin. :-o 

Worth an entry in your FAQs?

closed 2022-01-30 by kavousan