pbatard/rufus #1862: "Persistent Partition Size" bar keeps disappearing and reappearing while making the window longer when the same ISO Image is selected over and over again

batu-0, 2022-01-29:
﻿

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
First of all, I'd like to thank you for creating this awesome and easy to use tool! I've been using it for a long while now. Anyways, this just seems to be an UI bug and it only happens on the alpha builds, the latest release doesn't seem to have this issue. Since I'm bad at describing, here's a short video that showcases it.

https://user-images.githubusercontent.com/76159136/151652119-0a9d0b2e-b7b0-4f04-bae8-580258649148.mp4

Log
---
```

<FULL LOG>Rufus x86 v3.18.1863 (Alpha)
Windows version: Windows 10 Home Single Language, 64-bit (Build 19044.1415)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.06
System locale ID: 0x0409 (en-US)
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Notice: The ISO download feature has been deactivated because 'Check for updates' is disabled in your settings.
Found USB 2.0 device 'General UDisk USB Device' (ABCD:1234)
1 device found
Disk type: Removable, Disk size: 8 GB, Sector size: 512 bytes
Cylinders: 979, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 1
Disk GUID: {E16A6853-36F6-4BCA-B856-4C3695917E2C}
Max parts: 128, Start Offset: 17408, Usable = 8053029376 bytes
Partition 1:
  Type: Microsoft Basic Data Partition
  Name: 'Main Data Partition'
  ID: {97522AC3-40DD-47EC-89D1-15BD6FB54B7F}
  Size: 7.5 GB (8051998208 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
Scanning image...
ISO analysis:
  Image is an ISO9660 image
  Will use '/syslinux/syslinux.cfg' for Syslinux
  Detected Syslinux version: 6.04/20200816 (from '/syslinux/isolinux.bin')
Disk image analysis:
  Image has an unknown Master Boot Record
  Image is a bootable disk image
ISO label: 'GParted-live'
  Size: 417.6 MB (Projected)
  Uses: Syslinux/Isolinux v6.04
  Uses: EFI
Using image: gparted-live-1.3.1-3-amd64.iso (419 MB)
Scanning image...
ISO analysis:
  Image is an ISO9660 image
  Will use '/syslinux/syslinux.cfg' for Syslinux
  Detected Syslinux version: 6.04/20200816 (from '/syslinux/isolinux.bin')
Disk image analysis:
  Image has an unknown Master Boot Record
  Image is a bootable disk image
ISO label: 'GParted-live'
  Size: 417.6 MB (Projected)
  Uses: Syslinux/Isolinux v6.04
  Uses: EFI
Using image: gparted-live-1.3.1-3-amd64.iso (419 MB)
Scanning image...
ISO analysis:
  Image is an ISO9660 image
  Will use '/syslinux/syslinux.cfg' for Syslinux
  Detected Syslinux version: 6.04/20200816 (from '/syslinux/isolinux.bin')
Disk image analysis:
  Image has an unknown Master Boot Record
  Image is a bootable disk image
ISO label: 'GParted-live'
  Size: 417.6 MB (Projected)
  Uses: Syslinux/Isolinux v6.04
  Uses: EFI
Using image: gparted-live-1.3.1-3-amd64.iso (419 MB)
Scanning image...
ISO analysis:
  Image is an ISO9660 image
  Will use '/syslinux/syslinux.cfg' for Syslinux
  Detected Syslinux version: 6.04/20200816 (from '/syslinux/isolinux.bin')
Disk image analysis:
  Image has an unknown Master Boot Record
  Image is a bootable disk image
ISO label: 'GParted-live'
  Size: 417.6 MB (Projected)
  Uses: Syslinux/Isolinux v6.04
  Uses: EFI
Using image: gparted-live-1.3.1-3-amd64.iso (419 MB)
Scanning image...
ISO analysis:
  Image is an ISO9660 image
  Will use '/syslinux/syslinux.cfg' for Syslinux
  Detected Syslinux version: 6.04/20200816 (from '/syslinux/isolinux.bin')
Disk image analysis:
  Image has an unknown Master Boot Record
  Image is a bootable disk image
ISO label: 'GParted-live'
  Size: 417.6 MB (Projected)
  Uses: Syslinux/Isolinux v6.04
  Uses: EFI
Using image: gparted-live-1.3.1-3-amd64.iso (419 MB)
Scanning image...
ISO analysis:
  Image is an ISO9660 image
  Will use '/syslinux/syslinux.cfg' for Syslinux
  Detected Syslinux version: 6.04/20200816 (from '/syslinux/isolinux.bin')
Disk image analysis:
  Image has an unknown Master Boot Record
  Image is a bootable disk image
ISO label: 'GParted-live'
  Size: 417.6 MB (Projected)
  Uses: Syslinux/Isolinux v6.04
  Uses: EFI
Using image: gparted-live-1.3.1-3-amd64.iso (419 MB)
Scanning image...
ISO analysis:
  Image is an ISO9660 image
  Will use '/syslinux/syslinux.cfg' for Syslinux
  Detected Syslinux version: 6.04/20200816 (from '/syslinux/isolinux.bin')
Disk image analysis:
  Image has an unknown Master Boot Record
  Image is a bootable disk image
ISO label: 'GParted-live'
  Size: 417.6 MB (Projected)
  Uses: Syslinux/Isolinux v6.04
  Uses: EFI
Using image: gparted-live-1.3.1-3-amd64.iso (419 MB)
Scanning image...
ISO analysis:
  Image is an ISO9660 image
  Will use '/syslinux/syslinux.cfg' for Syslinux
  Detected Syslinux version: 6.04/20200816 (from '/syslinux/isolinux.bin')
Disk image analysis:
  Image has an unknown Master Boot Record
  Image is a bootable disk image
ISO label: 'GParted-live'
  Size: 417.6 MB (Projected)
  Uses: Syslinux/Isolinux v6.04
  Uses: EFI
Using image: gparted-live-1.3.1-3-amd64.iso (419 MB)

```

pbatard, 2022-01-30:
Thanks for the report. I can indeed replicate the issue on my side.

If you have a chance, would it be possible for you to try to find the first version since 3.17 where this issue started to appear (by testing the builds [for each commit](https://github.com/pbatard/rufus/commits/master), which you can access by clicking the green ✓ next to it).

If not I'll figure the commit that introduce this regression myself so it's no big deal. But if you can do the above, it should help with speeding up the fix.

batu-0, 2022-01-30:
I traced it down to this commit: https://github.com/pbatard/rufus/commit/c28f9bc49184af4062cac63d9dc78450c5731418
It throws out some errors when trying to run it, but I ignored those errors and proceeded with the testing anyways. This commit seems to have caused it since the commit before this one doesn't have the issue. Hope this helps.

pbatard, 2022-01-31:
Thanks a lot for pinpointing the commit! This really helps, and I should have a fix now.

commit 036f6260 (2022-01-31): [ui] fix regression in conditional expression and use %c always

* This fixes the regression introduced in c28f9bc49184af4062cac63d9dc78450c5731418.
* 'if ((a && !b) || (!a && b))' can not always be simplified as 'if (a != b)' when the types for 'a' and 'b' are not straight booleans.
* Closes #1862
* Also drop the use of '%C' in printf() expression, as it is intended to print wide characters and not turn a char to uppercase.
closed 2022-01-31 by pbatard