pbatard/rufus #1864: FEATURE REQUEST

willzyx-hub, 2022-02-08:
Hello dev sorry for opening issue on this repository but i wanna request something can you make rufus available on Linux too ? I like this app but it is not available on Linux :( so i hope you can make an AppImage on Linux maybe

9001, 2022-02-08:
https://github.com/pbatard/rufus/wiki/FAQ#do-you-plan-to-port-rufus-to-linuxmac-ossome-other-os

depending on what features you need from Rufus, you may be able to use [USBimager](https://bztsrc.gitlab.io/usbimager/) instead. Rufus is much more powerful, but USBimager can do a lot of the basic / common stuff

9001, 2022-02-08:
I dunno dude, try it and tell us how it goes?

Just guessing but it machine *should* be fine for booting win10 on an uefi machine

pbatard, 2022-02-08:
If you want to create Windows bootable drives on Linux you should be able to use [WoeUSB](https://github.com/WoeUSB)/[WoeUSB ng](https://github.com/WoeUSB/WoeUSB-ng).

For othet images, `dd` might work.

At any rate, the question is whether I can make Rufus available to Linux, to which the answer is no, so I will close this issue.

closed 2022-02-08 by pbatard

willzyx-hub, 2022-02-09:
> If you want to create Windows bootable drives on Linux you should be able to use [WoeUSB](https://github.com/WoeUSB)/[WoeUSB ng](https://github.com/WoeUSB/WoeUSB-ng).
>
> For othet images, `dd` might [...]

Well can i compile the rufus on Linux manually then ?

gdegliesposti, 2022-02-09:
Linux platform is not supported and according to the FAQ there are no plans
to port it any non-WIndows system.
(see
https://github.com/pbatard/rufus/wiki/FAQ#do-you-plan-to-port-rufus-to-linuxmac-ossome-other-os
)

Being an open source project, you can get the sources and try to port it to
linux yourself, it would be very welcomed!

It uses a lot of window-specific functions so it looks like a huge
work, comparable to a complete rewrite.

regards