pbatard/rufus #1865: Volume ID could not be found

github-fanuser, 2022-02-08:
Issue description
-----------------
I am trying to create a bootable Windows 11 flash drive using a Win 11 iso. Rufus gets to the point of writing MBR and then it displays an error "The volume ID could not be found". I have tried 2 different usb drives with the same error occurring.  I also tried 2 different usb ports. I have rebooted the computer.  I am using a Windows 10 Pro to run Rufus. 

Here is the latest log file.

Rufus x86 v3.17.1846 (Portable)
Windows version: Windows 10 Pro, 64-bit (Build 19044.1466)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.06
System locale ID: 0x0409 (en-US)
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Notice: The ISO download feature has been deactivated because 'Check for updates' is disabled in your settings.
Found USB 2.0 device 'hp v125w USB Device' (03F0:3307)
1 device found
No volume information for drive 0x81
Disk type: Removable, Disk size: 16 GB, Sector size: 512 bytes
Cylinders: 1950, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 1
Disk GUID: {557E378D-28C6-4F9B-ABCD-1FA39A299751}
Max parts: 128, Start Offset: 17408, Usable = 16043178496 bytes
Partition 1:
  Type: Microsoft Basic Data Partition
  Name: 'Main Data Partition'
  ID: {99496243-C3F6-4200-9736-91654073BBA3}
  Size: 14.9 GB (16042147328 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
Scanning image...
ISO analysis:
  Image is a UDF image
Disk image analysis:
  Image does not have a Boot Marker
ISO label: 'ESD-ISO'
  Detected: Windows 11 ISO (Build 22000.194)
  Size: 4.2 GB (Projected)
  Note: File on disk is larger than reported ISO size by 552 KB...
  Uses: EFI
  Uses: Bootmgr (BIOS and UEFI)
  Uses: Install.esd (version 0.14.0+)
Using image: Windows11.iso (4.2 GB)

Format operation started
Requesting disk access...
Will use 'F:' as volume mountpoint
Opened \\.\PhysicalDrive1 for exclusive write access
Analyzing existing boot records...
Drive has a Zeroed Master Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Initializing disk...
Partitioning (GPT)...
● Creating Main Data Partition (offset: 1048576, size: 14.9 GB)
Waiting for logical drive to reappear...
Formatting to FAT32 (using IFS)
Using cluster size: 8192 bytes
Quick format was selected
Creating file system...
Format completed.
Writing Master Boot Record...
Using Rufus protective MBR
Writing protective message SBR
Timeout while waiting for logical drive
Could not get volume name

Found USB 2.0 device 'hp v125w USB Device' (03F0:3307)
Could not open GUID volume '\\?\Volume{03f51aca-891d-11ec-9f5a-54a050b5ba35}': [0x00000005] Access is denied.
No logical drive found (unpartitioned?)
1 device found
Could not open GUID volume '\\?\Volume{03f51aca-891d-11ec-9f5a-54a050b5ba35}': [0x00000005] Access is denied.
No volume information for drive 0x81
Disk type: Removable, Disk size: 16 GB, Sector size: 512 bytes
Cylinders: 1950, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 1
Disk GUID: {2EE9BDEC-3C90-45BE-B480-1CCC308D8FB4}
Max parts: 128, Start Offset: 17408, Usable = 16043178496 bytes
Partition 1:
  Type: Microsoft Basic Data Partition
  Name: 'Main Data Partition'
  ID: {4F65D1B6-B587-444A-B20B-AB21134CC66C}
  Size: 14.9 GB (16042147328 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000

pbatard, 2022-02-08:
Would have preferred if you followed the check list.

Have you tried a bad blocks check?

Also you may want to try <kbd>Alt</kbd>-<kbd>Z</kbd> to see how it goes.

github-fanuser, 2022-02-09:
I tried Alt-Z...no help. I tried different usb drives...same error. I also tried MBR and GPT. I also tried FAT-32 and NTFS. I get the same error.

JonnyTech, 2022-02-09:
@github-fanuser do you have any security or anti-virus software running on your computer?

github-fanuser, 2022-02-10:
I have MS Defender and Malwarebytes. I disabled them both. It didn't help.
I gave up and used Media Creation Tool to create the USB drive. Maybe
someone else will have the problem and the solution will be found. Go ahead
and close the case; or you can leave open and see if anyone else post the
problem or possible solution.

pbatard, 2022-02-11:
Yeah, I'm afraid that this issue is tied to your environment, so it's going to be very difficult to guess what's really happening, let alone propose a possible workaround.

The fact that you are getting `Could not open GUID volume '\\?\Volume{03f51aca-891d-11ec-9f5a-54a050b5ba35}': [0x00000005] Access is denied.` messages seems to imply that there is an application, external to Rufus, that is interfering with access to the drive, which is probably the reason you are getting the `Timeout while waiting for logical drive`.

Please see [the relevant FAQ entry](https://github.com/pbatard/rufus/wiki/FAQ#Error_0x00000005_Access_is_denied) (which I'm afraid only tells you that you need to identify the application that interferes with Rufus on your own) as well as [this other FAQ entry listing incompatible software](https://github.com/pbatard/rufus/wiki/FAQ#Incompatible_Software), knowing that this list is far from being exhaustive, and there exists other software, not listed there, that can have the same effect.

Considering that this looks like a purely environmental issue, that is next to impossible to troubleshoot without having access to a machine that exhibits the problem, I'm going to close this.

closed 2022-02-11 by pbatard