pbatard/rufus #1868: Any way to disable `rufus.log` file creation?

Hlsgs, 2022-02-09:
﻿

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [ ] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [ ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
<Please describe your issue here>

I have set `VerboseUpdateCheck = 0` in `rufus.ini`, yet Rufus always creates `/Rufus/rufus.log`. Any way to prevent said file creation?

pbatard, 2022-02-09:
Can you explain in what way this is a problem to you?

Also `VerboseUpdateCheck` has nothing to do with log creation. As its name indicate, this settings relates to the __update check__ not the log.

I would also have liked to see a log, as I suspect you are using the portable version, but obviously can't tell without the log.

Hlsgs, 2022-02-09:
Hi @pbatard and thank you for the quick reply.

I am indeed using the portable version and this is not a problem, I would just prefer it if Rufus didn't create said log file as it pollutes my folder structure uselessly(as far as I'm concerned). This is not a feature request, I'm just asking if there is already a way to prevent that file getting created for the portable version. If there isn't, nevermind.

pbatard, 2022-02-09:
There is no option to prevent the creation of the `.log` and I'm afraid I am not planning to add one at this stage.

Also, unless you are moving from computer to computer, I would suggest that you use the non-portable version, as it will create the log and store other resources in the AppData folder, away from view. My understanding is that most people who use the portable version of Rufus are doing it for the wrong reason, because they erroneously believe that it means that Rufus will not alter the registry, which is not true. The only difference between portable and regular, really, is that Rufus will store its settings in a local `.ini` file rather than the registry, but it still needs to alter the registry to (temporarily) modify group policies and other elements. And the whole point of the portable version is that if you move the Rufus directory from one PC to another, it will preserve your setting such as language and update check. But if you aren't planning the same executable on different computers, I would really advise you to use the regular version, because it usually makes very little sense to use the portable one, _especially_ if you are annoyed by the presence of application files next to the application executable...

For more on this, please see [this FAQ entry](https://github.com/pbatard/rufus/wiki/FAQ#What_is_the_difference_between_the_portable_and_the_regular_version).

As I believe I have answered your question, I will close this issue.

closed 2022-02-09 by pbatard

Hlsgs, 2022-02-09:
> As I believe I have answered your question, I will close this issue.

Indeed, you have.

I am using the portable version of Rufus, as well as other tools that natively support this, so they keep their settings across multiple win installs. I understand it must alter the registry and have no qualms with that.

As always, thank you for the info and your great work, kind Sir!

vitusb, 2022-04-22:
Maybe this is a workaround for you ...

You may create under the execution-path of Rufus-Portable a directory with name "Rufus" and in that directory a directory with the name "rufus.log" ... That's it ...

So we have a directory-path like:
&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp; "C:\temp\Rufusportable-Dir\Rufus\rufus.log\"

When you launch rufusp.exe (my naming of this exe here) from "C:\temp\Rufusportable-Dir", the Logfile will never be created and the directory ".\rufus.log\" will ever be emty 👍🏼 

I think, this will work under Windows 10 Rel 21H2 x64 ...

Best Regards ...

Hlsgs, 2022-04-22:
@vitusb Interesting hack, thank you! But I have arrived at a simpler soluton by installing Rufus portable with Chocolatey and adding a shortcut to it where I need to find/run it. What happens in the Chocolatey folder where the executable gets added is of no concern.