pbatard/rufus #1869: 32-bit program can't run in WinRE or WinPE, requiring a full VM install to use on Linux

Callistemon, 2022-02-11:
Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
Alterative programs that work on Linux, such as balenaEtcher, only allow legacy MBR formatting. This is because those tools use DD copy mode, using the MBR or GPT inbuilt to the ISO. Since ISOs are meant for DVDs, they usually have MBR. It 's a common problem to be unable to create GPT-based Windows media in Linux, as Rufus does not work in Linux. A slow workaround is to install Windows in a virtual machine, which takes hours to complete. If Rufus worked in the Recovery Environment, it can be used in the Windows virtual machine immediately after starting it, by pressing Shift + F10 and inputting a path to Rufus 64, without having to continue with installation.

Rufus is a 32-bit program for compatibility with both 32-bit and 64-bit CPUs. This has one disadvantage. The Windows Recovery Environment on 64-bit systems is not backwards compatible with 32-bit applications, displaying an error while trying to run 32-bit executables. The error message says the program is not compatible on this system and to contact the software developer. I have tested another program, openSeaChest, in the Recovery Environment, and found this error does not appear with the 64-bit version. Please provide a 64-bit compiled version of Rufus. Thank you!

Log
---
```
Rufus x86 v3.17.1846 (Portable)
Windows version: Windows 11 Core, 64-bit (Build 22543.1000)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.06
System locale ID: 0x0409 (en-US)
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
0 devices found
Checking for Rufus updates...
Checking release channel...
No new release version found.
Found USB 3.0 device 'SanDisk SanDisk Ultra USB Device' (0781:5581)
No logical drive found (unpartitioned?)
1 device found
No volume information for drive 0x81
Disk type: Removable, Disk size: 64 GB, Sector size: 512 bytes
Cylinders: 7687, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 0
Disk GUID: {74E8D326-71E1-4F60-93FB-096B9F9F257B}
Max parts: 128, Start Offset: 17408, Usable = 63233292800 bytes
Found USB 3.0 device 'SanDisk SanDisk Ultra USB Device' (0781:5581)
No logical drive found (unpartitioned?)
1 device found
No volume information for drive 0x81
Disk type: Removable, Disk size: 64 GB, Sector size: 512 bytes
Cylinders: 7687, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 0
Disk GUID: {74E8D326-71E1-4F60-93FB-096B9F9F257B}
Max parts: 128, Start Offset: 17408, Usable = 63233292800 bytes
Scanning image...
ISO analysis:
  Image is a UDF image
Disk image analysis:
  Image does not have a Boot Marker
ISO label: 'CCCOMA_X64FRE_EN-US_DV9'
  Detected: Windows 11 ISO (Build 22533.1001)
  Size: 4.6 GB (Projected)
  Uses: EFI
  Uses: Bootmgr (BIOS and UEFI)
  Uses: Install.wim (version 0.13.1)
Using image: Windows11_InsiderPreview_Client_x64_en-us_22533.iso (4.6 GB)

Format operation started
Requesting disk access...
No drive letter was assigned...
[... 976 lines cut ...]
mounted 'E:\sources\boot.wim [2]' on 'C:\Users\Spindle\AppData\Local\Temp\RufAAA.tmp'
Mounted offline registry hive 'C:\Users\Spindle\AppData\Local\Temp\RufAAA.tmp\Windows\System32\config\SYSTEM' to 'HKLM\RUFUS_OFFLINE_HIVE'
Created 'HKLM\SYSTEM\Setup\LabConfig\BypassTPMCheck' registry key
Created 'HKLM\SYSTEM\Setup\LabConfig\BypassSecureBootCheck' registry key
Unmounted offline registry hive 'HKLM\RUFUS_OFFLINE_HIVE'
Unmounting 'E:\sources\boot.wim'...
Unmounted 'C:\Users\Spindle\AppData\Local\Temp\RufAAA.tmp [2]'
Finalizing, please wait...
E:autorun.inf already exists - keeping it

Found USB 3.0 device 'SanDisk SanDisk Ultra USB Device' (0781:5581)
1 device found
Disk type: Removable, Disk size: 64 GB, Sector size: 512 bytes
Cylinders: 7687, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 1
Disk GUID: {7DB0E910-D52C-4FFC-9B5D-F5B37DF3E9AC}
Max parts: 128, Start Offset: 17408, Usable = 63233292800 bytes
Partition 1:
  Type: Microsoft Basic Data Partition
  Name: 'Main Data Partition'
  ID: {0EF9ACE6-9D42-4342-A3DF-48097712C98D}
  Size: 58.9 GB (63232261632 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
```
The log isn't relevant, because this is on a full Windows environment supporting 32-bit, but it's not possible to create a log when it can't be opened, and this post will be deleted if there isn't a log.

pbatard, 2022-02-11:
> Please provide a 64-bit compiled version of Rufus.

You can find a 64-bit version of Rufus in the GitHub Actions artefacts of the VS2022 builds, such as https://github.com/pbatard/rufus/actions/runs/1773813069.

These won't be signed versions, but they are as official as you're gonna get, since they are built in the same manner as the release.

Or if you don't want to run ALPHA/BETA software, you can similarly pick the executables from the latest release [here](https://github.com/pbatard/rufus/runs/3984540484) by clicking the Artifacts button at the top.

Since very few people need to download a straight 64-bit release, and the ones that want one can just install Rufus from the Windows Store (since it includes all of the x86 32+64 and ARM 32+64 executables), I am not planning to officially publish a 64-bit release in the Rufus downloads, on the grounds that I don't want to spend time telling people, who might not have much of a clue about it, which of the 32-bit or 64-bit version they should use, whereas, unless you are explicitly limiting yourself to Windows environments where full 32-bit compatibility is not installed (in which case I will expect __you__ to be tech-savvy enough to locate a 64-bit executable from the GitHub Actions artifacts), the single 32-bit version I am publishing as the standalone executable will always do.

As I believe I have answered your query, I will close this issue, adding that, if you are using Rufus in limited Windows environments, such as RE/PE, I'm afraid that you are on your own (especially if you are doing so because you are using Linux and want to run an application that, as much as I'd like everybody to use Linux instead of proprietary OSes, was never designed to cater for Linux users).

closed 2022-02-11 by pbatard

Callistemon, 2022-02-12:
It works! Yes, that's all I wanted, and will suffice.

> Or if you don't want to run ALPHA/BETA software, you can similarly pick the executables from the latest release [here](https://github.com/pbatard/rufus/runs/3984540484) by clicking the Artifacts button at the top.

The `VS2019.zip` downloads for the regular releases are `(Expired)` and can't be downloaded for that reason. Only the Alpha builds can.

[image]

pbatard, 2022-02-12:
I usually release more frequently than the artifacts expire, but, for varied reasons that I'd rather not go into, I haven't been able to produce a 3.18 release before the 3.17 artifacts expire. At any rate, this is not a big deal since you can __always__ find artifacts with 64-bit builds, be it release or ALPHA/BETA.

monique994, 2022-02-13:
New coding?