pbatard/rufus #1870: Support for cloning from one medium to another directly

MichaIng, 2022-02-13:
﻿

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [ ] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [ ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Feature request
-----------------
Currently with Rufus one can create a VHS from one medium and flash it back to another medium in two steps. I wonder whether a feature could be added to clone a drive directly to another medium, basically another type of input, allowing to select a drive instead of an image file or download.

A question would be how to deal with input drives which are larger than the target/output drive, but basically it could be handled the same way as with an image file. I don't know how it is handled there, never had that case, whether Rufus is strict to not allow flashing then or gives the option to write until the target drive is full (or until a defined size), requiring that the user is sure that partitions on the input end before the target drive ends, to have a working result.

pbatard, 2022-02-14:
What you are asking for is [feature creep](https://en.wikipedia.org/wiki/Feature_creep). Rufus is not and never will be a cloning application, and this feature would sit well outside the mandate of what I am designing Rufus for. Save to VHD is as far as I will ever go with regards to "cloning" capabilities, and even that is outside the mandate of what Rufus was designed for.

The problem is that if I start adding what you ask for, then I'm going to have to satisfy the next person that comes and wants cloning to multiple drives at the same time, or to internal drives, or to different types of drives, and soon enough I have to design __and support__ Rufus as a full blown cloning application, which will take time away from designing and supporting core features, that a lot more people will actually benefit from.

As a result, I have to put a clear limit on how far I am willing to take features that don't fall within the core mandate of Rufus, and cloning drive to drive is clearly out of scope as far as I'm concerned.

If you want to clone drives, then please use a dedicated cloning application. Oh, and you may also want to read [this](https://github.com/pbatard/rufus/wiki/FAQ#Why_cant_you_just_add_the_feature_I_want).

MichaIng, 2022-02-14:
Understood, thanks for your explanation. I was hoping that since saving to VHD and flashing both already works, short-cutting these two steps, skipping the raw => VHD resp. format => raw conversion, would be an obvious and trivial extension. But naturally one oversees the rat's tail that comes with it. Basically I proved what you stated, since only as you added support for saving to VHD, I requested cloning 😄. Good to stick with the original design goal 👍.