pbatard/rufus #1871: Request: Windows Folder Redirection honored

Surf4799, 2022-02-14:
﻿

Checklist
---------
- [X] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [X] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [X] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [X] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
Functionality Impact: None

Request for change:  Honor Windows Folder Redirection path.

Input: Running Rufus 3.17 on Windows computer and have Folder Redirection enabled for the Downloads folder.  The program uses the Downloads path of C:\users\%username%\Downloads.

Expected outcome: Upon clicking the 'Select' button to identify image file, leverage the true user path for the folder.  In my case D:\Downloads. 

Possible evaluation methods: The true path can be found by querying the registry key HKEY_CURRENT_USER\SOFTWARE\Microsoft\Windows\CurrentVersion\Explorer\User Shell Folders.  
An Active Directory policy may also be used to set the path for Folder Redirection (and therefore registry setting noted above).  GPO polices could be applied using https://admx.help/?Category=Citrix_XenApp_and_XenDesktop&Policy=Citrix::00000057FRDownloadsEnable 

Thank you for this great program!  Be well.

Log
---
```
Rufus x86 v3.17.1846 (Portable)
Windows version: Windows 10 Pro, 64-bit (Build 19044.1526)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.06
System locale ID: 0x0409 (en-US)
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Notice: The ISO download feature has been deactivated because 'Check for updates' is disabled in your settings.
Found non-USB removable device 'Samsung SSD 850 EVO 500GB' => Eliminated
Found USB 3.0 device 'Generic MassStorageClass USB Device' (05E3:0749)
Device eliminated because it appears to contain no media
Found USB 3.0 device 'Generic MassStorageClass USB Device' (05E3:0749)
1 device found
Disk type: Removable, Disk size: 64 GB, Sector size: 512 bytes
Cylinders: 7764, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x00000000
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: exFAT (0x07)
  Size: 59.5 GB (63847792640 bytes)
  Start Sector: 32768, Boot: No
Scanning image...
ISO analysis:
  'D:\Downloads\FPP-v5.5-Pi.img\FPP-v5.5-Pi.img' doesn't look like an ISO image
Disk image analysis:
  Image has an unknown Master Boot Record
  Image is a bootable disk image
Using image: FPP-v5.5-Pi.img (3.7 GB)
Search for conflicting processes was interrupted due to timeout

Format operation started
Requesting disk access...
No drive letter was assigned...
Will use 'G:' as volume mountpoint
Opened \\.\PhysicalDrive3 for exclusive write access
Requesting logical volume handle...
Bad Blocks: Checking from block 0 to 121811 (1 block = 512 KB)
Bad Blocks: Using offset 77197 for fake device check
Bad Blocks: Writing test pattern 0x55
```

pbatard, 2022-02-14:
> leverage the true user path for the folder.

You do realize that Rufus runs as Administrator and not the standard user. So `C:\users%username%\Downloads` will resolve to ``C:\users\Administrator\Downloads` which is probably not the user you want.

So my first question is: When running `notepad.exe` elevated on your machine, does it honour your redirection path?
Because if it doesn't, I'm just going to ask you to just browse to `D:\Downloads` if that's where your redirection is, as adding this "feature" sounds like a lot of trouble for very little benefit.

pbatard, 2022-03-03:
No update &rarr; Closed.

closed 2022-03-03 by pbatard