pbatard/rufus #1875: Question about download

cokolwiek555666, 2022-02-22:
﻿

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
<I was using rufus mostly to download older versions of windows. I switched from windows to linux and I see that there isn't version for linux. So I am curious if I could just download these older versions without installing rufus but from the same source?>

Log
---
```
<I don't have it installed so there is no log>
```

JonnyTech, 2022-02-22:
https://tb.rg-adguard.net/

pbatard, 2022-02-22:
> I see that there isn't version for linux.

The reason why there are downloads for Windows is that Microsoft makes it very difficult to gain access to the ISO easily.
On the other hand, downloading Linux versions is usually very easy.

I have considered adding Linux downloads, but this would require a lot of work, with things like having to deal with different mirrors for each distro, making sure everything is being kept up to date and so on (that's the drawback of having to deal with dozens of purveyors of ISOs, as opposed to a single one for Microsoft), so, at this stage, I am not sure I will ever add Linux downloads. Instead, you can just visit the website of whatever Linux distribution strikes your fancy and download the ISO from there, as it's usually a couple clicks away, and you don't really need a utility like Rufus to help you with it.

Hopefully this answers your question.

closed 2022-02-22 by pbatard

cokolwiek555666, 2022-02-22:
OK, thanks @JonnyTech but unfortunately it will be the easiest for me to just use rufus in windows VM or something like that. 
The context is:
Only via rufus I can download very very old version of windows 10, like 1511, and that one is cool because it's small of size and I can burn it in "small" dvd