pbatard/rufus #1877: Report of DLL Hijacking vulnerability cause arbitrary code execution and privilege escalation

ScriptIdiot, 2022-02-24:
**Vulnerable Software and Version**:<br />
1) Rufus 3.17.1846 executable 
2) Rufus 3.17.1846 portable executable 

**Vulnerable software download link**:<br />
https://rufus.ie/en/
https://github.com/pbatard/rufus/releases/tag/v3.17

**Date discovered and reported**: <br />
25 Feb 2022<br />

**Description**: <br />
**Both Rufus 3.17.1846 executable AND portable executable** are suffering from **DLL Hijacking** by placing x86 **MSAN1.dll** or **VERSION.dll** in the same directory as the same directory as the exeutables, which could cause **arbitrary code execution and privilege escalation**.

Taking MSAN1.dll as an example, craft a malicious x86 DLL with an entry point with DllMain and place in the same directory, once double click the executable, an x86 admin shell could be obtained as the executable requires admin right to run by design.

**Attack vector**: <br />
Taking MSAN1.dll as an example
PoC code of dll can be found in my repository

**Attack steps**:<br />
1) Craft and drop a malicious DLL named as "MSAN1.dll" with entry point DllMain
[image]

3) Double click the executable "Rufus", administrator privilege is required to run
4) Malicious DLL has been called and an admin shell can be obtained as PoC
[image]

PoC code can be found here:
https://github.com/ScriptIdiot/Dll-Hijacking-Rufus-3.17.1846

pbatard, 2022-02-24:
I believe this or part of this should have been fixed in f6ac559f4d0a9a2e5043a972442d272c2da71de3 and ef2ff7179d09a7d1ac9e9943d901bd823e422958 as we got the report of the vulnerability with `version.dll` in #1838. Not sure from which dependency the ASN1 DLL comes into play though.

Of course, I'll take a look at it myself, but can you please test with the latest builds that you'll find in the artifacts from https://github.com/pbatard/rufus/actions/runs/1878105382 and https://github.com/pbatard/rufus/actions/runs/1878105381.

I would expect the VS2022 builds to no longer be vulnerable to DLL sideloading on account of ef2ff7179d09a7d1ac9e9943d901bd823e422958, and I also believe that you should no longer be able to use `version.dll` with the MinGW version, though it's possible that `MSASN1.dll` will still work there.

ScriptIdiot, 2022-02-24:
[image]
Just checked the X86 version, its suffering from the hijacking of dll named as MSASN1, will try for others as well

pbatard, 2022-02-24:
Thanks.

So that means that the "delay load" mitigation that we applied to VS builds does not work as expected... That's quite disappointing, but then again, since we can't do that for MinGW builds, it wasn't meant to be used as a bulletproof method to avoid sideloading.

ScriptIdiot, 2022-02-24:
It would be great if CVE can be assigned on DLL Hijacking vulnerability found in both Rufus 3.17.1846 executable
Rufus 3.17.1846 portable executable by hijacking MSASN1.dll

pbatard, 2022-02-25:
Okay, looking at it more closely, I found out that DLL delay loading, that was introduced in ef2ff7179d09a7d1ac9e9943d901bd823e422958, was only applied to MSVC __Debug__ builds and not the Release builds, which explains why the MSVC artifacts were still subject to side loading. With the new patch applied above, you should find that the artifacts from https://github.com/pbatard/rufus/actions/runs/1900092903 are no longer subject to the vulnerability.

Of course, that still doesn't address the issue for MinGW, which I'm planning to do next (since only the Windows Store version is built with MSVC whereas the executables from https://rufus.ie are build with MinGW).

From what I can see ([here](https://www.trustedsec.com/blog/object-overloading/) and [here](https://windows10dll.nirsoft.net/wintrust_dll.html)), `msasn1.dll` is picked from `wintrust`, which means that I may have to drop linking with `wintrust.lib` altogether, which might require some work...

assarbad, 2022-02-28:
> So that means that the "delay load" mitigation that we applied to VS builds does not work as expected... That's quite disappointing, but then again, since we can't do that for MinGW builds, it wasn't meant to be used as a bulletproof method to avoid sideloading.

@pbatard does [this method](https://stackoverflow.com/a/70416894) not work for you? It's a relatively new feature, though. But Clang should also be able to offer the functionality with `clang-cl`.

pbatard, 2022-02-28:
@assarbad, that's good info, though it might be a bit cumbersome to set up. I'll give it a whirl, as I'd sure prefer to stop the madness of having to hook into almost every DLL manually...

pbatard, 2022-03-03:
Since there's only one API call we pick from WinTrust, I'm going to continue going with direct DLL hooking as a workaround for the time being. So the next patch will close this issue and you should no find that none of the executables builds from [GitHub Actions](https://github.com/pbatard/rufus/actions) should be subject to DLL hijacking.

I have however made a note to explore delay loading for MinGW for the 3.19 release of Rufus, as I obviously don't want to end up in the situation where hooking into new DLLs makes us vulnerable to sideloading again, and I also wouldn't mind just being able to call DLL APIs without having to do the whole direct hook gymnastics.

closed 2022-03-03 by pbatard

pbatard, 2022-03-03:
Btw, feel free to test the [newly released Rufus 3.18 BETA](https://github.com/pbatard/rufus/releases/tag/v3.18_BETA) for DLL hijacking, and let me know if you still see anything amiss there.

ScriptIdiot, 2022-03-04:
@pbatard Just tested against Rufus 3.18 BETA, don think its vulnerable now, thanks for the update!

pbatard, 2022-04-12:
> does [this method](https://stackoverflow.com/a/70416894) not work for you?

It works, and I have just applied a patch where we use `dlltool --output-delaylib ...` for MinGW so that we can simplify the code by not having to hook into DLLs manually to prevent side loading, but the `dlltool` approach is more painful than the stackoverflow post makes it look, because 32-bit and 64-bit use different calling conventions, and therefore they cannot use the same `.def` (the 32-bit version requires stdcall `@##` decorations where the 64-bit version just uses plain cdecl undecorated names). So we have to perform extra processing of the `.def` depending on the target arch, and, whereas [the end result looks simple](https://github.com/pbatard/rufus/blob/master/.mingw/Makefile.am#L12-L28), it was a lot of trial and error to get there.

Anyway, moving forward, when linking with new libraries may be subject to side-loading before the application gets a chance to restrict the loading of DLLs to only system directories, we should be able to just add them to the existing MSVC and MinGW mitigation framework, so that delay-loading is used, which should hopefully keep us safe.

pbatard, 2022-04-26:
Yet one more entry in the never-ending story of DLL side-loading prevention.

Unfortunately, it looks like the MinGW method of delay-loading is not yet up to par with the MSVC one, because if you delay-load wininet with MinGW (which we do in 3194a4dac4c1cff66e812c0fc3375189e0b10d31), then Rufus exits before it launches its UI...

I guess the MinGW devs will have to weather-test their delay-loading some more, because (at least for wininet) you get a literal _"That's it, I'm outta here"_ from the runtime when it encounters an issue while trying to delay-load the DLL.

I kind of suspect that this might have to do with the fact that we start a separate thread with some networking tasks on startup, and the current MinGW delay-load hooks may not like that.

At any rate, it looks like I'm going to revert commit 3194a4dac4c1cff66e812c0fc3375189e0b10d31 yet again (which is annoying, since `net.c` was the one place that greatly benefited from no longer having to hook directly into the DLL), so that the MinGW version of Rufus can actually launch.

What a massive pain in the ass this DLL side-loading mitigation continued to be!

commit 19472668 (2022-03-03): [pki] don't link with wintrust.lib

* WinTrust.lib is responsible for the MSASN1.dll sideloading issue described in #1877,
  so, since we only use it for WinVerifyTrustEx(), hook into that function manually.
* Closes #1877 for the MinGW side.
* Note that we will probably try to use the method suggested by @assarbad and documented at
  https://stackoverflow.com/questions/1851267/mingw-gcc-delay-loaded-dll-equivalent/70416894#70416894
  to try to put an end to the problem of DLL side loading.