pbatard/rufus #1878: rufus wreaked 3 usb

grasshoper420, 2022-02-26:
﻿

Checklist
---------
- [ ] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [ ] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [ ] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [ ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
<I put proxmox on 3 different USB and 2 different computers and all 3 USB and not write-protected and makes it into 2 partitions how can I fix the USB? and do you have any other USB tool that won't break my USBs I saw that the dev doesn't want to look into this so I'm I going to get a fix or a reason to why this is happening?>

Log
---
![hgfuvkvk](https://user
[image]
-images.githubusercontent.com/95900766/155857848-22c0300d-c9f0-480d-95ad-100de61e7d08.png)

```
<FULL LOG>
```

pbatard, 2022-02-26:
Your USB aren't "wreaked". They just have data that Windows cannot display, because it's using a file system that it doesn't have native support for.

Please try [this answer from the FAQ](https://github.com/pbatard/rufus/wiki/FAQ#Help_how_do_I_restore_my_USB_to_how_it_was_before_I_created_a_bootable_drive) if you want to restore them.

grasshoper420, 2022-02-26:
make that 4 usb now so what you say I do then

grasshoper420, 2022-02-26:
it says write protected and not 2 partitions and can not formatted can not do anything is there a place qi can put a video and show what I'm doing and show that its rufis

grasshoper420, 2022-02-26:
this [this answer from the FAQ](https://github.com/pbatard/rufus/wiki/FAQ#Help_how_do_I_restore_my_USB_to_how_it_was_before_I_created_a_bootable_drive) has nothing to do with what I'm trying to do I'm trying to put proxmox on USB to boot in my server been doing this for 20 tears but last night it all changed and 4 USB will not work and are white protection can not format can not be used so what now

pbatard, 2022-02-26:
> what you say I do then

1. Post the logs of how you created the drives.
2. __FOLLOW THE CHECK LIST THAT YOU SAW WHEN YOU CREATED THE ISSUE (not it's not optional)__ that asked you to test the drives for bad blocks. If you are seeing failure after failure, chances are that you were sold _fake_ drives, which a bad blocks check would have detected. So please post the result of a log with a full bad blocks check if you want to proceed further, else we're not going to progress any further.

pbatard, 2022-02-26:
> I put proxmox on 3 different USB and 2 different computers and all 3 USB and not write-protected and makes it into 2 partitions

I read this as a statement that your drives were __NOT__ write protected but contained 2 partitions, which will only happen if you write a drive in DD mode using an ISOHybrdid image.

Again, if you want to receive help, please be very careful about the data you provide, as you originally wrote that the drives were not write protected and contained 2 partitions, and now you indicate that they are actually write protected (and empty).

A __FULL LOG__ from Rufus on how you create the drives is indispensable at this stage, along with a bad blocks check, because I'm afraid that the information you provide is not enough to help you.

pbatard, 2022-02-26:
You may be able to find a log in `C:\Users\<your user name>\AppData\Local\Rufus` though I would prefer a log with a bad blocks check. Or you can access the log in Rufus with the small log button (left of START) or <kbd>Ctrl</kbd>-<kbd>L</kbd>.

I will remind you that you __DID__ put a check mark in the following:

- [ ] I ran a bad blocks check, by clicking Show advanced format options then Check device for bad blocks, and confirmed that my USB is not defective.

Was that not true? If it was true, what was the outcome of that check?

grasshoper420, 2022-02-26:
i put the usb in loaded rufis picked USB and picked proxmox it did its thing then at the end its in 2 partitions and are write-protected as for not its to say now and I cant format cant do anything

pbatard, 2022-02-26:
Again, I would like to see a log of how you create your Proxmox drive, along with a bad blocks check, else we're not going to go anywhere.

grasshoper420, 2022-02-26:
there is no log so ill take the loss and won't use it again

pbatard, 2022-02-26:
If you use a modern version of Rufus, it will always save a log at the location I provided. So it's very surprising to hear you say that there is no log. And also, I am puzzled as to why you wouldn't run a bad blocks check if you got an inkling that your drives were failing.

I'm all for trying to help you, but you gotta give me the data I request, and not cherry pick what you like from the checklist. The checklist is not there to annoy you, but to provide the __mandatory__ data that is needed to troubleshoot problems, so if you choose to ignore/dismiss parts of it, then, yes, I guess you're better off not using the application altogether, as we clearly won't be able to come to a resolution of your issue.

closed 2022-02-26 by pbatard