pbatard/rufus #1879: Feature request: Exclude device / card / drive from the list

SCiunczyk, 2022-03-03:
﻿

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [x] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [x] I also tried one or more of the following:
  - [x] Using a different USB drive.
  - [x] Plugging the USB into a different port.
  - [x] Running Rufus on a different computer.
- [x] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

FR description
-----------------
In my Surface Book Pro I have big (512G) SD card installed permanently as and storage expand. This is also popular on Mac users to have similar flush sitting adapters like BaseQi, Transcend JetDrive etc. 

But every time when I'm using Rufus to flash a SD card (very often as I'm designing IoT systems) my heart stops beating if by mistake I don't mess up by flashing fresh Armbian/Raspbian on my precious data storage card. 

The option to exclude some devices from the list would be very useful. 

Maybe in the setup the user could select/check "untouchable devices". And those devices will be hidden from drop-down list on top of the screen or "greyed" on the drop-down list.

Log
---
```
<FULL LOG>
```

pbatard, 2022-03-03:
Please provide a full log (because I need to see the details of your card reader detection to determine whether I want to consider your suggestion or not). It doesn't matter if you don't create a media, but I need to see what Rufus says about your environment. You may also want to use <kbd>Alt</kbd>-<kbd>.</kbd> to enable enumeration debug, which will increase the log verbosity.

Right now, I am considering that this won't affect that many users, and I also have a potential feature in the cards (can't commit to it) where Rufus will add an extra warning, with delay, if it detects that you are trying to format media that isn't the smallest size of all the media it detected, which I think might also achieve what you want.

At this stage, the best I can see me doing would be to allow you to __manually__ set a registry/settings key with part of the ID of the device you want to exclude... but to do that, I __MUST__ see your log.

Also, don't put checkmarks on items that you obviously didn't carry out, like _"I ran a bad blocks check"_. It should come as no surprise that the more a developer obviously sees somebody misfilling an issue report, the less they am inclined to help them... So please, when you are asked to provide data, provide it even if it seems irrelevant, and be truthful with what you __actually__ did.

SCiunczyk, 2022-03-03:
Voila! [rufus.log](https://github.com/pbatard/rufus/files/8180066/rufus.log)

I checked all check marks because I was not sure if some "superbot" will remove my Issue automatically ;-) Sorry.

pbatard, 2022-03-03:
Thanks. I need to think about this, as I'm still not sure the cost/benefit of adding and especially __maintaining__ such a feature will be worth it...

SCiunczyk, 2022-03-03:
Good question. And this feature could be a potential source of complains "why I can't see  my card in drop-down list". 

In my case I'm flashing mostly 16-32 G cards and my storage is 512G. Maybe just some warning if target is bigger than xxx ?

YoYoK3nny, 2022-04-13:
An alternative to "just some warning":

An option `List Large Removable Drives` that when unchecked (by default) Rufus will not list drives matching the following conditions:

- Total capacity is above 32GB (or whatever the actual capacity is for a standard chip-based 32GB drive)
- Wouldn't be eliminated by other features (USB Hard Drives, Internal Drives, etc)

The tooltip for such an option could read:
```
Enable the listing of Large Removable Drives (USB, SD, etc)
USE AT YOUR OWN RISKS!!!
```

P.S: Of course, only if such a feature is decided upon. I feel this is a better alternative to just some warning (since the tooltip will have the warning anyway, much like `List USB Hard Drives` does)

pbatard, 2022-05-25:
Considering that, from what I am seeing, much of the scares come from having a large drive selected by mistake, I have now added code [c76327f96ebbe0f8c54a4f7dc3cd09b80097838e] to list drives by increasing order of size (so that the smallest drive will always appear first). I believe this should _mitigate_ most of the issues that were reported about erroneous drive selection (and yes, I understand that this will __not__ address everything you want, but I also don't think it's healthy to expect that an application that is specifically designed to erase and format drives will be able to read your mind and skip that one drive you also happen to have plugged and don't ever want to ever erase).

At this stage, I am still pondering whether I also want to add a settings-based (registry/rufus.ini) means of skipping specific devices... I'll make a decision about this before the 3.19 release.

closed 2022-06-14 by pbatard

SCiunczyk, 2022-06-15:
In my case I don't know how tu use the solution - my Surface Book Pro internal SD card reader is reported in device manager as disk not an USB device:
[image]

pbatard, 2022-06-15:
Yes, I'm afraid this is only a partial solution, that will only work for USB devices. I am not planning to expand it for SD cards, sorry.

commit cd9d3737 (2022-06-14): [core] add the ability to ignore USBs through VID:PID

* This enables the provision of Registry/Settings key IgnoreUsb01 to IgnoreUsb08 where
  one can specify a USB device to ignore by providing its VID:PID as a 32-bit hex value.
* Closes #1879.
* Also update rufus.ini sample for current Rufus version.
* Also fix status display for Alt-Q.