pbatard/rufus #1883: "no TPM / no Secure Boot menu" option related!

karakarga70, 2022-03-07:
There is an "no TPM / no Secure Boot menu" image option starting from version 3.17. I want to be sure, this image option menu also disables CPU check together. If so, a Rufus user let's say, have a TPM 2.0 module but, may not have a supported processor.

In this case, image maker may not wish to disable the TPM 2.0 module, but only wish to override CPU check!

To do this, a separate 3rd. menu can be added, to only bypass CPU requirement.

pbatard, 2022-03-07:
Yes, the checklist, which you ignored, and that advised to look for similar issue does apply to __YOU__!

Closed as duplicate of #1791, #1809 and so on.

closed 2022-03-07 by pbatard

karakarga70, 2022-03-07:
Disk, Ram, Storage etc. can be upgraded but CPU class can not. I really think, a "CPU bypass only" menu is exactly essential! Yes I know the other options also may arise.

**--> Why not thick boxes for each of the options then? <--**

Your reply: Closed as duplicate of https://github.com/pbatard/rufus/pull/1791, https://github.com/pbatard/rufus/issues/1809 and so on.

Well, close but not I really what I mean....

pbatard, 2022-03-07:
If you read those other issues you will see that I have yet to find evidence that `BypassCPUCheck` actually exists and does anything.

__NOBODY__ so far has been able to point to an example where Windows 11 would not install without this option, but would with.

So, if you think this is "exactly essential", I'll be happy to hear about your evidence. What model of CPU do you have where you can __demonstrate__ that `BypassCPUCheck` works and is required?

Because, until I see verifiable, I'll stick to the idea that `BypassCPUCheck` was proposed, without evidence to back it up, by people who thought that, if there exists a `BypassRAMCheck` and ``BypassTMPCheck` there must also exist a `BypassCPUCheck` or `BypassDiskCheck`, and tried to champion these registry entries, even as they never ever validated that they actually did anything.

karakarga70, 2022-03-07:
I have to choose no TPM option to continue installation for my Xeon Broadwell E5-2682 v4 CPU. There is no TPM option on my Huananzhi X99-TF mainboard.

But, I have an Asus Rampage IV Black edition mainboard as well with an Intel® Core™ i7-4960X Extreme Edition CPU on it. It has an optional feature for a 20-1 pin TPM 2.0 connector. I may order a TPM module to it, and with registry hack option I may try it later some time to see, a CPU incompatibility halts the installation or not.

Bye for now then, good luck....

pbatard, 2022-03-07:
I installed Windows 11 __without `BypassCPUCheck`__ on a machine with a much older CPU, so I'm pretty sure you're going to be fine.

karakarga70, 2022-03-07:
Yes me too, to a 1st. generation Intel(R) Core(TM) i7-2960XM based notebook as well. Even nVidia GT 540M Windows 10 drivers worked as well.

But, with TPM 2.0 installed, unsupported processor? What will be then? Can an original untouched image be truly installable at this case? I am not sure....

pbatard, 2022-03-07:
> But, with TPM 2.0 installed, unsupported processor? What will be then?

The `ByPassTPMCheck` has no incidence whatsoever on the CPU check, so, __no__ installing Windows 11 without `ByPassTPMCheck` will not change anything about the CPU.

You are trying to solve (or rather invite other people to solve) a nonexistent problem that you __believe__ may happen. Please try to solve problems that are backed up with evidence, and next time, only log an issue when you have actual proof that there indeed exists a problem.

karakarga70, 2022-03-07:
You are rejecting to believe it. Well why then there is **BypassCPUCheck** thing....

OK, enough for me, I have tried to inform the possibility about it. You want proof....